raspberrypi/linux #4818: Request to fix compilation errors in downstream patches reported by clang

rsglobal, 2022-01-10:
### Describe the bug

Dear RPI community,  

Please see diff below.  
Please enable clang build in CI to avoid such cases in the future.

### Steps to reproduce the behaviour

Build v5.15 branch using clang

### Device (s)

Raspberry Pi 4 Mod. B

### System

Local build

### Additional context

```diff
diff --git a/drivers/gpu/drm/vc4/vc4_plane.c b/drivers/gpu/drm/vc4/vc4_plane.c
index b323d18b51199..4b51130427efa 100644
--- a/drivers/gpu/drm/vc4/vc4_plane.c
+++ b/drivers/gpu/drm/vc4/vc4_plane.c
@@ -686,7 +686,7 @@ static int vc4_plane_mode_set(struct drm_plane *plane,
 	u32 tiling, src_y;
 	u32 hvs_format = format->hvs;
 	unsigned int rotation;
-	int ret, i;
+	int ret, i = 0;
 
 	if (vc4_state->dlist_initialized)
 		return 0;
@@ -811,7 +811,7 @@ static int vc4_plane_mode_set(struct drm_plane *plane,
 	case DRM_FORMAT_MOD_BROADCOM_SAND128:
 	case DRM_FORMAT_MOD_BROADCOM_SAND256: {
 		uint32_t param = fourcc_mod_broadcom_param(fb->modifier);
-		u32 tile_w, tile, x_off, pix_per_tile;
+		u32 tile_w = 1, tile, x_off, pix_per_tile;
 
 		if (fb->format->format == DRM_FORMAT_P030) {
 			/*
diff --git a/drivers/media/i2c/ov9281.c b/drivers/media/i2c/ov9281.c
index 7d939128899b9..0a5f9e0927aec 100644
--- a/drivers/media/i2c/ov9281.c
+++ b/drivers/media/i2c/ov9281.c
@@ -1124,7 +1124,7 @@ static int ov9281_check_sensor_id(struct ov9281 *ov9281,
 				  struct i2c_client *client)
 {
 	struct device *dev = &ov9281->client->dev;
-	u32 id = 0, id_msb;
+	u32 id = 0, id_msb = 0;
 	int ret;
 
 	ret = ov9281_read_reg(client, OV9281_REG_CHIP_ID + 1,
diff --git a/drivers/media/platform/bcm2835/bcm2835-unicam.c b/drivers/media/platform/bcm2835/bcm2835-unicam.c
index 61b08f655058b..2c2930dc34c52 100644
--- a/drivers/media/platform/bcm2835/bcm2835-unicam.c
+++ b/drivers/media/platform/bcm2835/bcm2835-unicam.c
@@ -3100,7 +3100,7 @@ static int unicam_async_complete(struct v4l2_async_notifier *notifier)
 {
 	struct unicam_device *unicam = to_unicam_device(notifier->v4l2_dev);
 	unsigned int i, source_pads = 0;
-	int ret;
+	int ret = -EINVAL;
 
 	unicam->v4l2_dev.notify = unicam_notify;
 
diff --git a/drivers/misc/bcm2835_smi.c b/drivers/misc/bcm2835_smi.c
index f1a7f6a3e966b..831b7038c78d6 100644
--- a/drivers/misc/bcm2835_smi.c
+++ b/drivers/misc/bcm2835_smi.c
@@ -689,7 +689,7 @@ void bcm2835_smi_write_buf(
 			inst->dev,
 			(void *)buf,
 			n_bytes,
-			DMA_MEM_TO_DEV);
+			DMA_TO_DEVICE);
 		struct scatterlist *sgl =
 			smi_scatterlist_from_buffer(inst, phy_addr, n_bytes,
 				&inst->buffer_sgl);
@@ -702,7 +702,7 @@ void bcm2835_smi_write_buf(
 		smi_dma_write_sgl(inst, sgl, 1, n_bytes);
 
 		dma_unmap_single
-			(inst->dev, phy_addr, n_bytes, DMA_MEM_TO_DEV);
+			(inst->dev, phy_addr, n_bytes, DMA_TO_DEVICE);
 	} else if (n_bytes) {
 		smi_write_fifo(inst, (uint32_t *) buf, n_bytes);
 	}
@@ -748,7 +748,7 @@ void bcm2835_smi_read_buf(struct bcm2835_smi_instance *inst,
 	if (n_bytes > DMA_THRESHOLD_BYTES) {
 		dma_addr_t phy_addr = dma_map_single(inst->dev,
 						     buf, n_bytes,
-						     DMA_DEV_TO_MEM);
+						     DMA_FROM_DEVICE);
 		struct scatterlist *sgl = smi_scatterlist_from_buffer(
 			inst, phy_addr, n_bytes,
 			&inst->buffer_sgl);
@@ -758,7 +758,7 @@ void bcm2835_smi_read_buf(struct bcm2835_smi_instance *inst,
 			goto out;
 		}
 		smi_dma_read_sgl(inst, sgl, 1, n_bytes);
-		dma_unmap_single(inst->dev, phy_addr, n_bytes, DMA_DEV_TO_MEM);
+		dma_unmap_single(inst->dev, phy_addr, n_bytes, DMA_FROM_DEVICE);
 	} else if (n_bytes) {
 		smi_read_fifo(inst, (uint32_t *)buf, n_bytes);
 	}
diff --git a/sound/soc/bcm/allo-piano-dac-plus.c b/sound/soc/bcm/allo-piano-dac-plus.c
index fd0fe58421b06..3efc898c601d7 100644
--- a/sound/soc/bcm/allo-piano-dac-plus.c
+++ b/sound/soc/bcm/allo-piano-dac-plus.c
@@ -63,7 +63,7 @@ static const char * const allo_piano_mode_texts[] = {
 	"2.2",
 };
 
-static const SOC_ENUM_SINGLE_DECL(allo_piano_mode_enum,
+static SOC_ENUM_SINGLE_DECL(allo_piano_mode_enum,
 		0, 0, allo_piano_mode_texts);
 
 static const char * const allo_piano_dual_mode_texts[] = {
@@ -72,7 +72,7 @@ static const char * const allo_piano_dual_mode_texts[] = {
 	"Dual-Stereo",
 };
 
-static const SOC_ENUM_SINGLE_DECL(allo_piano_dual_mode_enum,
+static SOC_ENUM_SINGLE_DECL(allo_piano_dual_mode_enum,
 		0, 0, allo_piano_dual_mode_texts);
 
 static const char * const allo_piano_dsp_low_pass_texts[] = {
@@ -93,7 +93,7 @@ static const char * const allo_piano_dsp_low_pass_texts[] = {
 	"200",
 };
 
-static const SOC_ENUM_SINGLE_DECL(allo_piano_enum,
+static SOC_ENUM_SINGLE_DECL(allo_piano_enum,
 		0, 0, allo_piano_dsp_low_pass_texts);
 
 static int __snd_allo_piano_dsp_program(struct snd_soc_pcm_runtime *rtd,
diff --git a/sound/soc/codecs/ma120x0p.c b/sound/soc/codecs/ma120x0p.c
index c447d37450b7b..ac812e83178da 100644
--- a/sound/soc/codecs/ma120x0p.c
+++ b/sound/soc/codecs/ma120x0p.c
@@ -888,7 +888,7 @@ static const int pwr_mode_values[] = {
 		0x70,
 	};
 
-static const SOC_VALUE_ENUM_SINGLE_DECL(pwr_mode_ctrl,
+static SOC_VALUE_ENUM_SINGLE_DECL(pwr_mode_ctrl,
 	ma_pm_man__a, 0, 0x70,
 	pwr_mode_texts,
 	pwr_mode_values);
```

pelwell, 2022-01-10:
Submit these as Pull Requests against their respective drivers and they will be merged,

6by9, 2022-01-10:
The changes to vc4_plane.c are wrong. There is an error here, but it's not just a case of initialising the variables.

```
			pix_per_tile = tile_w / fb->format->cpp[0];
			x_off = (vc4_state->src_x % pix_per_tile) /
				(i ? h_subsample : 1) * fb->format->cpp[i];
```
looks to have been copied from the DRM_FORMAT_MOD_LINEAR case, where it inside a ```for (i = 0; i < num_planes; i++) {```` loop.

The use of id_msb in ov9281_check_sensor_id is largely irrelevant. In the case that the first I2C transaction fails and sets ret to non-zero which skips the second read. The ```if``` is then ```if (ret || id != CHIP_ID) {``` so whilst ```id``` does have a bogus value, it's never used.
Possibly simpler to move the ```id |= (id_msb << 8);``` inside the conditional as 
```
	if (!ret) {
		ret = ov9281_read_reg(client, OV9281_REG_CHIP_ID,
				      OV9281_REG_VALUE_08BIT, &id_msb);
		id |= (id_msb << 8);
	}
```

rsglobal, 2022-01-10:
Thanks for quick feedback. Unfortunately I do not have much free time here to go through review process.

popcornmix, 2022-01-10:
The `tile_w` issue has a fix submitted upstream https://www.spinics.net/lists/dri-devel/msg326550.html
(but not merged).

popcornmix, 2022-01-24:
I've replaced commits (in rpi-5.15.y):
drm/vc4: plane: Add support for YUV color encodings and ranges
drm/vc4: plane: Add support for DRM_FORMAT_P030

with the versions merged upstream which should resolve the `tile_w` and `i` issues (in the correct way).

rsglobal, 2022-02-17:
Thanks for the fix. I can't see no more clang build related issues on latest v5.15 on my side.

closed 2022-02-17 by popcornmix