raspberrypi/linux #4823: mpv --hwdec=v4l2m2m-copy issue

Dark-Sky, 2022-01-11:
### Describe the bug

Using mpv --hwdec=v4l2m2m-copy to play a some (not all) .mp4 video results in no video but has audio. I went back installing kernels and the last good one was 5.10.52. Then the next kernel I have 5.10.64 was when the issue started. This is the issue started trying to boot with V3D so I do not have any kernels between these kernel versions to test. Eventually this issue has crept in to all of newer kernel trees.

### Steps to reproduce the behaviour

Play the attached video with mpv --hwdec=v4l2m2m-copy ./2minvideo.mp4

### Device (s)

Raspberry Pi 4 Mod. B

### System

[ray@pi4 ~]$ vcgencmd version
Jan  4 2022 18:09:05 
Copyright (c) 2012 Broadcom
version 89012f19c33b91cbe9df3933e2a21375bb4f0274 (clean) (release) (start)

### Logs

Here are the logs with mpv with --msg-level=all=v First the last working kernel then the first non working kernel playing a .mp4 that has the issue.

Kernel 5.10.52 --hwdec=v4l2m2m-copy working:

[ray@pi4 Desktop]$ mpv --msg-level=all=v --hwdec=v4l2m2m-copy ./2minvideo.mp4 
[cplayer] Command line options: '--msg-level=all=v' '--hwdec=v4l2m2m-copy' './2minvideo.mp4'
[cplayer] mpv 0.34.0-dirty Copyright © 2000-2021 mpv/MPlayer/mplayer2 projects
[cplayer]  built on UNKNOWN
[cplayer] FFmpeg library versions:
[cplayer]    libavutil       56.70.100
[cplayer]    libavcodec      58.134.100
[cplayer]    libavformat     58.76.100
[cplayer]    libswscale      5.9.100
[cplayer]    libavfilter     7.110.100
[cplayer]    libswresample   3.9.100
[cplayer] FFmpeg version: n4.4.1
[cplayer] 
[cplayer] Configuration: /usr/bin/waf configure --prefix=/usr --confdir=/etc/mpv --enable-cdda --enable-dvb --enable-dvdnav --enable-libarchive --enable-libmpv-shared --disable-build-date --disable-caca
[cplayer] List of enabled features: alsa asm cdda cplayer cplugins cuda-hwaccel cuda-interop debug-build drm dvbin dvdnav egl egl-drm egl-helpers egl-x11 ffmpeg ffmpeg-aviocontext-bytes-read ffnvcodec gbm gbm.h gl gl-wayland glibc-thread-name glob glob-posix gpl iconv jack javascript jpeg lcms2 libarchive libass libavdevice libbluray libdl libm libmpv-shared libplacebo librt linux-fstatfs linux-input-event-codes lua lua52 memfd_create optimize plain-gl posix posix-or-mingw pthreads pulse rubberband shaderc shaderc-shared stdatomic uchardet vaapi vaapi-drm vaapi-egl vaapi-vulkan vaapi-wayland vaapi-x-egl vaapi-x11 vdpau vector vt.h vulkan wayland wayland-protocols x11 xv zimg zlib
[cplayer] Reading config file /etc/mpv/encoding-profiles.conf
[cplayer] Applying profile 'default'...
[cplayer] Reading config file /home/ray/.config/mpv/config
[cplayer] Applying profile 'default'...
[cplayer] Setting option 'ao' = 'pulse' (flags = 4)
[cplayer] Setting option 'autofit-smaller' = '90%' (flags = 4)
[cplayer] Setting option 'autofit-larger' = '90%' (flags = 4)
[cplayer] Setting option 'volume-max' = '400' (flags = 4)
[cplayer] Setting option 'msg-level' = 'all=v' (flags = 8)
[cplayer] Setting option 'hwdec' = 'v4l2m2m-copy' (flags = 8)
[cplayer] Waiting for scripts...
[osd/libass] libass API version: 0x1502000
[osd/libass] libass source: commit: 0.15.2-0-gc967a5a3d9ec0d36af1148b3fdf2f307a21dd122
[osd/libass] Shaper: FriBidi 1.0.10 (SIMPLE) HarfBuzz-ng 3.2.0 (COMPLEX)
[osd/libass] Setting up fonts...
[osd/libass] Using font provider fontconfig
[osd/libass] Done.
[cplayer] Set property: shared-script-properties -> 1
[cplayer] Set property: shared-script-properties -> 1
[cplayer] Done loading scripts.
[cplayer] Running hook: ytdl_hook/on_load
[ytdl_hook] ytdl:// hook 
[ytdl_hook] not a ytdl:// url 
[cplayer] Set property: shared-script-properties -> 1
[ifo_dvdnav] Opening ./2minvideo.mp4
[bdmv/bluray] Opening ./2minvideo.mp4
[file] Opening ./2minvideo.mp4
[demux] Trying demuxers for level=normal.
[cplayer] Set property: shared-script-properties -> 1
[osd/libass] libass API version: 0x1502000
[... 311 lines cut ...]
AO: [pulse] 48000Hz stereo 2ch float
[cplayer] AO: Description: PulseAudio audio output
[autoconvert] inserting resampler
[swresample] format change, reinitializing resampler
[swresample] 48000Hz stereo floatp -> 48000Hz stereo float
[af] [out] 48000Hz stereo 2ch float
[cplayer] audio ready
[cplayer] starting audio playback
[ao/pulse] starting AO
[vo/gpu/x11] Disabling screensaver.
[cplayer] playback restart complete @ 0.000000, audio=playing, video=eof
[cplayer] Set property: shared-script-properties -> 1
AV: 00:00:01 / 00:02:00 (1%) A-V:  0.000
[cplayer] EOF code: 5  
[vo/gpu/x11] Enabling screensaver.
[cplayer] finished playback, success (reason 3)

Exiting... (Quit)
[cplayer] Set property: shared-script-properties -> 1

### Additional context

Here is a link to a sample 2 minute .mp4 video that has the issue. Like I said above not all .mp4 has the issue but if I remember right I had to change config.txt to get the ones that did work. I believe I used a config used with kodi (dtoverlay=rpivid-v4l2) which I never did when the video was playing correctly.

https://drive.google.com/file/d/1_hC8hIGIQFnyBVwIEanwG_iEbgc7LsuD/view?usp=sharing

stl3, 2022-02-24:
Have you gotten any further with this issue? I'm on the same pi model as you, using latest Dietpi and experiencing the same problem but on youtube videos with these formats:
```
ID  EXT  RESOLUTION FPS │   FILESIZE   TBR PROTO │ VCODEC        VBR ACODEC      ABR     ASR MORE INFO
───────────────────────────────────────────────────────────────────
137 mp4  1920x1080   30 │    2.11MiB 1803k https │ avc1.640028 1803k video only              1080p, mp4_dash
248 webm 1920x1080   30 │  965.31KiB  804k https │ vp9          804k video only              1080p, webm_dash

```
Although it didn't like vp9 on 720p, it still played both audio/video, and mp4/webm were fine with v4l2m2m at 720p
```
❯ vcgencmd version
Jan 20 2022 13:57:40
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start_x)
```

Dark-Sky, 2022-02-24:
No I have not. It seems to be kernel related. I can install an older kernel and it works fine but any kernel after that it will not work. Waiting for some one here to get time to look in to it......

stl3, 2022-02-24:
Yeah, I'm not finding much else info on it either, and when I searched about mpv and v4l2m2m in general I don't get much cept for these and no one else talking about the kernel issue. Guess we'll have to wait.
https://github.com/jellyfin/jellyfin/issues/4892
https://github.com/jellyfin/jellyfin/pull/7227

closed 2024-12-12 by Dark-Sky