raspberrypi/linux #4828: 5.10.89 crashes when booted from u-boot -> grub

colemickens, 2022-01-13:
### Describe the bug

If I build and boot the foundation kernel 5.10.89 (corresponding build for the current stable firmware), I get a problem.

I'm booting via u-boot, which then loads grub's efi application which then tries to boot the foundation kernel.

This is the crash I get early on:
```
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 5.10.89 #1-NixOS                                          
[    0.000000] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)                                                
[    0.000000] pstate: 80000085 (Nzcv daIf -PAN -UAO -TCO BTYPE=--)                                              
[    0.000000] pc : kthread_should_stop+0x20/0x48                                                                
[    0.000000] lr : add_hwgenerator_randomness+0x4c/0x140                                                        
[    0.000000] sp : ffffffe159ca3d90                                                                             
[    0.000000] x29: ffffffe159ca3d90 x28: 0000000014117380                                                       
[    0.000000] x27: 000000002386f040 x26: fffffffefe4bc050                                                       
[    0.000000] x25: 0000000000000018 x24: 0000000000000040                                                       
[    0.000000] x23: fffffffefe4fc044 x22: 0000000000000200                                                       
[    0.000000] x21: ffffffe159cb7c40 x20: fffffffefe4fc040                                                       
[    0.000000] x19: ffffffe159cb3178 x18: 0000000000000020                                                       
[    0.000000] x17: 0000000000001400 x16: 0000000000001c00                                                       
[    0.000000] x15: ffffffe159cb8190 x14: 52455345524d454d                                                       
[    0.000000] x13: 2030343030373833 x12: 3278303d474e5220                                                       
[    0.000000] x11: ffffffe159d473e0 x10: ffffffe159d2f3a0                                                       
[    0.000000] x9 : ffffffe159d2f3f8 x8 : 0000000000017fe8                                                       
[    0.000000] x7 : c0000000ffffefff x6 : 0000000000000001                                                       
[    0.000000] x5 : 0000000000000000 x4 : 0000000000000000                                                       
[    0.000000] x3 : 0000000000000001 x2 : 0000000000000000                                                       
[    0.000000] x1 : 0000000000200000 x0 : 0000000000000000                                                       
[    0.000000] Call trace:                                                                                       
[    0.000000]  kthread_should_stop+0x20/0x48                                                                    
[    0.000000]  add_bootloader_randomness+0x20/0x30                                                              
[    0.000000]  efi_config_parse_tables+0x148/0x270                                                              
[    0.000000]  efi_init+0x174/0x24c                                                                             
[    0.000000]  setup_arch+0x260/0x630                                                                           
[    0.000000]  start_kernel+0x98/0x574                                                                          
[    0.000000] Code: d503233f 36a800c1 f9431000 d50323bf (f9400000)                                              
[    0.000000] random: get_random_bytes called from print_oops_end_marker+0x34/0x60 with crng_init=1             
[    0.000000] ---[ end trace affd80da8a5899f8 ]---                                                              
[    0.000000] Kernel panic - not syncing: Attempted to kill the idle task!                                      
[    0.000000] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---   
```

### Steps to reproduce the behaviour

Use u-boot to boot grub, use grub to boot foundation kernel.

### Device (s)

Raspberry Pi 4 Mod. B

### System

I can't give exact outputs, my current boot process apparently doesn't have the DT configured correctly since I just get "VCHI Initialization failed".

pelwell, 2022-01-14:
1. Is this a new problem, i.e. does it work with earlier kernels?
2. What is the source of the Device Tree in this configuration?
3. The trace is missing the first few lines - please include the cut lines before and after.

P33M, 2022-01-14:
Sounds a lot like https://lore.kernel.org/all/YcTIM+MWEbMGLpRa@light.dominikbrodowski.net/

pelwell, 2022-01-14:
Thanks @P33M - see https://github.com/raspberrypi/linux/pull/4829 for a possible fix.

colemickens, 2022-01-22:
That patch works, I am sitting in `5.10.89` via `tow-boot` -> `grub2-efi` -> patched rpi kernel with #4829.

colemickens, 2022-01-22:
Closing since #4829 was merged. Thanks @P33M and @pelwell !

closed 2022-01-22 by colemickens