raspberrypi/linux #4833: RPi Zero (rev900092) with RPi BCM43143 dongle. WiFi gets kernel oops.

DougieLawson, 2022-01-17:
### Describe the bug

Raspberry Pi Zero (with official Raspberry Pi USB BCM43143 WiFi adapter) with 5.15.13+ c77e197d656c41d308d6f7306e2d7ece0cd72610.

lsusb
```
Bus 001 Device 002: ID 0a5c:bd1e Broadcom Corp. BCM43143 802.11bgn (1x1) Wireless Adapter
```

gets:
```
[   39.599372] 8<--- cut here ---
[   39.611255] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   39.628204] pgd = 456fdb51
[   39.639650] [00000000] *pgd=00000000
[   39.652138] Internal error: Oops: 5 [#1] ARM
[   39.665069] Modules linked in: brcmfmac(+) brcmutil sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon i2c_bcm2835 pwm_bcm2835 spi_bcm2835 bcm2835_codec(C) snd_bcm2835(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_pcm videobuf2_v4l2 videobuf2_common snd_timer snd vc_sm_cma(C) videodev mc uio_pdrv_genirq uio fixed fuse drm drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[   39.735435] CPU: 0 PID: 132 Comm: systemd-udevd Tainted: G         C        5.15.13+ #1510
[   39.753667] Hardware name: BCM2835
[   39.766946] PC is at strlcat+0x28/0xa4
[   39.780284] LR is at brcm_alt_fw_path+0xa0/0xe4 [brcmfmac]
[   39.796334] pc : [<c052ac60>]    lr : [<bf5c0b20>]    psr: 60000013
[   39.812201] sp : c3ae59f8  ip : c3ae5a18  fp : c3ae5a14
[   39.826977] r10: bf5d87c0  r9 : c5fbf880  r8 : c552ce40
[   39.841673] r7 : c5fbf8c0  r6 : bf5d1938  r5 : c3ae5a37  r4 : c0c1d028
[   39.857782] r3 : 00000000  r2 : 00000140  r1 : 00000000  r0 : c3ae5a24
[   39.873979] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   39.890912] Control: 00c5387d  Table: 02684008  DAC: 00000055
[   39.906377] Register r0 information: non-slab/vmalloc memory
[   39.921917] Register r1 information: NULL pointer
[   39.936433] Register r2 information: non-paged memory
[   39.951261] Register r3 information: NULL pointer
[   39.965545] Register r4 information: non-slab/vmalloc memory
[   39.980597] Register r5 information: non-slab/vmalloc memory
[   39.995496] Register r6 information: 75-page vmalloc region starting at 0xbf5a8000 allocated at load_module+0xa74/0x25f8
[   40.015938] Register r7 information: slab kmalloc-32 start c5fbf8c0 pointer offset 0 size 32
[   40.033967] Register r8 information: slab kmalloc-64 start c552ce40 pointer offset 0 size 64
[   40.052016] Register r9 information: slab kmalloc-32 start c5fbf880 pointer offset 0 size 32
[   40.070241] Register r10 information: 75-page vmalloc region starting at 0xbf5a8000 allocated at load_module+0xa74/0x25f8
[   40.091299] Register r11 information: non-slab/vmalloc memory
[   40.106869] Register r12 information: non-slab/vmalloc memory
[   40.122480] Process systemd-udevd (pid: 132, stack limit = 0x8dea4ec8)
[   40.138981] Stack: (0xc3ae59f8 to 0xc3ae6000)
[   40.153312] 59e0:                                                       c0c1d028 00000000
[   40.171614] 5a00: bf5d1938 c5fbf8c0 c3ae5b7c c3ae5a18 bf5c0b20 c052ac44 00000000 2e000000
[   40.189961] 5a20: 006e6962 6d637262 6372622f 616d666d 31333463 002e3334 00006e69 00000000
[   40.208388] 5a40: c0068ff8 c011b19c c3ae5a64 c3ae5a58 c011b19c c0095f3c c3ae5a74 c3ae5a68
[   40.226637] 5a60: c0068e90 c011b13c c3ae5acc c3ae5a78 c006b5ec c0068e44 c3ae5b2c c00f9c0c
[   40.244870] 5a80: c01ee99c 60000013 ffffffff c3ae5adc c3ae5b8c c3ae4000 00000000 00000000
[   40.263162] 5aa0: c0008c90 c0c1d028 c0c1d028 c5fbf880 bf5d8548 c3ae5b8c c3ae5b40 bf5d87c0
[   40.281506] 5ac0: c3ae5ae4 c3ae5ad0 c006b798 c006b458 c3ae5b2c c3ae5b34 c3ae5b04 c3ae5ae8
[   40.299901] 5ae0: c006bd64 c006b774 60000013 ffffffff c5fbf880 c0c1d028 c3ae5b24 c3ae5b08
[   40.318299] 5b00: c08c02bc c006bcf0 c557288a 00000004 c557288a c052aca8 bf5d8548 c0c1d028
[   40.336939] 5b20: c5fbf880 c01efab0 c1001100 c01ef998 c3ae5b7c c3ae5b40 c5fbf8c0 c01efb2c
[   40.355685] 5b40: 344d4342 33343133 00000010 bf5c1a3c c3ae5b7c c5fbf880 c187d880 bf5d1938
[   40.374444] 5b60: c55729c8 14d86122 c5fbf880 c187d880 c3ae5bac c3ae5b80 bf5c1a5c bf5c0a8c
[   40.393073] 5b80: c3ae5b8c 00000001 c3ae5b80 bf5e1c48 c5572800 c1855000 c0c1d028 c55729c8
[   40.411576] 5ba0: c3ae5c14 c3ae5bb0 bf5d32b8 bf5c19b0 00000002 bf5d88a8 c3ae5bec c3ae5bc8
[   40.430187] 5bc0: 00000002 c187d880 20000013 0000a887 00000002 00070000 60000000 00000000
[   40.448791] 5be0: 00000000 14d86122 c1861800 c1855020 c187d800 bf5e3cb8 00000000 c1855000
[   40.467462] 5c00: bf5d88a8 00000000 c3ae5c44 c3ae5c18 c06511b4 bf5d2e70 c06510d8 c1855020
[   40.486378] 5c20: 00000000 bf5e3cb8 c1855020 00000018 c553fb28 00000028 c3ae5c64 c3ae5c48
[   40.505532] 5c40: c05cf69c c06510e4 c1855020 bf5e3cb8 bf5e3cb8 c1855020 c3ae5c7c c3ae5c68
[   40.524750] 5c60: c05cf8f8 c05cf54c c0dcae14 c0dcae18 c3ae5ca4 c3ae5c80 c05cf99c c05cf874
[   40.543725] 5c80: 00000000 c1855020 bf5e3cb8 c0c1d028 00000002 c553fb28 c3ae5cc4 c3ae5ca8
[   40.562762] 5ca0: c05cfdfc c05cf968 c1878730 bf5e3cb8 c05cfd7c c0c1d028 c3ae5cf4 c3ae5cc8
[   40.581754] 5cc0: c05cd674 c05cfd88 c3ae5d00 c11f7dcc c1878730 14d86122 bf5e3cb8 c0d16750
[   40.600696] 5ce0: c553fb40 00000000 c3ae5d04 c3ae5cf8 c05cef7c c05cd600 c3ae5d2c c3ae5d08
[   40.619830] 5d00: c05ce8a0 c05cef60 bf5e2168 c3ae5d18 bf5e3cb8 bf5e3cb8 c0d16750 bf5e3f60
[   40.639280] 5d20: c3ae5d44 c3ae5d30 c05d04bc c05ce704 bf5e3c84 bf5e3cb8 c3ae5d6c c3ae5d48
[... 86 lines cut ...]
### System

```
root@falcon:~ # cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2
root@falcon:~ #
```
```
root@falcon:~ # vcgencmd version
Jan  6 2022 15:40:57
Copyright (c) 2012 Broadcom
version 58e03c94953762222f2b838390dde54d46c38381 (clean) (release) (start)
root@falcon:~ #
```

```
root@falcon:~ # uname -a
Linux falcon 5.15.13+ #1510 Wed Jan 5 12:22:55 GMT 2022 armv6l GNU/Linux
root@falcon:~ #
```

### Logs

See OOPs above

pelwell, 2022-01-17:
I don't have one of those dongles to hand, but by inspection of the code I'd say it's an upstream bug.

5.15 added the ability for board-specific firmware binaries - not just "nvram" files - based on the Device Tree compatible string for the board. This feature is implemented by the new function `brcm_alt_fw_path`, called from `brcmf_fw_get_firmwares` and `brcmf_fw_request_firmware`. `brcmf_fw_request_firmware` checka that `fwctx->req_->board_type` is non-NULL before passing it as a parameter to `brcm_alt_fw_path`, but `brcmf_fw_get_firmwares` doesn't. That NULL pointer is subsequently passed to `strlcat`, causing the failure you've seen.

I'll put a patch together and merge it speculatively, but I'm pretty sure it will prevent the crash.

DougieLawson, 2022-01-17:
Let me know where I can download your patched kernel and I'll give it a go.

pelwell, 2022-01-17:
It will be in the next `BRANCH=next` build (the patch is currently at a0bf936, but it will move with the next rebase).

DougieLawson, 2022-01-18:
Thanks @pelwell @popcornmix commit 6f487a7a1e9d2e73e5e40a89b5e02c96d391fb57 has fixed this one.

closed 2023-08-10 by popcornmix