raspberrypi/linux #4836: Compliation failure: Error .fnend directive without .fnstart

xandris, 2022-01-21:
### Describe the bug

Trying to compile and make 1.20220120 fails with a assember error in arch/arm/lib/memcpymove.h

### Steps to reproduce the behaviour

- Download the source tarball
- Configure and make it

### Device (s)

Raspberry Pi 1 Mod. B+

### System

I'm cross compiling:

- Gentoo on Linux 5.16.0
- Compiling 1.20220120
- 

### Logs

```
  arm-unknown-linux-gnueabihf-gcc -Wp,-MMD,arch/arm/lib/.memcpy_rpi.o.d -nostdinc -isystem /usr/lib/gcc/arm-unknown-linux-gnueabihf/11.2.1/include -I./arch/arm/include -I./arch/arm/include/generated  -I./include -I./arch/arm/include/uapi -I./arch/arm/include/generated/uapi 
-I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -mlittle-endian -fmacro-prefix-map=./= -D__ASSEMBLY__ -fno-PIE -mabi=aapcs-linux -mfpu=vfp -funwind-tables -marm -Wa,-mno-warn-deprecated -D__LINUX_ARM_ARCH__=6 -march=armv6k -mtune=
arm1136j-s -include asm/unified.h -msoft-float    -c -o arch/arm/lib/memcpy_rpi.o arch/arm/lib/memcpy_rpi.S                              
arch/arm/lib/memcpy_rpi.S: Assembler messages:                                                                                           
arch/arm/lib/memcpy_rpi.S:61: Error: .fnend directive without .fnstart                                                                   
arch/arm/lib/memcpy_rpi.S:61: Error: .fnend directive without .fnstart                                                                   
arch/arm/lib/memcpy_rpi.S:61: Error: .fnend directive without .fnstart                                                                   
arch/arm/lib/memcpy_rpi.S:61: Error: .fnend directive without .fnstart                                                                   
make[1]: *** [scripts/Makefile.build:361: arch/arm/lib/memcpy_rpi.o] Error 1
```

### Additional context

I think it was introduced in [a commit from 8 days ago](https://github.com/raspberrypi/linux/commit/6d5309c44a179c4651e9efaca3c0d5cf92105044#diff-03d6e24b6111065023505757d30e77963d512126d5532a116e9339aa680952d2R283).

This patch fixes the issue for me:

```
--- kernel-1.20220120.orig/arch/arm/lib/memcpymove.h    2022-01-20 19:33:57.659681651 -0800
+++ kernel-1.20220120/arch/arm/lib/memcpymove.h 2022-01-20 19:34:19.791681264 -0800
@@ -280,7 +280,6 @@
 199:
         pop     {DAT3, DAT4, DAT5, DAT6, DAT7}
         pop     {D, DAT1, DAT2, pc}
-        UNWIND( .fnend )
 .endm
 
 .macro memcpy_medium_inner_loop  backwards, align
```

xandris, 2022-01-21:
I take that back...the patch makes it build, but the resulting kernel doesn't boot

pelwell, 2022-01-21:
Which version of GCC is that? 9.3.0 doesn't complain (annoyingly).

The patch looks reasonable - I was concerned about the frame offsets caused by the multiple pops, but I guess the function return  makes that a non-issue.

> I take that back...the patch makes it build, but the resulting kernel doesn't boot-

If you think something about https://github.com/raspberrypi/linux/commit/6d5309c44a179c4651e9efaca3c0d5cf92105044#diff-03d6e24b6111065023505757d30e77963d512126d5532a116e9339aa680952d2R283 is stopping your system booting, revert it and rebuild.

pelwell, 2022-01-21:
The correction is in all the affected branches.

popcornmix, 2022-01-24:
@xandris okay to close?

xandris, 2022-01-24:
yes @popcornmix, tyvm