raspberrypi/linux #4837: 2711 HDMI interrupts possibly declared incorrectly

pelwell, 2022-01-21:
### Describe the bug

There are two elements of the HDMI interrupt declaration in bcm2711.dtsi that look incorrect:

1. The CEC interrupts for HDMI1 appear to be backwards w.r.t. HDMI0 (and reality):
    HDMI0:
    ```
			interrupts = <0>, <1>, <2>,
				     <3>, <4>, <5>;
			interrupt-names = "cec-tx", "cec-rx", "cec-low",
					  "wakeup", "hpd-connected", "hpd-removed";
    ```
    HDMI1:
    ```
			interrupts = <8>, <7>, <6>,
				     <9>, <10>, <11>;
			interrupt-names = "cec-tx", "cec-rx", "cec-low",
					  "wakeup", "hpd-connected", "hpd-removed";
    ```
    Notice the 0, 1, 2 and 8. 7, 6 for the CEC interrupts - the RDB headers suggest that HDMI1 should have 6, 7, 8 instead.

2. [An email on the linux-arm-kernel mailing list](http://lists.infradead.org/pipermail/linux-arm-kernel/2022-January/710292.html) mentioned that the declaration of the AON_INTR2 interrupt controller looked wrong - it is declared to take an `IRQ_TYPE_LEVEL_HIGH` interrupt from the GIC, but that the driver requests the edge interrupt flow (`handle_edge_irq`). I can see hotplug interrupts getting through with either the current `IRQ_TYPE_LEVEL_HIGH` or `IRQ_TYPE_EDGE_RISING`, but the fact that the interrupt controller has no form of acknowledge/clear register does suggest it uses edges rather than levels to indicate interrupts to the GIC.

### Steps to reproduce the behaviour

I've not seen a problem yet, but in theory CEC on HDMI1 looks like it could be broken.

### Device (s)

Other

### System

All current kernels.

pelwell, 2022-01-21:
Any thoughts, @mripard?

popcornmix, 2022-01-21:
This rings a bell (I remember debugging this). Ah yes:
https://github.com/popcornmix/linux/commit/e51c2f3429aa9ac1fb4c68e6d3cfd9d029a322ce

pelwell, 2022-01-21:
Ho hum - there's no audit trail or comment to explain.

That just leaves the question about the edge-triggered interrupt.

popcornmix, 2022-01-21:
I have recently tested cec on hdmi0 and hdmi1, so can confirm it that both are working.
I don't know whether we want edge or level (except I thought almost everything used edge).
I could test a patch if you want to propose something more correct, but don't have a cec device to test with.

pelwell, 2022-01-21:
Which kernel would you prefer to test?

pelwell, 2022-01-21:
See https://github.com/raspberrypi/linux/pull/4838 - it cherry-picks cleanly to 5.15 and 5.16.

popcornmix, 2022-01-21:
Any kernel is fine. I'll test when I'm home as cec display is more convenient.

sregoiac, 2022-01-22:
Linux raspberrypi3b 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux

kern.log:
Jan 22 16:28:27 raspberrypi3b kernel: [ 8118.915283] [drm:drm_atomic_helper_wait
_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
Jan 22 16:28:37 raspberrypi3b kernel: [ 8129.155522] [drm:drm_atomic_helper_wait
_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out

Since last kernel update

sregoiac, 2022-01-22:
I start to X. Display goes to blank after inactivity and I'm not able wake up the display with the mouse. But can connect through ssh to the machine. no relevant log on X.org

mripard, 2022-01-24:
I guess using an edge vs level would be fine in most cases, hence why it could work. I couldn't find any info regarding this, so I guess at the time I just assumed it would be level since the rest of the interrupts were level triggered.

popcornmix, 2022-01-24:
I commented on #4838 that the change seems fine with cec and hotplug detect.

pelwell, 2022-01-24:
Since #4838 is merged and cherry-picked, this issue can be closed.

closed 2022-01-24 by pelwell