raspberrypi/linux #4839: Pi ZeroW: Direct firmware load with error -2 (.bin not .txt)

macmpi, 2022-01-21:
### Describe the bug

Seems a new `dmesg` error since 5.15, related to missing `brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.bin` file.

There used to be similar message for `brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt` in the past, which got fixed a while back by a symlink to `brcmfmac43430-sdio.txt` NVRAM parameter file.

But this time it is about actual firmware **.bin** instead...
Do we also need explicit symlink to baseline `brcm/brcmfmac43430-sdio.bin` file now?

### Steps to reproduce the behaviour

Just boot with Alpine 3.15 image

### Device (s)

Raspberry Pi Zero W / WH

### System

```
cat /etc/alpine-release 
3.15.0

sudo /opt/vc/bin/vcgencmd version
Oct 29 2021 10:49:08 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)

uname -a
Linux tiny-radiocd 5.15.4-0-rpi #1-Alpine Mon Nov 22 09:13:22 UTC 2021 armv6l Linux
```

### Logs

```
dmesg | grep brcmfmac
[   53.092882] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   53.103596] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   53.104244] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.bin failed with error -2
[   53.171774] usbcore: registered new interface driver brcmfmac
[   53.628615] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   53.629023] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   53.638602] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Mar 30 2021 01:12:21 version 7.45.98.118 (7d96287 CY) FWID 01-32059766
[   55.550645] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

### Additional context

```
ls -ls /lib/firmware/brcm/brcmfmac43430*
     1 -rwxr-xr-x    1 root     root           875 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.AP6212.txt
     1 -rwxr-xr-x    1 root     root           928 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.Hampoo-D2D3_Vi8A1.txt
     1 -rwxr-xr-x    1 root     root           831 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.MUR1DX.txt
   410 -rwxr-xr-x    1 root     root        419798 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.bin
     5 -rwxr-xr-x    1 root     root          4733 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.clm_blob
     1 -rwxr-xr-x    1 root     root           874 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt
     1 -rwxr-xr-x    1 root     root           874 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt
     1 -rwxr-xr-x    1 root     root           875 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-plus.txt
     1 -rwxr-xr-x    1 root     root           875 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-ultra.txt
     1 -rwxr-xr-x    1 root     root           875 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-zero.txt
     1 -rwxr-xr-x    1 root     root           875 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m3.txt
     1 -rwxr-xr-x    1 root     root           874 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430-sdio.txt
     1 -rwxr-xr-x    1 root     root           989 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430a0-sdio.ONDA-V80 PLUS.txt
   373 -rwxr-xr-x    1 root     root        382455 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430a0-sdio.bin
     1 -rwxr-xr-x    1 root     root           908 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430a0-sdio.ilife-S806.txt
     1 -rwxr-xr-x    1 root     root           951 Oct 29 10:09 /lib/firmware/brcm/brcmfmac43430a0-sdio.jumper-ezpad-mini3.txt
```

pelwell, 2022-01-21:
> Do we also need explicit symlink to baseline brcm/brcmfmac43430-sdio.bin file now?

You only need the symlink if it is important to avoid the error message - the driver uses the unqualified name as a fallback.

macmpi, 2022-01-21:
Thanks Phil.
Just for record, is it new as of 3.15.x, or did it come earlier (never noticed so far)?

pelwell, 2022-01-21:
It's new in 5.15 - https://github.com/raspberrypi/linux/commit/5ff013914c62c493c206d70554cfb1d311ea481a.