raspberrypi/linux #4843: Yellow Tinted HDMI display

vade, 2022-01-24:
### Describe the bug

Hello

My Raspberry Pi 4 via HDMI is showing a yellow tint on screen, regardless if I start with X server or to console.

Here you can see my Raspberry Pi 3's in the background. All devices are running Legacy Raspbian. All devices are using Adafruit HDMI backpack. I can confirm switching displays does not resolve the issue. The Raspberry Pi 4 system showing the issue has HDMI 0 plugged in via a Micro -> full size HDMI cable.

[image]

I'm running a Raspberry Pi 4, with legacy OS. Ive attempted to run `rpi-update` which did not resolve my issue.

Ive looked at both

https://github.com/raspberrypi/linux/issues/3331

and 

https://github.com/raspberrypi/linux/issues/3139

which lead me to `rpi-update` - but did not resolve the issue. 

### Steps to reproduce the behaviour

Unclear, this occurred after a fresh install of Ubuntu via the macOS `Raspberry Pi Imager` using Legacy. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
cat /etc/rpi-issue 
Raspberry Pi reference 2021-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fa45ccf5a4b183ee566b36d74fb4b65bf9358bed, stage4
```

```
vcgencmd version
Jan 24 2022 18:01:22 
Copyright (c) 2012 Broadcom
version 94562b1518ca82ece28042cca1e5cdbbb43c8bda (clean) (release) (start_x)
```

`Linux T4OU1 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux`

### Logs

```
fbset 

mode "800x480"
    geometry 800 480 800 480 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,8/24
endmode
```

```
dmesg | grep -i fb
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:82:E8:9D vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=3ab77013-02 rootfstype=ext4 fsck.repair=yes rootwait
[    1.278414] bcm2708_fb soc:fb: FB found 1 display(s)
[    1.290089] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 800x480
```

```
dmesg | grep -i frame
[    0.002129] LSM: Security Framework initializing
[    1.286710] Console: switching to colour frame buffer device 100x30
[    1.290089] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 800x480
```

My config.txt:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=0

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=800
#framebuffer_height=480

# uncomment if hdmi display is not detected and composite is being output
#Vhdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (here we are forcing 800x480!)
hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 480 60 6 0 0 0
hdmi_drive=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
start_x=1
gpu_mem=256
```

vade, 2022-01-24:
I should also note that the 'gradient screen' that flashes for a moment also has yellow tint - if that is helpful context

popcornmix, 2022-01-25:
> All devices are using Adafruit HDMI backpack. I can confirm switching displays does not resolve the issue.

Was this switching display to another Adafruit HDMI backpack? Can you test with a standard hdmi display (e.g. TV).