raspberrypi/linux #4844: Raspberry Pi 4 : USB3 SSD Connected via USB3 Hub : BOOT Files Corrupted Following Reboot

reraikes, 2022-01-25:
### Describe the bug

Files written to the FAT32 BOOT filesystem of a USB3 SSD connected via a USB3 hub sporadically get corrupted when followed immediately by a reboot.  This issue does not occur when the SSD is connected directly to one of the Raspberry Pi 4 USB3 ports.  This issue only occurs on the FAT32 BOOT filesystem, never on the EXT4 ROOT filesystem.  A 'sync' command prior to the reboot does not eliminate the issue.  This issue does not occur if the hub is connected to one of the Raspberry Pi 4 USB2 ports.

### Steps to reproduce the behaviour

Test environment:

1. Raspbberry Pi 4B 4GB (revision 1.1 or 1.2)

2. SSD: Samsung EVO 860 or Corsair Neutron

3. SATA to USB3 adapter:
Asmedia-based (from 5 different manufacturers) [Bus 002 Device 003: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge]
or
Seagate [Bus 002 Device 003: ID 0bc2:50a0 Seagate RSS LLC FA GoFlex Desk]

4. Powered USB3 Hub:
Realtek-based [Bus 002 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. Hub / Bus 001 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub]
or
VIA Labs-based [Bus 002 Device 004: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub / Bus 001 Device 005: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub]

5. Rapberry Pi OS Bullseye booted on the SSD (not running on an SD card)

The attached 'test' script performs the following actions:

1. Copy the entire file structure from /boot to /SAVED-BOOT
2. Copy the entire file structure from /SAVED-BOOT to /boot
3. Run 'diff -r /SAVED-BOOT /boot' to ensure /boot is identical to /SAVED-BOOT
4. Reboot

Following the reboot, executing a 'diff -r /SAVED-BOOT /boot' sporadically reveals that one or more /boot files are corrupt (frequency and degree of corruption varies).
[test.zip](https://github.com/raspberrypi/linux/files/7930631/test.zip)

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2

Jan 20 2022 13:56:48
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux

pelwell, 2022-01-27:
The firmware isn't writing to the boot partition, and the hardware doesn't know the difference between the boot partition and the rest of the drive, so the problem is likely to be in the kernel or higher.

I would start by putting an explicit `sync` in your test script after the copy. You can also use `sudo sh -c "echo 3 > /proc/sys/vm/drop_caches"` to cause any cached files to be invalidated, meaning that your diff will actually be checking the drive contents.

pelwell, 2022-01-27:
The good news is that I think adding the NONEXCLUSIVE support to DT is straightforward:
1. Add an entry to the `of_gpio_flags` enum.
2. Add an equivalent flag to `include/dt-bindings/gpio/gpio.h`.
3. Check for the new bit in `gpiod_get_from_of_node`, setting it in `dflags`.

I'm hoping that's it.

popcornmix, 2022-01-27:
@pelwell was this comment intended for https://github.com/raspberrypi/linux/pull/4812

pelwell, 2022-01-27:
Yes.

pelwell, 2022-01-27:
The point is to empty the cache before the verification, to ensure that the files are being read from the drive.

reraikes, 2022-01-28:
>The point is to empty the cache before the verification, to ensure that the files are being read from the drive.

Adding the 'drop_caches' allows the problem to be observed without the need for a reboot and eliminates reboots as the source of the problem.  The problem appears to simply be intermittent disk write failures to the FAT32 filesystem when a USB3 hub is used.

The problem occurs when a USB3 SSD or HD is connected via a powered USB3 hub (Realtek- or VIA Labs-based) but never occurs when it's connected directly to one of the Raspberry Pi 4 USB ports. 

The updated 'test' script attached to this comment contains the added 'drop_caches' and reliably displays the intermittent failures without the need for a reboot and manual 'diff' afterward.
[test.zip](https://github.com/raspberrypi/linux/files/7955376/test.zip)

XECDesign, 2022-01-28:
May or may not be relevant: https://forums.raspberrypi.com/viewtopic.php?t=244948

pelwell, 2022-01-28:
As was asked on that thread:

> What does hdparm -W report? Do you get the same corruption if you disable the drive write cache with hdparm -W0?

reraikes, 2022-01-28:
>What does hdparm -W report?

write-caching =  1 (on)

>Do you get the same corruption if you disable the drive write cache with hdparm -W0?

Yes.  The problem is unchanged with 'write-caching =  0 (off)'.

P33M, 2022-02-02:
Does disabling LPM on the device make a difference?

From lsusb find the vid:pid pair for the device, for example:
```
$ lsusb
Bus 002 Device 003: ID 05e3:0743 Genesys Logic, Inc. SDXC and microSDXC CardReader
Bus 002 Device 004: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
The mass-storage device in question is `05e3:0743`

Add the following parameter to `/boot/cmdline.txt`:
`usbcore.quirks=05e3:0743:k`

And reboot, then run your test.

reraikes, 2022-02-02:
Bus 002 Device 002: ID 174c:55aa ASMedia Technology Inc. ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ...

Adding 'usbcore.quirks=174c:55aa:k' to /boot/cmdline.txt has no effect on the problem using the attached 'test' script:

Ok to run test (y/n)? y

Saving BOOT partition

Restoring BOOT partition

Syncing + Dropping Caches

Comparing BOOT partition
Binary files /SAVED-BOOT/overlays/spi5-1cs.dtbo and /boot/overlays/spi5-1cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi5-2cs.dtbo and /boot/overlays/spi5-2cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi6-1cs.dtbo and /boot/overlays/spi6-1cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi6-2cs.dtbo and /boot/overlays/spi6-2cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-gpio35-39.dtbo and /boot/overlays/spi-gpio35-39.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-gpio40-45.dtbo and /boot/overlays/spi-gpio40-45.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-rtc.dtbo and /boot/overlays/spi-rtc.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1306.dtbo and /boot/overlays/ssd1306.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1306-spi.dtbo and /boot/overlays/ssd1306-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1331-spi.dtbo and /boot/overlays/ssd1331-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1351-spi.dtbo and /boot/overlays/ssd1351-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/superaudioboard.dtbo and /boot/overlays/superaudioboard.dtbo differ
Binary files /SAVED-BOOT/overlays/sx150x.dtbo and /boot/overlays/sx150x.dtbo differ

Restored BOOT partition is corrupt

[test.zip](https://github.com/raspberrypi/linux/files/7989509/test.zip)

pelwell, 2022-02-02:
Can you put a sample of those corrupted files and the originals in one or two zip files and attach them? I'm curious to see the nature of the corruption.

reraikes, 2022-02-02:
Ok to run test (y/n)? y

Saving BOOT partition

Restoring BOOT partition

Syncing + Dropping Caches

Comparing BOOT partition
Binary files /SAVED-BOOT/overlays/spi5-1cs.dtbo and /boot/overlays/spi5-1cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi5-2cs.dtbo and /boot/overlays/spi5-2cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi6-1cs.dtbo and /boot/overlays/spi6-1cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi6-2cs.dtbo and /boot/overlays/spi6-2cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-gpio35-39.dtbo and /boot/overlays/spi-gpio35-39.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-gpio40-45.dtbo and /boot/overlays/spi-gpio40-45.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-rtc.dtbo and /boot/overlays/spi-rtc.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1306.dtbo and /boot/overlays/ssd1306.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1306-spi.dtbo and /boot/overlays/ssd1306-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1331-spi.dtbo and /boot/overlays/ssd1331-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1351-spi.dtbo and /boot/overlays/ssd1351-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/superaudioboard.dtbo and /boot/overlays/superaudioboard.dtbo differ

Restored BOOT partition is corrupt

[files.zip](https://github.com/raspberrypi/linux/files/7990031/files.zip)

pelwell, 2022-02-03:
Interesting - the corrupted files appear to contain valid chunks of overlays in a scrambled order. Running `md5sum` on individual sectors shows some whole sectors migrating between files. For example, the first sector of the BAD spi5-2cs.dtbo is identical to the third sector of the GOOD spi5-2cs.dtbo and spi6-2cs.dtbo. There would probably be more evidence of duplication if not for the fact that the short files mean a large percentage of the sectors are not used in their entirety.

The fact that the ext4 partition never shows this problem suggests this is the result of an interaction between the FAT filesystem and the SSD. Does `sudo fsck.vfat /dev/sda1` (or whatever the device name is) show any problems after the copy?

reraikes, 2022-02-05:
This issue is definitely related to writing files to a FAT32 filesystem if the USB3 SSD/HD is connected via a USB3 hub (the problem does not occur if the USB3 SSD/HD is connected directly to a Raspberry Pi 4 USB3 port):

1. The problem occurs with any FAT32 filesystem on any partition (for example, a /dev/sda3 partition formatted as FAT32), not just the BOOT partition (/dev/sda1).

2. The problem goes away if the FAT32 filesystem is reformatted to EXT4.

It appears there is a bug in the Raspberry Pi 4 USB3 driver when a USB3 hub is used in conjunction with a FAT32 filesystem.

P33M, 2022-02-07:
Does an explicit unmount of the target filesystem before reboot prevent corruption from happening?
What happens if you mount the fat32 partition with "-o flush" and do the test?

P33M, 2022-02-07:
I'm using your second test script on a fresh (apt updated) Bullseye install. It's on a 2GB Pi 4 but that shouldn't matter. I'm using an SSD with a standard Raspberry Pi OS partition layout - /dev/sda1 is a 512M fat32 partition - and the fat32 partition is mounted on e.g. /home/pi/fat32 in place of the /boot location.

I don't see any filesystem corruption. Can you clarify
- Where the fat32 FS is being mounted
- Whether you have overlayFS enabled or not

reraikes, 2022-02-07:
There is no mounting of anything nor any rebooting involved.  There is no overlayFS involved. 

Just a virgin Raspberry Pi OS loaded/booted/running on a USB3 SSD **which is connected to a USB3 powered hub which is connected to a Raspberry Pi 4 USB3 port**:

`
echo "Saving BOOT partition"
rm -r /SAVED-BOOT &> /dev/null
mkdir /SAVED-BOOT
cp -p -r -T /boot /SAVED-BOOT

echo ""
echo "Restoring BOOT partition"
cp --preserve=timestamps -r -T /SAVED-BOOT /boot

echo ""
echo "Syncing + Dropping Caches"
sync
echo 3 > /proc/sys/vm/drop_caches

echo ""
echo "Comparing BOOT partition"
diff -r /SAVED-BOOT /boot
if [ $? -ne 0 ]; then
  errexit "Restored BOOT partition is corrupt"
fi

echo ""
echo "Restored BOOT partition is intact"
echo ""
`

P33M, 2022-02-07:
Please post the output of `mount`.

reraikes, 2022-02-07:
root@raspberrypi:~# mount
/dev/sda2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=1799320k,nr_inodes=449830,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=786008k,nr_inodes=819200,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate,memory_recursiveprot)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
/dev/sda1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/0 type tmpfs (rw,nosuid,nodev,relatime,size=393000k,nr_inodes=98250,mode=700)

P33M, 2022-02-08:
I can get corruption of the sort you describe to varying degrees of brokeneness if the adapter I am using has a known interop issue with UAS. The adapter can reliably boot, but using the test script has a high chance of causing transfer timeouts if there is a hub in the path. After running your script and getting a bad result, please post the full output of `dmesg`.

If there are errors relating to UAS, then follow the guide here and re-test:

https://forums.raspberrypi.com/viewtopic.php?t=245931

reraikes, 2022-02-08:
The 'test' script failed 6 times in a row before I was able to get a run that didn't fail.  The only thing that showed up in dmesg was:

[35096.662229] test (1893): drop_caches: 3
[35098.417446] test (1893): drop_caches: 3
[35100.206815] test (1893): drop_caches: 3
[35102.131658] test (1893): drop_caches: 3
[35104.506642] test (1893): drop_caches: 3
[35106.296305] test (1893): drop_caches: 3
[35108.065804] test (1893): drop_caches: 3

I added 'usb-storage.quirks=hhhh::hhhh:u' anyway, but it did not have any effect on this issue.

I have 6 different manufacturer's Asmedia-based SATA-to-USB adapters, all of which have proper support of UAS but exhibit this issue, but 'usb-storage.quirks' has no effect on any of them regarding this issue.  I had tested all of them and ruled out UAS as the source of the problem before opening this ticket.  They all work100% reliably connected directly to the Raspberry Pi 4B (no USB3 hub in the loop).  I don't own any adapters with UAS problems.

This issue occurs on all MSD's, including a new Samsung EVO 860 SSD, a Corsair Neutron SSD, and a Seagate BarraCuda HD (ST4000LM024).  It even occurs on faster USB3 flash drives such as a 64 GB Sandisk Extreme (but with less frequency) when a USB3 hub is used.

I have a hard time believing this is a UAS problem since only FAT32 filesystems are affected.

reraikes, 2022-02-09:
Attached is an enhanced version of the 'test' script you've been using to reproduce the FAT32 file corruption issue.

The attached 'test-retry' script performs the same operations as the 'test' script, but by default, suppresses the output of 'diff' and retries the restoration copy until it succeeds, displaying the retry count along the way:

***
root@raspberrypi:~# ./test-retry

Ok to run test (y/n)? y

Saving BOOT partition

Restoring BOOT partition
[ Failed / Retrying: 9 ]

BOOT partition successfully restored
***

This is a more convenient way to illustrate the severity of problem and ensures you're not left with corrupted files at the conclusion of the test.

If the 'test-retry' script is run with a '--noretry' option, it functions identically to the 'test' script you've been using:

***
root@raspberrypi:~# ./test-retry --noretry

Ok to run test (y/n)? y

Saving BOOT partition

Restoring BOOT partition
Binary files /SAVED-BOOT/overlays/spi5-1cs.dtbo and /boot/overlays/spi5-1cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi5-2cs.dtbo and /boot/overlays/spi5-2cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi6-1cs.dtbo and /boot/overlays/spi6-1cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi6-2cs.dtbo and /boot/overlays/spi6-2cs.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-gpio35-39.dtbo and /boot/overlays/spi-gpio35-39.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-gpio40-45.dtbo and /boot/overlays/spi-gpio40-45.dtbo differ
Binary files /SAVED-BOOT/overlays/spi-rtc.dtbo and /boot/overlays/spi-rtc.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1306.dtbo and /boot/overlays/ssd1306.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1306-spi.dtbo and /boot/overlays/ssd1306-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1331-spi.dtbo and /boot/overlays/ssd1331-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/ssd1351-spi.dtbo and /boot/overlays/ssd1351-spi.dtbo differ
Binary files /SAVED-BOOT/overlays/superaudioboard.dtbo and /boot/overlays/superaudioboard.dtbo differ

Restored BOOT partition is corrupt
***

[test-retry.zip](https://github.com/raspberrypi/linux/files/8029026/test-retry.zip)

P33M, 2022-02-09:
I can run your script dozens of times without a broken filesystem happening.

Whatever the issue is, it must be down to your hardware. Booting from USB SSDs is a very common use-case, and it is recommended to use a powered hub as SSDs can draw in excess of 1.5A - causing brownouts on the Pi's USB ports. If there was a software bug, then every time people ran rpi-update or updated the apt package that pulled a new kernel in, we would have thousands of reports of corrupt boot filesystems.
```
[ 130.998022] test (728): drop_caches: 3
[ 132.699318] usb 2-1.3: reset SuperSpeed Gen 1 USB device number 4 using xhci_hcd
```
This line implies that Linux is unhappy with the device for some reason. Resetting the device will obviously cause corruption if there are pending writes.

Please attach the output of the `raspinfo` script.

reraikes, 2022-02-09:
>I can run your script dozens of times (with UAS quirk applied) without a broken filesystem happening.

So you're testing on a system that requires a UAS quirk to be applied for normal operation?  Do you not have an adapter that properly supports UAS?  I can't test using an adapter that requires a UAS quirk as I don't own one.

>Whatever the issue is, it must be down to your hardware. Booting from USB SSDs is a very common use-case, and it is recommended to use a powered hub as SSDs can draw in excess of 1.5A - causing brownouts on the Pi's USB ports. 

But I have absolutely zero problems when no powered hub is used.  It's a powered hub (multiple brands) that's causing the problem, not eliminating it.  I can use any combination of multiple Raspberry Pi 4B's, multiple adapters, multiple hubs, multiple MSD's, and the problem only occurs on a partition that is formatted FAT32, which can be located anywhere on the drive, not just /dev/sda1, and only when a hub is used.

>This line implies that Linux is unhappy with the device for some reason. Resetting the device will obviously cause corruption if there are pending writes.

I agree the resets look out of place, but as many test runs illustrate, the file corruption often occurs without any intervening resets.

[raspinfo.zip](https://github.com/raspberrypi/linux/files/8034502/raspinfo.zip)

reraikes, 2022-02-09:
As further evidence this issue is not related to a particular Raspberry Pi 4B nor a particular USB3 MSD nor a particular USB3 SATA-to-USB adapter nor a particular USB3 hub, I ran tests using a 64G USB3 Sandisk Extreme flash drive.

The first tests were with no hub plugged into the Raspberry Pi 4B (rev 1.2).  The USB3 flash drive was plugged directly into one of the Pi's USB3 ports (both USB3 ports exhibit the problem and no other USB ports were in use).  I ran 12 tests with no corrupted files encountered (errors never occur with a USB3 MSD plugged directly into one of the Pi's USB3 ports):

[  148.134491] test (700): drop_caches: 3
[  183.063953] test (708): drop_caches: 3
[  190.340006] test (716): drop_caches: 3
[  203.408303] test (724): drop_caches: 3
[  216.643790] test (732): drop_caches: 3
[  231.370521] test (740): drop_caches: 3
[  241.080061] test (749): drop_caches: 3
[  248.930747] test (757): drop_caches: 3
[  270.774901] test (765): drop_caches: 3
[  274.984036] test (773): drop_caches: 3
[  280.912935] test (781): drop_caches: 3
[  304.480693] test (789): drop_caches: 3

See the attached dmesg-no-hub.txt.

Then I ran the same tests with the same USB3 flash drive plugged into a USB3 hub (VIA Labs-based) which was plugged into the same Pi's same USB3 port.  I ran the same 12 tests with no corrupted files encountered until the very last one:

[   40.038678] test (713): drop_caches: 3
[   51.059094] test (721): drop_caches: 3
[   63.064206] test (731): drop_caches: 3
[  103.491772] usb 2-1.4: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  104.133090] test (739): drop_caches: 3
[  115.591212] test (748): drop_caches: 3
[  123.479040] test (756): drop_caches: 3
[  160.838845] usb 2-1.4: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  163.515253] test (764): drop_caches: 3
[  176.261146] test (772): drop_caches: 3
[  202.890249] test (780): drop_caches: 3
[  211.017506] test (788): drop_caches: 3
[  219.464898] test (796): drop_caches: 3
[  227.152182] usb 2-1.4: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  227.803767] test (804): drop_caches: 3
[  229.869840] test (804): drop_caches: 3

See the attached dmesg-with-hub.txt.

I then substituted a Realtek-based USB3 hub for the VIA Labs-based USB3 hub used in the previous tests, with no change.

I then substituted another Raspberry Pi 4B (rev 1.1) for the Raspberry Pi 4B (rev 1.2) used in the previous tests, with no change.

The problem appears to be related to the Raspberry Pi 4B handling of MSD's connected to USB3 hubs when writing large quantities of files to FAT32 filesystems.

[dmesg-more.zip](https://github.com/raspberrypi/linux/files/8036061/dmesg-more.zip)

P33M, 2022-02-10:
There is a bug here. I swapped to using a mass-storage-only drive caddy, but on a default install it still didn't register corruption. I noticed that you are using the v8 kernel (64bit) and I _do_ get a failing test in this case. It happens on both 5.10 and 5.15.

Using a protocol analyser, the device is being reset because the previous write command failed and the endpoint stalled. The write fails because the amount of data transferred to the OUT endpoint exceeds the length in the CDB - and this appears to only happen if there is a large contiguous transfer - in my case it's failing on a 710KiB write length, and an excess 8KiB of data was sent.

So to trigger the bug we need at least:
- large contiguous blocks of data queued for a write
- Possibly with multiple outstanding writes
- With an AARCH64 kernel?

Can you repeat your test with the 32-bit kernel. It's possible that whatever set of circumstances is required to trigger the bug are simply less likely on ARMv7 (and I'd like to eliminate it as a contributing factor).

P33M, 2022-02-10:
Do you believe this bug to be a regression? What is the first kernel version or operating system release that exhibits it?

reraikes, 2022-02-10:
It's my belief this bug has been present since the Raspberry Pi 4B was first released (I pray it's not in the Raspberry Pi hardware).

I first noticed it when I wrote MultiBoot back in mid-2020:

https://forums.raspberrypi.com/viewtopic.php?t=282912

Since I could only reproduce the file corruption by copying the entire contents of /boot back from a saved location, I couldn't bring myself to believe it was a bug in the OS.  I assumed I had bad hardware of some sort. I finally concluded it was not my hardware and that there was not a reasonable work-around and opened this issue.

reraikes, 2022-02-15:
Any progress in isolating and fixing the bug?

pelwell, 2022-02-15:
The bug can be seen in a USB trace, and it's not currently looking like a software issue (although there may be a workaround).

P33M, 2022-02-15:
Yes. Lots.

As a workaround, add "flush" to the mount options for /boot in /etc/fstab. FAT volumes mounted by the desktop automounter already have this option set and seem to be immune to corruption.

reraikes, 2022-02-15:
>The bug can be seen in a USB trace, and it's not currently looking like a software issue

Does that equate to a suspected flaw in silicon?

>As a workaround, add "flush" to the mount options for /boot in /etc/fstab.

While at first glance this appears to eliminate the file corruption, it causes the file copy (cp --preserve=timestamps -r -T /SAVED-BOOT /boot) time to go from about 0.5 seconds to more than 30 seconds on a fast SSD (a 60X performance hit)!  The horrible kludge I'm currently using of repeatedly issuing the copy over and over until all files arrive intact is considerably faster.

P33M, 2022-02-15:
> > The bug can be seen in a USB trace, and it's not currently looking like a software issue
> 
> Does that equate to a suspected flaw in silicon?

There is as yet insufficient data for a meaningful answer.

> While at first glance this appears to eliminate the file corruption, it causes the file copy (cp --preserve=timestamps -r -T /SAVED-BOOT /boot) time to go from about 0.5 seconds to more than 30 [...]

Using the workaround is recommended, as depending on the behaviour of the USB to SSD bridge the corruption may not be limited to the file clusters being written by the copy command.

reraikes, 2022-02-15:
>Using the workaround is recommended, as depending on the behaviour of the USB to SSD bridge the corruption may not be limited to the file clusters being written by the copy command.

I haven't been thoroughly convinced at this point that the addition of the 'flush' option is a guaranteed fix.

This is extremely troubling, at best.

max17048, 2022-02-15:
I am seen and fix this a couple of months ago. I used a lot of high write intensive home-brewn apps on SD-Card in an Argon ONE V2 case. Then I change to the M.2-Version with 1TB WD Red SA500 SSD, connectet to USB3. Fixed the brownouts by spending an extra power supply to the SSD. No Problems on installing Buster-Lite (~9/2021) /Rpi-Imager. After installing my apps, the OS stalls after reboot; appr. 8/10 trys.

By try & error, I found the reason in my case in systemd (.path). I use a lot of `recodemedia@<directory_name>.path` tasks (e.g. `recodemedia@Documentations.path`) which are enabled; eg. `systemctl enable recodemedia@Documentations.path`. After reboot, the OS stalls, login by terminal randomly possible but full of worst behavior. The `*@<directory_name>.path` is officially not supportet but works well. On a SD-Card w/ exact the same setup, there are no problems. 

After I does‘nt enable the `.path`-file for each needed directory, the system works like a charm (now they are started by script and timer).

Maybe this points to a solution.

reraikes, 2022-02-15:
>Using the workaround is recommended, as depending on the behaviour of the USB to SSD bridge the corruption may not be limited to the file clusters being written by the copy command.

Why isn't the ROOT filesystem (or any filesystem connected via a USB hub) also potentially vulnerable due to this bug?

reraikes, 2022-02-16:
Thanks for the detailed explanation.

>I have a fix that subdivides any SG buffer into two chained TRBs if it's not a multiple of 1024 and the failing test script now passes. I tested up to a hub depth of 4 and it still seems to work.

Any thoughts on when you might have something available for others to test?

P33M, 2022-02-17:
I have at least one UAS device that fails in the same manner with the test script and is fixed with buffer splitting. So that's enough to apply buffer splitting to all modes of bulk OUT endpoint operation.

P33M, 2022-02-17:
To jump the queue on an rpi-update release, follow the kernel building guide here - https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

and instead clone the branch `git clone --depth=1 --branch rpi-5.15.y https://github.com/raspberrypi/linux`

Note: given that this bug involves data loss, I wouldn't recommend testing with any SSD that has data on that you care about.

reraikes, 2022-02-18:
I built 5.10.95 and 5.15.23 kernels (both 32- and 64-bit versions of each).  While the frequency of file corruption is considerably reduced, it's by no means eliminated.  The "reset SuperSpeed Gen 1 USB device number 4 using xhci_hcd" entries in dmesg have been completely eliminated, however.

The attached 'test-repeat' script performs the same copy/compare operations as before but repeats until corrupted files occur or a 'q' or 'Q' is entered to stop it.  Dots are displayed for each successful pass.

Be patient, as it often succeeds for quite a while before corrupted files occur.  If you get tired of waiting, stop it ('q' or 'Q') and restart it (answering 'y' to "Saved BOOT partition exists, use it (y/n)?").  It's not clear why the failures seem to subside periodically but stopping the test and restarting it usually make them resume for a while.

[test-repeat.zip](https://github.com/raspberrypi/linux/files/8094802/test-repeat.zip)

reraikes, 2022-02-18:
Sometimes the errors take a really long time to occur (other times, not so much):

*****
root@raspberrypi:~# ./test-repeat

Ok to run test (y/n)? y

Saving BOOT partition

Restoring BOOT partition
....................................................................................................................Binary files /SAVED-BOOT/overlays/qca7000-uart0.dtbo and /boot/overlays/qca7000-uart0.dtbo differ

Restored BOOT partition is corrupt

root@raspberrypi:~# ./test-repeat

Ok to run test (y/n)? y

Saved BOOT partition exists, use it (y/n)? y

Restoring BOOT partition
...Binary files /SAVED-BOOT/overlays/vc4-kms-dsi-lt070me05000-v2.dtbo and /boot/overlays/vc4-kms-dsi-lt070me05000-v2.dtbo differ
Binary files /SAVED-BOOT/overlays/vc4-kms-kippah-7inch.dtbo and /boot/overlays/vc4-kms-kippah-7inch.dtbo differ
Binary files /SAVED-BOOT/overlays/vc4-kms-v3d.dtbo and /boot/overlays/vc4-kms-v3d.dtbo differ
Binary files /SAVED-BOOT/overlays/vc4-kms-v3d-pi4.dtbo and /boot/overlays/vc4-kms-v3d-pi4.dtbo differ
Binary files /SAVED-BOOT/overlays/vc4-kms-vga666.dtbo and /boot/overlays/vc4-kms-vga666.dtbo differ
Binary files /SAVED-BOOT/overlays/vga666.dtbo and /boot/overlays/vga666.dtbo differ
Binary files /SAVED-BOOT/overlays/vl805.dtbo and /boot/overlays/vl805.dtbo differ
Binary files /SAVED-BOOT/overlays/w1-gpio.dtbo and /boot/overlays/w1-gpio.dtbo differ
Binary files /SAVED-BOOT/overlays/w1-gpio-pullup.dtbo and /boot/overlays/w1-gpio-pullup.dtbo differ
Binary files /SAVED-BOOT/overlays/w5500.dtbo and /boot/overlays/w5500.dtbo differ
Binary files /SAVED-BOOT/overlays/wittypi.dtbo and /boot/overlays/wittypi.dtbo differ
Binary files /SAVED-BOOT/overlays/wm8960-soundcard.dtbo and /boot/overlays/wm8960-soundcard.dtbo differ

Restored BOOT partition is corrupt
*****

P33M, 2022-02-18:
The lack of resets means the usb-storage device was ultimately happy with the length of data that was sent, so one failure mode has been squashed. I wonder if there isn't some other combination of lengths (e.g. 4K followed by 512B and 512B) that causes a different type of miscounting to occur. A lack of protocol error is going to make this hard to debug as there's no obvious trigger point I can use to stop the controller and examine the state.

Can you try modifying the TRB_MAX_BUFF_SHIFT define here - https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/usb/host/xhci.h#L1526

Try 12 through 9 (4K through 512B) and retest. Smaller buffers will ultimately reduce performance for situations where the bug isn't hit, so modifying the global define won't be a fix.

P33M, 2022-02-18:
Do the errors happen quicker if you add the usb-storage quirk `m` ? This splits up writes into max 64 sector operations, so more SG operations per run.

P33M, 2022-02-18:
Aha, managed to get corruption after some hacking around. If I limit TRB buffers to 4k max and use a different ASmedia controller in mass-storage mode, then after 10 loops I usually see a few broken files.

In all the cases I've seen so far, the corruption is now at a 512B offset and the trashed section is a multiple of 512B in length. In one case, kernel8.img got corrupted which is a huge file. The segment that got trashed was 1K in length at 0x1c7200 and this appeared to be a repeat of data found at >6MB. If I examine the FAT for the file, it has a single extent covering the entire file. The file, offset and repeated section are far larger than any chip SRAM.

Can you upload your corrupt files (after a run with just the quirk applied - no other length mangling) for comparison. I can't rule out introducing a software bug by dramatically amplifying the number of TRBs queued at once to a ring.

reraikes, 2022-02-20:
Sorry for the delayed response.  I was in the process of posting this comment on Friday when my ISP experienced a massive area-wide outage that lasted more than 36 hours.

Here's a set of differing files after adding 'usb-storage.quirks=hhhh:hhhh:k' to /boot/cmdline.txt.  This addition appears to have had no effect on the problem.  The file corruption is still quite reproducible here using 5.10.95 and 5.15.23 with two different USB3 powered hubs, two different USB3 SATA-to-USB adapters, and two different USB3 SSD's (any combination).

Testing with a Sandisk Extreme 64GB USB3 flash drive on any USB3 powered hub, I also see corrupted files and still see resets (which I haven't seen with SSD's using 5.10.95/5.15.23):

[  384.742728] xxx (951): drop_caches: 3
[  396.798035] usb 2-1.4: reset SuperSpeed USB device number 4 using xhci_hcd
[  398.269484] xxx (951): drop_caches: 3

With the same Sandisk Extreme 64GB USB3 flash drive plugged directly into a Raspberry Pi 4 USB3 port (no hub), neither corrupted files nor resets occur after several hours of running the 'test-repeat' script.

[files.zip](https://github.com/raspberrypi/linux/files/8103502/files.zip)

P33M, 2022-02-22:
Your files appear to have sector-length segments trashed, and I see no instances of the granular mangling that was seen originally.

If I use the most terrible USB3 pendrive that I have then I get device resets with the latest rpi-update kernel. The random read speed is sub-1MB/s and sequential write is 30MB/s. Because of the slow flash, it frequently signals flow control on the bulk endpoints. It reliably fails the consistency check in the first few loops of the script.

The failure mode in this instance is insufficient data being transmitted - usually a multiple of 512B. The device never acknowledges completion of the CDB since it's still waiting for the total write data length, and eventually it gets reset because of a timeout.

If I disable the quirk and iterate over buffer segmentation thresholds, I get different types of corruption.

| TRB_BUFF_SHIFT | Failure mode |
| ---- | ---- |
| 7 (128B) | 768B sections of files are trashed, aligned to 128B offsets |
| 8 (256B) | 512B sections of files are trashed, aligned to 512B offsets |
| 9 (512B) | 1024B sections of files are trashed, aligned to 512B offsets | 
| 10 (1K) | Short transfers after 5-10 loops - device resets |
| 11 (2K) | Short transfers - devices resets |
| 11 (2K) with quirk | Short transfers - device resets |

If I repeat the test with 2 sectors per FAT cluster, I don't get corruption with 1K or 2K TRB sizes. If I reduce TRB sizes below 1K, I again get similar trashing.

If an endpoint is flow controlled, the USB3.0 spec requires that the controller "parks" it and ignores it until an EP Ready packet is sent from the device. From what I've seen of the controller behaviour, it looks like the rewind I was seeing on PCIe was a consequence of flow control - which should flush the internal FIFO, subtract the number of bytes in the FIFO from the PCIe fetch count, and update the transfer ring dequeue pointer to point to where the transfer should be resumed from.

It does seem like a different manifestation of the same bug - buffers of large non-integer multiples of wMaxPacket cause overflow, and TD buffer combinations of integer multiples of wMaxPacket with shorter lengths cause a miscalculation of the dequeue pointer/TRB residue.

reraikes, 2022-02-23:
https://github.com/raspberrypi/linux/issues/4844#issuecomment-1040053575

reraikes, 2022-02-25:
Since it appears the mount 'flush' option is being looked at as the solution to this issue, and since the addition of the 'flush' option causes such a major performance degradation to 'cp', I decided to try a different approach to saving and restoring the BOOT partition by using 'dd' instead.  'dd' doesn't suffer the same performance hit that 'cp' does.  I modified the 'test' script to use 'dd' as well as adding 'flush' to the /boot entry in fstab.  Much to my surprise, errors still show up in dmesg:

*****
[ 7947.724969] test-dd (963): drop_caches: 3
[ 7951.062729] usb 2-1.3: reset SuperSpeed USB device number 4 using xhci_hcd
[ 7951.094622] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
[ 7951.094648] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 36 63 6d 00 00 02 00 00
[ 7951.094657] blk_update_request: I/O error, dev sda, sector 912485632 op 0x0:(READ) flags 0x80700 phys_seg 5 prio class 0

[10665.283681] test-dd (5598): drop_caches: 3
[10667.114599] usb 2-1.3: reset SuperSpeed USB device number 4 using xhci_hcd
[10667.146613] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
[10667.146638] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 04 3b 18 00 01 00 00
[10667.146649] blk_update_request: I/O error, dev sda, sector 277272 op 0x0:(READ) flags 0x80700 phys_seg 3 prio class 0
*****

[test-dd.zip](https://github.com/raspberrypi/linux/files/8139617/test-dd.zip)

P33M, 2022-02-25:
DD is not a usual method by which files are written to disk. Flushing will break FAT filesystem write ops into single-file updates but dd'ing to the block device has no such restriction. It will copy blocks up to 4M in size as per your script.

reraikes, 2022-02-25:
The script is simply using 'dd' to save and restore the whole filesystem in (unmounted at the time) partition 1.  What's wrong with doing that?  And why would this cause the errors reported?  No corrupted data resulted in these instances, but why the dmesg errors?

P33M, 2022-02-25:
Where in my comment did I say it was the wrong tool to use? 

There is no filesystem corruption because the errors are for reads, not writes, and appear to be a different symptom. The sector numbers are odd - that first offset is over 460GB, the second over 14GB. How big is the FAT filesystem?

reraikes, 2022-02-25:
Sorry if I misinterpreted your comment.

The BOOT partition (partition 1) is a standard Raspberry Pi OS BOOT partition (256 MiB).

The SSD:
Disk /dev/sda: 447.13 GiB, 480103980544 bytes, 937703087 sectors
Disk model:     FA GoFlex De
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: A27E94CB-D567-4DC3-8489-8BFA49CE252D

pelwell, 2022-02-25:
> Since it appears the mount 'flush' option is being looked at as the solution to this issue,

I don't think that's the case - it's being viewed as a workaround while the real investigation continues, because people don't like data corruption (who knew?).

reraikes, 2022-02-25:
>I don't think that's the case - it's being viewed as a workaround while the real investigation continues, because people don't like data corruption (who knew?).

That's a relief to hear.  When I saw the change to pi-gen with no other comment, I feared that was the end of discussion.

pelwell, 2022-02-25:
You don't know @P33M very well.

xranby, 2022-02-28:
https://github.com/RPi-Distro/pi-gen/commit/1ad53be314dbc73d7117dbd610a982c90e115f37#diff-96afed30b2c1792627517acf0f5d507ee286dfac7d8c78ed3f595d7655c01f93R2 there is a typo in the commit
replace . with , to make the option have desired effect

pelwell, 2022-02-28:
Well spotted - we'll get that fixed.

pelwell, 2022-02-28:
See https://github.com/RPi-Distro/pi-gen/commit/a120c4128ece237754af9e8bef6679d6a51f73ec.

reraikes, 2022-03-03:
>Please post updates about the issue on the issue thread, not in commentary on a PR - it splits the discussion unnecessarily.

Sorry 'bout that.  I confess I didn't know the difference and wondered why the discussion moved elsewhere.

Another piece of info that may or may not be relevant...

While testing on the Corsair SSD again, I observed this in dmesg while running test-repeat:

*****
...
[  578.808731] test-repeat (754): drop_caches: 3
[  580.749600] test-repeat (754): drop_caches: 3
[  581.993803] usb 2-1.3: Disable of device-initiated U1 failed.
[  582.004427] usb 2-1.3: Disable of device-initiated U2 failed.
[  582.333689] usb 2-1.3: reset SuperSpeed USB device number 4 using xhci_hcd
[  584.594049] test-repeat (754): drop_caches: 3
[  586.524441] test-repeat (754): drop_caches: 3
...
*****

[24 more replies]