raspberrypi/linux #4847: Android Pixel 6 Pro unable to complete Bluetooth Connection Reason code 0x08 and Wrong link type (-22)

hando-gomes, 2022-01-25:
### Describe the bug

A few Android devices are unable to successfully establish a bluetooth connection to the Raspberry Pi 4.
The devices connect, and after 5 seconds, it is disconnected with a timeout reason code 0x08.

```
> HCI Event: Disconnect Complete (0x05) plen 4
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8
        LE Address: 54:FA:15:75:50:1B (Resolvable)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8
        LE Address: 54:FA:15:75:50:1B (Resolvable)
        Reason: Connection timeout (0x01)
```

The following Android devices have shown this behavior:
- Pixel Pro 6 - Android 12
- Samsung Galaxy S10 - Android 10
- Samsung Galaxy S21 Ultra

After connection, this message can be seen in the journal and dmesg:
`raspberrypi kernel: Bluetooth: Wrong link type (-22)`

The issue seems to be that during the discovery phase something goes wrong at the kernel level and the the device and raspi never completes the discovery phase. Bluez never receives the request for discovery and therefore the connection is closed due to timeout.

The log can be traced back to here: https://github.com/raspberrypi/linux/blob/2697f7403187bb2bb61cc716f33ee9f6cfb9af7c/net/bluetooth/l2cap_core.c#L6382

### Steps to reproduce the behaviour

- [Download the Bluez study guide for Linux](http://www.bluez.org/study-guide-for-linux-developer/) and transfer the code to the pi
- Install python3-dbus `sudo apt-get install python3-dbus`
- stop the bluetooth service `sudo systemctl stop bluetooth`
- Verify the path for the bluetoothd ExecPath: `$ grep Exec /lib/systemd/system/bluetooth.service`

- Run bluetooth service manually in debug and detached mode: `sudo /usr/libexec/bluetooth/bluetoothd -d -n`
- Run the sample code ` sudo python server_advertisement.py`
- Launch nRF Connect app on one of these Android devices:
  -  Pixel Pro 6 - Android 12, Samsung Galaxy S10 - Android 10, Samsung Galaxy S21 Ultra
- Scan and connect to raspi

The client will connect, start discovering services and after 5 seconds on inactivity the device will be disconnected.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2
```

```
$ vcgencmd version
Oct 29 2021 10:47:33
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
```

```
$ uname -a
Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```

```
[... 610 lines cut ...]
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service0023/char002a/desc002c
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char002e
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char002e/desc0030
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char002e/desc0031
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char0032
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char0032/desc0034
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char0032/desc0035
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char0036
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_EA_49_22_94_D5/service002d/char0036/desc0038
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:btd_gatt_client_ready() Features 0x00
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/gatt-client.c:btd_gatt_client_ready() Update Features 0x00
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_68_EA_49_22_94_D5 err 0
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/device.c:store_gatt_db() Can't store GATT db for private addressed device /org/bluez/hci0/dev_68_EA_49_22_94_D5
Jan 25 21:18:50 raspberrypi bluetoothd[951]: src/device.c:gatt_debug() Registered handler for "Service Changed": 1
Jan 25 21:18:51 raspberrypi bluetoothd[951]: profiles/gap/gas.c:read_device_name_cb() GAP Device Name: iPhone
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/device.c:btd_device_device_set_name() /org/bluez/hci0/dev_68_EA_49_22_94_D5 iPhone
Jan 25 21:18:51 raspberrypi bluetoothd[951]: profiles/gap/gas.c:read_appearance_cb() GAP Appearance: 0x0040
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/adapter.c:user_confirm_request_callback() hci0 68:EA:49:22:94:D5 confirm_hint 0
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/device.c:new_auth() Requesting agent authentication for 68:EA:49:22:94:D5
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/agent.c:agent_ref() 0x153fe68: ref=2
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.21, path=/org/bluez/agent, passkey=309570
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/device.c:gatt_debug() Read By Type - start: 0x0001 end: 0x0005
Jan 25 21:18:51 raspberrypi bluetoothd[951]: src/gatt-database.c:gap_device_name_read_cb() GAP Device Name read request
```

hando-gomes, 2022-01-25:
Running the newest Kernel, there's a new log message that helps narrow the issue:
```
Jan 25 18:59:15 bryte bluetoothd[23308]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 517
Jan 25 18:59:15 bryte bluetoothd[23308]: src/device.c:gatt_debug() (chan 0x1204940) ATT PDU received: 0x10
Jan 25 18:59:15 bryte bluetoothd[23308]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0001 end: 0xffff
Jan 25 18:59:15 bryte kernel: Bluetooth: Unknown LE signaling command 0x17
Jan 25 18:59:15 bryte kernel: Bluetooth: Wrong link type (-22)
Jan 25 18:59:15 bryte bluetoothd[23308]: src/device.c:gatt_debug() (chan 0x1204940) ATT PDU received: 0x08
Jan 25 18:59:15 bryte bluetoothd[23308]: src/device.c:gatt_debug() Read By Type - start: 0x0001 end: 0xffff
```

This commit refers to fixing the issue with the 0x17 which is the Enhanced Credit Based Flow Control
Mode. 
https://github.com/raspberrypi/linux/commit/69d67b461a180144ad1d31174fadf3e3eda78e56#diff-21cdf1a8784787f91c59b5750610c31d15712ee406b107c21951f81c1b48c188

pelwell, 2022-01-26:
Without changing the configuration, do other Android devices connect? If so, are they also attempting BLE connections, or is BLE the common factor in the failure cases?

pelwell, 2022-01-26:
Wait - I see you have successful Pixel 1 and iPhone 6 logs there, both of them using BLE, which answers both my questions.

The patch you link to is interesting, but it is already in the newest kernel and the problem clearly isn't fixed.

Time to expense a Pixel 6 Pro?

hando-gomes, 2022-01-26:
I was able to replicate this issue with these phones:

- Pixel Pro 6 - Android 12
- Samsung Galaxy S10 - Android 10
- Samsung Galaxy S21 Ultra - Android 12

These phones work without an issue:

- Google Pixel 5a Android 11
- Google Pixel 1
- iPhone 12 Pro iOS 15.1.1
- iPhone 12 Pro Max iOS 15.2.1
- iPhone 6s iOS 14.7.1
- iPhone 7 iOS 14.4 Vitaliy b03114 YES
- iPhone SE (1st gen) iOS 15.0
- iPhone X iOS 14.7.1
- iPhone XR iOS 14.8.1
- Samsung Galaxy S20 Ultra 5G Android 12
- Samsung Galaxy S21 Android 12
- Samsung Galaxy S21 Plus Android 12
- Samsung Galaxy S8 Android 9
- Samsung Galaxy S9 Android 10
- Samsung S21 (SM-G991B/DS) Android 11

pelwell, 2022-01-26:
Thanks for the confirmation. I was slightly joking with my previous comment re: expenses, but investigating this issue is going to depend on getting hold of one of the devices that shows the problem.

hando-gomes, 2022-01-26:
:) I hope you can find one of these devices.

Here is something else that could help to narrow this issue. 
The bluetooth parameter for ecred is by default disabled on the latest kernel.

Mod info:
```
$ modinfo bluetooth
filename:       /lib/modules/5.10.63-v7l+/kernel/net/bluetooth/bluetooth.ko
alias:          net-pf-31
license:        GPL
version:        2.22
description:    Bluetooth Core ver 2.22
author:         Marcel Holtmann <marcel@holtmann.org>
srcversion:     45D3D4C44F9D5986B018E4D
depends:        rfkill,ecdh_generic
intree:         Y
name:           bluetooth
vermagic:       5.10.63-v7l+ SMP mod_unload modversions ARMv7 p2v8
parm:           disable_esco:Disable eSCO connection creation (bool)
parm:           disable_ertm:Disable enhanced retransmission mode (bool)
parm:           enable_ecred:Enable enhanced credit flow control mode (bool) 
```

```
$ head /sys/module/bluetooth/parameters/*
==> /sys/module/bluetooth/parameters/disable_ertm <==
N

==> /sys/module/bluetooth/parameters/disable_esco <==
N

==> /sys/module/bluetooth/parameters/enable_ecred <==
N
```

Then by changing the content to `Y` for enable_ecred, some connections will actually succeed, some of the time, but not always.
`sudo nano /sys/module/bluetooth/parameters/enable_ecred` and replace the `N` with a `Y`
Do not reboot.

Here are some notable differences:
With the feature enable=N (the default value is disabled) the Pixel 6 Pro will always connect and then get disconnected. The journal will show this error during connection:
```Jan 26 16:46:06 raspberrypi bluetoothd[2376]: src/device.c:load_gatt_db() Restoring 6D:B1:54:80:B7:EF gatt database from file
Jan 26 16:46:06 raspberrypi bluetoothd[2376]: src/device.c:load_gatt_db() No cache for 6D:B1:54:80:B7:EF
Jan 26 16:46:06 raspberrypi bluetoothd[2376]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
Jan 26 16:46:06 raspberrypi kernel: Bluetooth: Wrong link type (-22)
Jan 26 16:46:06 raspberrypi bluetoothd[2376]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0001 end: 0xffff
Jan 26 16:46:07 raspberrypi bluetoothd[2376]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 517```

The btmon will display the command been rejected:
```> ACL Data RX: Handle 66 flags 0x02 dlen 26
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 18
        PSM: 39 (0x0027)
        MTU: 256
        MPS: 251
        Credits: 65535
        Source CID: 251
        Source CID: 65535
        Source CID: 94
        Source CID: 87
        Source CID: 88
        Source CID: 78
        Source CID: 75
< ACL Data TX: Handle 66 flags 0x00 dlen 10
      LE L2CAP: Command Reject (0x01) ident 2 len 2
        Reason: Command not understood (0x0000)
```

Changing the value to Y (enabling ecred) when the phone successfully connects, there are no warnings on the journal, and the btmon shows a valid RX and TX for the  Enhanced Credit Connection:

[... 402 lines cut ...]
< ACL Data TX: Handle 65 flags 0x00 dlen 9                                                                                                                #2994 [hci0] 2022-01-26 17:03:01.237981
      ATT: Error Response (0x01) len 4
        Read By Type Request (0x08)
        Handle: 0x0005
        Error: Attribute Not Found (0x0a)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                    #2995 [hci0] 2022-01-26 17:03:01.251886
        Num handles: 1
        Handle: 65
        Count: 2
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                    #2996 [hci0] 2022-01-26 17:03:01.259099
        Num handles: 1
        Handle: 65
        Count: 2
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                            #2997 [hci0] 2022-01-26 17:03:06.262317
        Status: Success (0x00)
        Handle: 65
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                      {0x0001} [hci0] 2022-01-26 17:03:06.262386
        LE Address: 61:DD:CB:12:94:BE (Resolvable)
        Reason: Connection timeout (0x01)
```

The client starts a `gatt.discoverServices()` but never get data back from the pi, and 5 seconds later it is disconnected.

Please let me know if there is anything I can do to help.

jbraswell, 2022-03-25:
We are also encountering the exact behavior described in this issue. Would love to see it fixed, and happy to help. Is there anything I can do to help move this along?

hando-gomes, 2022-03-28:
I just left a comment on the link above.

jbraswell, 2022-03-28:
@hando-gomes Sorry, not seeing a new comment on https://github.com/RPi-Distro/bluez-firmware/issues/9. Were you referring to another link?

Vudentz, 2022-04-20:
```
> ACL Data RX: Handle 65 flags 0x02 dlen 26                                                                                                               #2945 [hci0] 2022-01-26 17:03:00.636995
      LE L2CAP: Enhanced Credit Connection Request (0x17) ident 2 len 18
        PSM: 39 (0x0027)
        MTU: 256
        MPS: 251
        Credits: 65535
        Source CID: 251
        Source CID: 65535
        Source CID: 89
        Source CID: 90
        Source CID: 91
        Source CID: 92
        Source CID: 93
```

First this is against the spec since there is way too many Source CID:

```
BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 3, Part A
page 1063
Logical Link Control and Adaptation Protocol Specification

• Source CID – (2 to 10 octets)
The Source CID is an array of up to 5 two-octet values and represents the
channel endpoints on the device sending the request. Once a channel has
been created, data packets flowing to the sender of the request shall be sent to
```

It also makes no sense to limit the MTU to 256 for EATT when for ATT the MTU is 517.

cspensky, 2022-05-06:
Just to follow up.  My previous fix only seems to fix part of the issue.  After a few days, the phone seems to fail to connect again.  Has anyone had any progress on this?

ln-12, 2022-07-25:
I am experiencing the exact same issue. Using a Pixel 6 (non Pro, Android 13 Beta 4), I get the following messages running `sudo /usr/libexec/bluetooth/bluetoothd -d -n` alongside [cputemp](https://github.com/Douglas6/cputemp) when trying to connect from `nRF Connect`:

```
bluetoothd[1396]: src/adapter.c:connected_callback() hci0 device 4A:D4:07:B8:A6:99 connected eir_len 0
bluetoothd[1396]: src/device.c:device_create() dst 4A:D4:07:B8:A6:99
bluetoothd[1396]: src/device.c:device_new() address 4A:D4:07:B8:A6:99
bluetoothd[1396]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_4A_D4_07_B8_A6_99
bluetoothd[1396]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
bluetoothd[1396]: attrib/gattrib.c:g_attrib_ref() 0x6e0060: g_attrib_ref=1 
bluetoothd[1396]: src/device.c:load_gatt_db() Restoring 4A:D4:07:B8:A6:99 gatt database from file
bluetoothd[1396]: src/device.c:load_gatt_db() No cache for 4A:D4:07:B8:A6:99
bluetoothd[1396]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
bluetoothd[1396]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0001 end: 0xffff
bluetoothd[1396]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 517
bluetoothd[1396]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0013 end: 0xffff
bluetoothd[1396]: src/device.c:gatt_debug() Read By Grp Type - start: 0x001b end: 0xffff
bluetoothd[1396]: src/device.c:gatt_debug() Primary services found: 2
bluetoothd[1396]: src/device.c:gatt_debug() start: 0x0001, end: 0x0009, uuid: 00001801-0000-1000-8000-00805f9b34fb
bluetoothd[1396]: src/device.c:gatt_debug() start: 0x0014, end: 0xffff, uuid: 00001800-0000-1000-8000-00805f9b34fb
bluetoothd[1396]: src/device.c:gatt_debug() Read By Type - start: 0x0001 end: 0x0005
bluetoothd[1396]: src/device.c:gatt_debug() Read By Type - start: 0x0001 end: 0x0005
bluetoothd[1396]: src/device.c:gatt_debug() Server Features found: handle 0x0005 length 0x0001 value 0x01
bluetoothd[1396]: src/device.c:gatt_debug() Read By Type - start: 0x0005 end: 0x0005
bluetoothd[1396]: src/device.c:gatt_debug() Read By Type - start: 0x0006 end: 0x000f
bluetoothd[1396]: src/device.c:gatt_debug() Read By Type - start: 0x0006 end: 0x000f
bluetoothd[1396]: src/adapter.c:dev_disconnected() Device 4A:D4:07:B8:A6:99 disconnected, reason 1
bluetoothd[1396]: src/adapter.c:adapter_remove_connection() 
bluetoothd[1396]: plugins/policy.c:disconnect_cb() reason 1
bluetoothd[1396]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 4A:D4:07:B8:A6:99 type 2 status 0xe
bluetoothd[1396]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[1396]: src/device.c:device_bonding_failed() status 14
bluetoothd[1396]: src/adapter.c:resume_discovery() 
bluetoothd[1396]: src/device.c:gatt_debug() service disappeared: start 0x0001 end 0x0009
bluetoothd[1396]: src/device.c:gatt_debug() service disappeared: start 0x0014 end 0xffff
bluetoothd[1396]: src/device.c:gatt_debug() Failed to initialize gatt-client
bluetoothd[1396]: src/device.c:gatt_client_ready_cb() status: failed, error: 0
bluetoothd[1396]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_4A_D4_07_B8_A6_99 err -5
bluetoothd[1396]: src/device.c:att_disconnected_cb() 
bluetoothd[1396]: src/device.c:att_disconnected_cb() Connection timed out (110)
bluetoothd[1396]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
bluetoothd[1396]: src/device.c:att_disconnected_cb() Automatic connection disabled
bluetoothd[1396]: src/gatt-database.c:btd_gatt_database_att_disconnected() 
bluetoothd[1396]: attrib/gattrib.c:g_attrib_unref() 0x6e0060: g_attrib_unref=0 
bluetoothd[1396]: src/device.c:device_remove() Removing device /org/bluez/hci0/dev_4A_D4_07_B8_A6_99
bluetoothd[1396]: src/device.c:btd_device_unref() Freeing device /org/bluez/hci0/dev_4A_D4_07_B8_A6_99
bluetoothd[1396]: src/device.c:device_free() 0x6e1308
```

ln-12, 2022-07-28:
I finally got it working, but only with an external USB bluetooth dongle. I had a random one lying around, plugged it in, disabled the integrated bluetooth adapter using `sudo hciconfig hci0 down` and ran the [cputemp example](https://github.com/Douglas6/cputemp) again. My Pixel 6 connects immediately without any issue.

jbraswell, 2022-07-30:
> I finally got it working, but only with an external USB bluetooth dongle. I had a random one lying around, plugged it in, disabled the integrated bluetooth adapter using `sudo hciconfig hci0 down` [...]

Which dongle did you use?

ln-12, 2022-08-01:
> Which dongle did you use?

It's a TP-Link UB400, but I would guess that other dongles would also work.

Path-A, 2022-08-08:
Another workaround on the Pixel 6 with Android 12 is to enable Gabeldorsche Bluetooth in the phone's developer settings. Unfortunately, it seems that Gabeldorsche is going away with Android 13 from what I can tell, so this may not work for very long.

zhonghuiwen, 2022-09-04:
> It's a TP-Link UB400, but I would guess that other dongles would also work.

Thank you so much @ln-12, i just bought the TP-Link UB400 you mentioned and i am able to connect my OPPO1920 to the pi

Path-A, 2022-10-05:
Would like to confirm that some newer phones on latest release of Raspbian (Linux 5.15 and BlueZ 5.55) are still not working. The following phones failed to connect via Bluetooth:

Pixel 6, Android 12
Pixel 6, Android 13
Pixel 6 Pro, Android 13
Samsung Ultra 22, Android 12
Samsung Ultra 22, Android 13

It appears that the new Gabeldorsche Bluetooth stack was widely adopted in Android 13, but it didn't fix the issue. Perhaps the stack wasn't entirely adopted yet because it previously allowed my Pixels to connect when I enabled Gabeldorsche in Developer settings.

cspensky, 2022-10-05:
I am still having issues on my pixel 6 with android 13. Would love some
insight on a long-term fix.

pelwell, 2022-10-05:
Locking thread to stop the cross-posting. Please use https://github.com/RPi-Distro/bluez-firmware/issues/9.