raspberrypi/linux #4848: please fix ahci driver when in pcie-32bit-dma mode

FrancescoRestelli, 2022-01-25:
### Describe the bug

when setting dtoverlay=pcie-32bit-dma (which is necessary for some dvb cards i am playing with) the ahci driver does not work and crashes

### Steps to reproduce the behaviour

install ASM1064 based sata card with a connected ssd
set dtoverlay=pcie-32bit-dma in /boot/config.txt
reboot

dmesg shows errors
lspci crashes

no /dev/sda* exists

### Device (s)

Raspberry Pi CM4

### System

os:
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage4

firmware:
Jan 20 2022 13:56:48
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

uname:
Linux raspberrypi 5.10.92-v7l+ #1 SMP Tue Jan 25 18:46:39 GMT 2022 armv7l GNU/Linux

### Logs

[   12.771119] ata1.00: qc timeout (cmd 0xec)
[   12.911545] 8<--- cut here ---
[   12.914604] Unhandled fault: asynchronous external abort (0x1211) at 0x00000000
[   12.921915] pgd = 7ed806c6
[   12.924617] [00000000] *pgd=042cc003, *pmd=1fd26d003
[   12.976726] 8<--- cut here ---
[   12.979781] Unhandled fault: asynchronous external abort (0x1211) at 0x00000000
[   12.987083] pgd = bc4626ff
[   12.989783] [00000000] *pgd=80000000004003, *pmd=00000000
[   12.995185] Internal error: : 1211 [#1] SMP ARM
[   12.999710] Modules linked in: snd_soc_hdmi_codec cfg80211 rfkill 8021q garp stp llc v3d gpu_sched i2c_brcmstb raspberrypi_hwmon v4l2_mem2mem(O) bcm2835_mmal_vchiq(C) videobuf2_dma_contig(O) videobuf2_vmalloc(O) videobuf2_memops(O) videobuf2_v4l2(O) videobuf2_common(O) snd_bcm2835(C) videodev(O) mc(O) vc_sm_cma(C) vc4 ahci libahci cec drm_kms_helper libata snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[   13.051725] CPU: 1 PID: 200 Comm: scsi_eh_0 Tainted: G         C O      5.10.92-v7l+ #1
[   13.059721] Hardware name: BCM2711
[   13.063159] PC is at ata_wait_register+0xb0/0x134 [libata]
[   13.068651] LR is at _cond_resched+0x34/0x6c
[   13.072915] pc : [<bf2443ac>]    lr : [<c0bc404c>]    psr: 60000013
[   13.079175] sp : c37fdb70  ip : c37fdb28  fp : c37fdbb4
[   13.084393] r10: c377e000  r9 : c37fc000  r8 : 00008000
[   13.089619] r7 : 00000001  r6 : c1203d00  r5 : 00000001  r4 : c377c000
[   13.096149] r3 : f0868118  r2 : deaddead  r1 : 00000000  r0 : 00000000
[   13.102681] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   13.109828] Control: 30c5383d  Table: 04655e40  DAC: 55555555
[   13.109835] Process scsi_eh_0 (pid: 200, stack limit = 0xe537caac)
[   13.109838] Stack: (0xc37fdb70 to 0xc37fe000)
[   13.109843] db60:                                     00008001 0000041a 000003e8 00008000
[   13.109850] db80: f0868118 ffff9008 c377e000 00000001 000001f4 c377c000 c377e000 c377e000
[   13.142442] dba0: deaddead c377d780 c37fdbdc c37fdbb8 bf1953dc bf244308 00000001 000001f4
[   13.142447] dbc0: c377c000 60000113 f0868100 c237f240 c37fdc0c c37fdbe0 bf194ca8 bf195348
[   13.142451] dbe0: 60000113 60000113 c377d000 60000113 c377c000 c377dac0 c377d6b0 00000001
[   13.142455] dc00: c37fdc1c c37fdc10 bf194d5c bf194c6c c37fdc94 c37fdc20 bf246c7c bf194d40
[   13.142460] dc20: c37fdc64 000000ec c377d770 00000000 00000000 00000000 00000000 fafbfcfd
[   13.171252] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   13.175126] dc40: c37fdce8 60000113 c377d000 c37fdcd0 bf19694c 00000001 00030003 c37fdc5c
[   13.183340] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[... 105 lines cut ...]
                LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis+
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [130 v1] Secondary PCI Express
                LnkCtl3: LnkEquIntrruptEn- PerformEqu-
                LaneErrStat: 0
        Kernel driver in use: ahci
        Kernel modules: ahci

### Additional context

also @6by9  noticed the bug on his hardware

pelwell, 2022-01-26:
Does running in 64-bit mode (`arm_64bit=1` in config.txt) make any difference?

FrancescoRestelli, 2022-01-26:
well it crashes harder (as in i loose ssh and had to uart to get the logs)

uname -ar
Linux raspberrypi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

[   20.748085] SError Interrupt on CPU0, code 0xbf000002 -- SError
[   20.748087] CPU: 0 PID: 212 Comm: scsi_eh_0 Tainted: G         C        5.10.63-v8+ #1459
[   20.748088] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   20.748090] pstate: 80000085 (Nzcv daIf -PAN -UAO -TCO BTYPE=--)
[   20.748091] pc : el1_irq+0x84/0x180
[   20.748092] lr : ahci_post_internal_cmd+0x28/0x38 [libahci]
[   20.748094] sp : ffffffc01220b780
[   20.748095] x29: ffffffc01220b8b0 x28: ffffff81062c8000
[   20.748100] x27: 0000000000000200 x26: 0000000000000001
[   20.748103] x25: ffffff8103efe040 x24: 0000000000000000
[   20.748106] x23: 0000000080000005 x22: ffffffc008e1eeb8
[   20.748109] x21: ffffffc01220b8d0 x20: 0000007fffffffff
[   20.748112] x19: ffffffc011b8e100 x18: 0000000000000020
[   20.748115] x17: 0000000000000000 x16: 0000000000000000
[   20.748118] x15: ffffffffffffffff x14: ffffffc011288948
[   20.748121] x13: ffffffc011471c10 x12: ffffffc0113154b8
[   20.748124] x11: 0000000000000003 x10: ffffffc0112fd478
[   20.748127] x9 : ffffffc008e1ef98 x8 : 0000000000017fe8
[   20.748130] x7 : c0000000ffffefff x6 : ffffffc01220b880
[   20.748133] x5 : 0000000000000000 x4 : ffffff81062c8000
[   20.748135] x3 : 0000000000000001 x2 : 0000000000000000
[   20.748139] x1 : 0000000000000100 x0 : ffffff8103efc000
[   20.748142] Kernel panic - not syncing: Asynchronous SError Interrupt
[   20.748144] CPU: 0 PID: 212 Comm: scsi_eh_0 Tainted: G         C        5.10.63-v8+ #1459
[   20.748145] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   20.748147] Call trace:
[   20.748148]  dump_backtrace+0x0/0x1b8
[   20.748149]  show_stack+0x20/0x30
[   20.748150]  dump_stack+0xf0/0x158
[   20.748151]  panic+0x18c/0x38c
[   20.748153]  nmi_panic+0x6c/0xa0
[   20.748154]  arm64_serror_panic+0x7c/0x90
[   20.748155]  do_serror+0x38/0x98
[   20.748156]  el1_error+0x90/0x110
[   20.748157]  el1_irq+0x84/0x180
[   20.748159]  ahci_kick_engine+0x48/0x100 [libahci]
[   20.748160]  ahci_post_internal_cmd+0x28/0x38 [libahci]
[   20.748161]  ata_exec_internal_sg+0x26c/0x450 [libata]
[   20.748163]  ata_exec_internal+0x84/0xc8 [libata]
[   20.748164]  ata_dev_read_id+0x27c/0x578 [libata]
[   20.748165]  ata_eh_recover+0x6d4/0x1248 [libata]
[   20.748167]  sata_pmp_error_handler+0x514/0x978 [libata]
[   20.748168]  ahci_error_handler+0x48/0x90 [libahci]
[   20.748169]  ata_scsi_port_error_handler+0x240/0x600 [libata]
[   20.748171]  ata_scsi_error+0xa0/0xe0 [libata]
[   20.748172]  scsi_error_handler+0xa8/0x388
[   20.748173]  kthread+0x148/0x158
[   20.748174]  ret_from_fork+0x10/0x30
[   20.748191] SMP: stopping secondary CPUs
[   20.748192] Kernel Offset: disabled
[   20.748194] CPU features: 0x0240022,61002000
[   20.748195] Memory Limit: none

6by9, 2022-01-26:
Does it work without the PCIe switch in the way but ```dtoverlay=pcie-32bit-dma```?

I think I have a spare SATA card at home (should be a https://www.amazon.co.uk/Ziyituod-Controller-Expansion-Profile-Non-Raid/dp/B07TD57RGZ), so I can bring it in if it helps.

FrancescoRestelli, 2022-01-26:
nope tried with arm_64bit=1 and without same results once i set dtoverlay=pcie-32bit-dma

crash from arm64 without PCIe switch in the middle :
[    1.432005] hub 2-0:1.0: config failed, hub doesn't have any ports! (err -19)
[   10.252211] SError Interrupt on CPU1, code 0xbf000002 -- SError
[   10.252215] CPU: 1 PID: 224 Comm: scsi_eh_0 Tainted: G         C        5.10.                                       63-v8+ #1459
[   10.252217] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   10.252218] pstate: 80000085 (Nzcv daIf -PAN -UAO -TCO BTYPE=--)
[   10.252220] pc : el1_irq+0x84/0x180
[   10.252221] lr : ahci_post_internal_cmd+0x28/0x38 [libahci]
[   10.252223] sp : ffffffc0120eb780
[   10.252224] x29: ffffffc0120eb8b0 x28: ffffff81061ddac0
[   10.252230] x27: 0000000000000200 x26: 0000000000000001
[   10.252234] x25: ffffff8104356040 x24: 0000000000000000
[   10.252238] x23: 0000000080000005 x22: ffffffc008d3eeb8
[   10.252242] x21: ffffffc0120eb8d0 x20: 0000007fffffffff
[   10.252245] x19: ffffffc011c66100 x18: 0000000000000020
[   10.252248] x17: 0000000000000000 x16: 0000000000000000
[   10.252251] x15: ffffffffffffffff x14: ffffffc011288948
[   10.252255] x13: ffffffc011471c10 x12: ffffffc0113154b8
[   10.252258] x11: 0000000000000003 x10: ffffffc0112fd478
[   10.252261] x9 : ffffffc008d3ef98 x8 : 0000000000017fe8
[   10.252264] x7 : c0000000ffffefff x6 : ffffffc0120eb880
[   10.252267] x5 : 0000000000000000 x4 : ffffff81061ddac0
[   10.252270] x3 : 0000000000000001 x2 : 0000000000000000
[   10.252273] x1 : 0000000000000100 x0 : ffffff8104354000
[   10.252277] Kernel panic - not syncing: Asynchronous SError Interrupt
[   10.252278] CPU: 1 PID: 224 Comm: scsi_eh_0 Tainted: G         C        5.10.                                       63-v8+ #1459
[   10.252280] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   10.252281] Call trace:
[   10.252283]  dump_backtrace+0x0/0x1b8
[   10.252284]  show_stack+0x20/0x30
[   10.252285]  dump_stack+0xf0/0x158
[   10.252286]  panic+0x18c/0x38c
[   10.252287]  nmi_panic+0x6c/0xa0
[   10.252289]  arm64_serror_panic+0x7c/0x90
[   10.252290]  do_serror+0x38/0x98
[   10.252291]  el1_error+0x90/0x110
[   10.252292]  el1_irq+0x84/0x180
[   10.252293]  ahci_kick_engine+0x48/0x100 [libahci]
[   10.252295]  ahci_post_internal_cmd+0x28/0x38 [libahci]
[   10.252296]  ata_exec_internal_sg+0x26c/0x450 [libata]
[   10.252298]  ata_exec_internal+0x84/0xc8 [libata]
[   10.252299]  ata_dev_read_id+0x27c/0x578 [libata]
[   10.252300]  ata_eh_recover+0x6d4/0x1248 [libata]
[   10.252302]  sata_pmp_error_handler+0x514/0x978 [libata]
[   10.252303]  ahci_error_handler+0x48/0x90 [libahci]
[   10.252305]  ata_scsi_port_error_handler+0x240/0x600 [libata]
[   10.252306]  ata_scsi_error+0xa0/0xe0 [libata]
[   10.252307]  scsi_error_handler+0xa8/0x388
[   10.252308]  kthread+0x148/0x158
[   10.252309]  ret_from_fork+0x10/0x30
[   10.252326] SMP: stopping secondary CPUs
[   10.252327] Kernel Offset: disabled
[   10.252329] CPU features: 0x0240022,61002000
[   10.252330] Memory Limit: none

6by9, 2022-01-26:
> nope tried with arm_64bit=1 and without same results once i set dtoverlay=pcie-32bit-dma
>
> crash from arm64 without PCIe switch in the middle :

In some ways this is good, as it simplifies the configuration.

FrancescoRestelli, 2022-01-26:
i forgot to add this is the card i am using 
https://www.amazon.de/gp/product/B08NXTF7BM/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1
and my latest test are on a fresh vanilla flash of 5.10.63

6by9, 2022-01-27:
Testing in the office and it seems happy, at least with no drives connected.
I am on 5.15.15, but I'd hope that wouldn't make a significant difference.

FWIW my card reports as
```
01:00.0 SATA controller: Marvell Technology Group Ltd. Device 9215 (rev 11) (prog-if 01 [AHCI 1.0])
	Subsystem: Marvell Technology Group Ltd. Device 9215
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 84
	Region 0: I/O ports at 0000
	Region 1: I/O ports at 0000
	Region 2: I/O ports at 0000
	Region 3: I/O ports at 0000
	Region 4: I/O ports at 0000
	Region 5: Memory at 600040000 (32-bit, non-prefetchable) [size=2K]
	Expansion ROM at 600000000 [size=256K]
	Capabilities: <access denied>
	Kernel driver in use: ahci
	Kernel modules: ahci
```
There is an error message in dmesg
```
[    6.636876] ahci 0000:01:00.0: version 3.0
[    6.637324] ahci 0000:01:00.0: arch assigned 64-bit MSI address 0xffffffffc but device only supports 32 bits
[    6.642158] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    6.647690] ahci 0000:01:00.0: AHCI 0001.0000 32 slots 4 ports 6 Gbps 0xf impl SATA mode
[    6.647725] ahci 0000:01:00.0: flags: 64bit ncq sntf led only pmp fbs pio slum part sxs 
```

I do have a suspicion I may have had msi disabled (```pci=nomsi``` in cmdline.txt) when I last tested, which may well explain it. I was debugging a load of probing issues with numerous cards under https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/57 at the time.
Please check that you haven't done the same.

FrancescoRestelli, 2022-01-27:
correct with no drive connected the error does not present itself...

i do not have pci=nomsi in my cmdline.txt

6by9, 2022-01-27:
OK, I'll take a drive into the office tomorrow.

6by9, 2022-02-01:
I forgot to pick up a drive until today.
Tested and largely seems OK without the PCIe switch using a 5.15.18 kernel. I've mounted my SSD and done some basic reads off it.

Doing a comparison with and without dtoverlay=pcie-32bit-dma.
Without it I get an MSI address assigned of 0xffff fffc at https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/pci/msi.c#L516. 
With it I get 0xf ffff fffc.

There seems to be something funky going on, and ```__pci_write_msi_msg``` stores away the last message that was written.
And it seems to be down to pcie-brcmtb
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/pci/controller/pcie-brcmstb.c#L149
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/pci/controller/pcie-brcmstb.c#L947
```
	/*
	 * We ideally want the MSI target address to be located in the 32bit
	 * addressable memory area. Some devices might depend on it. This is
	 * possible either when the inbound window is located above the lower
	 * 4GB or when the inbound area is smaller than 4GB (taking into
	 * account the rounding-up we're forced to perform).
	 */
	if (rc_bar2_offset >= SZ_4G || (rc_bar2_size + rc_bar2_offset) < SZ_4G)
		pcie->msi_target_addr = BRCM_MSI_TARGET_ADDR_LT_4GB;
	else
		pcie->msi_target_addr = BRCM_MSI_TARGET_ADDR_GT_4GB;
```
I would chase down the window definition, but it looks like it's being done via some magic, and I'm out of time for today, but I think we want to define it as less than 4GB in this case.

6by9, 2022-02-02:
arm_64bit=1 issue appears to be different to 32bit. It crashes during brcm_pcie_probe, with callstack
```
el1_error
pci_generic_config_read
pci_bus_read_config_dword
pcie_capability_read_dword
pcie_aspm_init_link_state
pci_scan_slot
pci_scan_child_bus_extend
pci_scan_bridge_extend
pci_scan_child_bus_extend
pci_scan_root_bus_bridge
pci_host_probe
```
And that is with or without a drive connected.

32bit issue does seem to be down to that MSI address. With pci-32bit-dma we have an offset of 0 and size of 0x100000000 (4GB). Without we have an offset of 0x400000000 (16GB) and size of 0x100000000 (4GB). I can't immediately deduce where these window values come from.
@pelwell could you advise?

pelwell, 2022-02-02:
The base address and window size are provided by the firmware (via the `dma-ranges` property) based on the RAM size and SoC stepping (the PCIe interface on 2711B0 couldn't address the full 4GB range, only the first 3GB). The effect of the `pcie-32bit-dma` overlay is to disable the rewriting by the firmware, leaving you with the values in the dtb, which specify an inbound window of 3GB at offset 0, leaving space for the MSI vector at 0x0_fffffffc.

The first thing to try would be to set `total_mem=3072` in config.txt and see if that changes anything.

6by9, 2022-02-02:
With ```pcie-32bit-dma``` the firmware can't fix anything up, so ```total_mem=3072``` will do nothing. Or am I missing something?

Issue found anyway.
rc_bar2 is computed at https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/pci/controller/pcie-brcmstb.c#L819
```
	/* System memory starts at this address in PCIe-space */
	*rc_bar2_offset = lowest_pcie_addr;
	/* The sum of all memc views must also be a power of 2 */
	*rc_bar2_size = 1ULL << fls64(size - 1);
```
The 3GB size was therefore being rounded up to 4GB, and the MSI vector couldn't fit in at 0x0_fffffffc.

Drop the dma-range to 2GB with 
```
--- a/arch/arm/boot/dts/bcm2711.dtsi
+++ b/arch/arm/boot/dts/bcm2711.dtsi
@@ -542,7 +542,7 @@
                         * memory.
                         */
                        dma-ranges = <0x02000000 0x0 0x00000000 0x0 0x00000000
-                                     0x0 0xc0000000>;
+                                     0x0 0x80000000>;
                        brcm,enable-ssc;
                };
 ```
and the MSI vector slots in where expected, and the warning goes away.

64 bit also then seems to be happy to boot as well.

Is that a reasonable change to make to the base DT? I don't know the full implications.

pelwell, 2022-02-02:
Rather than patch the base dts file I would prefer that change to go into the pcie-32bit-dma overlay, making it live up to the name.

6by9, 2022-02-03:
pcie-32bit-dma patched, and PR #4870 created.
It makes my SATA card happy when not through a switch. I don't have a switch here to test with, but can try tonight.

@FrancescoRestelli any chance you could test with your setup?

FrancescoRestelli, 2022-02-03:
i can confirm that my ssd drive is now mounted and working with dtoverlay=pcie-32bit-dma enabled

with and without the pcie switch!

since the PR was already merged i just compiled current upstream version and a 32bit kernel  5.10.95-v7l+

thanks a lot to both of you

will test more over the weekend report and then closed the issue

br francesco

closed 2022-02-07 by FrancescoRestelli