raspberrypi/linux #4857: CM4s: vc4-kms-v3d fails if hdmi1 is missing

nbuchwitz, 2022-01-30:
### Describe the bug

On compute module 4s the kms v3d driver overlay fails. Could this caused by the deletation in the device tree (https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/bcm2711-rpi-cm4s.dts#L209)?

```
005050.304: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
005064.801: dterror: can't find symbol 'hdmi1'
005064.811: Failed to resolve overlay 'vc4-kms-v3d'
```

### Steps to reproduce the behaviour

1.  Edit /boot/config.txt and enable `dtoverlay=vc4-kms-v3d`
2.  Reboot
3. Run vcdbg log msg

### Device (s)

Raspberry Pi CM4

### System

**os**
```
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4
```

**firmware**
```
Dec  1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)
```

**kernel**
```
Linux RevPi6351 5.10.92-rt60-v7l+ #8 SMP PREEMPT_RT Mon, 24 Jan 2022 15:51:23 +0000 armv7l GNU/Linux
```

### Logs

```
003798.674: arasan: arasan_emmc_open
004276.158: brfs: File read: /mfs/sd/config.txt
004276.988: brfs: File read: 1938 bytes
004282.491: brfs: File read: /mfs/sd/dt-blob.bin
004286.628: gpioman: configuration attempt 1 failed (error 1) - bad dt-blob.bin?
004314.535: gpioman: configuration attempt 2 succeeded - using built-in dt-blob
004379.896: HDMI1:EDID error reading EDID block 0 attempt 0
004385.921: HDMI1:EDID error reading EDID block 0 attempt 1
004391.943: HDMI1:EDID error reading EDID block 0 attempt 2
004397.967: HDMI1:EDID error reading EDID block 0 attempt 3
004403.990: HDMI1:EDID error reading EDID block 0 attempt 4
004410.014: HDMI1:EDID error reading EDID block 0 attempt 5
004416.036: HDMI1:EDID error reading EDID block 0 attempt 6
004422.060: HDMI1:EDID error reading EDID block 0 attempt 7
004428.083: HDMI1:EDID error reading EDID block 0 attempt 8
004434.107: HDMI1:EDID error reading EDID block 0 attempt 9
004435.120: HDMI1:EDID giving up on reading EDID block 0
004435.201: brfs: File read: 43162 bytes
004437.901: brfs: File read: /mfs/sd/config.txt
004438.706: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
004438.723: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
004448.237: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
004448.252: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
004448.281: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
004448.298: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
004737.972: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
004738.016: *** Restart logging
004738.036: brfs: File read: 1938 bytes
004803.933: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004809.959: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
004815.985: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
004822.012: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
004828.035: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
004838.239: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
004846.851: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
004852.905: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
004858.955: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
004865.007: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
004866.041: hdmi: HDMI1:EDID giving up on reading EDID block 0
004871.135: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004877.187: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
004883.250: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
004889.303: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
004895.352: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
004901.405: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
004907.455: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
004913.507: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
004919.557: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
004925.609: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
004926.642: hdmi: HDMI1:EDID giving up on reading EDID block 0
004934.372: HDMI0: hdmi_pixel_encoding: 300000000
004934.387: HDMI1: hdmi_pixel_encoding: 300000000
004935.204: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
004935.225: dtb_file 'bcm2711-rpi-cm4s.dtb'
004942.015: brfs: File read: /mfs/sd/bcm2711-rpi-cm4s.dtb
004942.033: Loading 'bcm2711-rpi-cm4s.dtb' to 0x100 size 0xbd8e
004954.556: brfs: File read: 48526 bytes
004965.798: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
005028.137: brfs: File read: 1703 bytes
005029.734: brfs: File read: /mfs/sd/config.txt
005030.210: dtparam: audio=on
005040.242: brfs: File read: 1938 bytes
005050.304: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
005064.801: dterror: can't find symbol 'hdmi1'
005064.811: Failed to resolve overlay 'vc4-kms-v3d'
005065.277: brfs: File read: 3823 bytes
005067.626: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
005073.511: Loaded overlay 'dwc2'
005073.523: dtparam: dr_mode=host
005083.516: brfs: File read: 801 bytes
005088.486: brfs: File read: /mfs/sd/overlays/revpi-core.dtbo
005107.724: Loaded overlay 'revpi-core'
005107.752: dtparam: eth0_mac_hi=0xc83ea700
005108.964: dtparam: eth0_mac_lo=0x2ae5
005366.634: brfs: File read: 3987 bytes
005368.376: brfs: File read: /mfs/sd/cmdline.txt
005368.453: Read command line from file 'cmdline.txt':
005368.481: 'dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait nosplash plymouth.ignore-serial-consoles'
005618.019: brfs: File read: 154 bytes
007061.453: brfs: File read: /mfs/sd/kernel7l.img
007061.489: Loading 'kernel7l.img' to 0x8000 size 0x6af9f0
007061.533: Device tree loaded to 0x2eff3600 (size 0xc933)
007069.240: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
009235.066: vchiq_core: vchiq_init_state: slot_zero = 0xead80000, is_master = 1
009237.131: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
009237.209: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
009243.169: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000004
009243.194: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000004
```

nbuchwitz, 2022-01-30:
I tried with the same kernel with an modified cm4s overlay, where I removed the `/delete-node/ &hdmi1`. With this overlay the driver loads as expected:

```
005081.790: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
005145.329: Loaded overlay 'vc4-kms-v3d'
```

nbuchwitz, 2022-01-30:
Looks like the problem is here https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/vc4-kms-v3d-pi4-overlay.dts#L40

If I remove the `hdmi1` reference from this overlay, the driver loads with the cm4s overlay. BTW: Is there any reason why `ddc1` is not removed from the tree aswell?

peterharperuk, 2022-01-30:
Thanks. The CM4S dtb was generated from the CM4 dtb. So ddc1 might just be something that we forgot to remove

nbuchwitz, 2022-01-30:
> Thanks. The CM4S dtb was generated from the CM4 dtb. So ddc1 might just be something that we forgot to remove

OK, I'll send a PR for the cm4s with ddc1 removed

pelwell, 2022-02-02:
It's a minor point, but it is the reference to the undefined symbol that is completely fatal - the whole overlay is skipped - rather than the failure to apply a fragment. Fragment application stops at the first failure - whether this is ideal behaviour is open to debate.

pelwell, 2022-02-02:
Feedback on the PR: https://github.com/raspberrypi/linux/pull/4863#issuecomment-1027954019

nbuchwitz, 2022-02-02:
Imho is having a separate overlay (`vc4-kms-cm4s`?) for this the only reasonable way to do. Unless there is any hidden DSL magic which allows for a reference check before the fragment is applied...

6by9, 2022-02-02:
It depends on exactly how things have been broken out on the board.

hdmi1 and ddc1 physically exists in the SoC so they can be enabled (which the existing vc4-kms-v3d-pi4 overlay will do). As long as the hot plug detect line has been wired low then it'll be enabled but just stay idle as there is no display connected.

nbuchwitz, 2022-02-02:
We haven't received schematics yet, so I can't check. Maybe @pelwell can clarify this internally.

pelwell, 2022-02-02:
The schematic shows, and the CM4S designer confirms, that the hotplug signal for HDMI1 is floating. Hopes of a hidden internal pull resistor, seem to be just that - hopes. It would therefore be wise not to enable the hotplug interrupt or the CEC I2C.

Someone here is going to have to come up with the least horrible solution for CM4S, and I think I know who that someone is.

pelwell, 2022-02-02:
See https://github.com/raspberrypi/linux/pull/4869 for a possible workaround.

nbuchwitz, 2022-02-03:
Fixed with  #4869

closed 2022-02-03 by nbuchwitz

pelwell, 2022-06-13:
My fix was merged in February, and the `stable` branch of the `rpi-firmware` repo was bumped on April 1st, including `bcm2711-rpi-cm4s.dtb`. As the latest 64-bit image was released on April 4th, the fix should already be available,

pelwell, 2022-06-13:
Ah yes, that's really an orthogonal issue, but it should be sorted now that CM4S (and Zero 2 W) can be built in the arm64 tree - see https://github.com/raspberrypi/linux/commit/fca0308be9e2d516cb38a1686050e931a740ebe2.