raspberrypi/linux #4858: CM4s: hdmi1 is probed for EDID even when removed from devicetree

nbuchwitz, 2022-01-30:
### Describe the bug

With the `vc4-fkms-v3d` overlay enabled, there is a error message regarding the EDID detection of the (disabled) hdmi1 port:

```
[   11.275757] vc4-drm gpu: bound fe600000.firmwarekms (ops vc4_fkms_ops [vc4])
[   11.276605] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 0
[   11.392165] vc4-drm gpu: [drm] HDMI-A-2: EDID is invalid:
[   11.392177]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392180]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392183]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392186]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392188]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392191]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392193]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.392196]  [00] ZERO 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[   11.663165] rpivid-mem feb00000.hevc-decoder: rpivid-hevcmem initialised: Registers at 0xfeb00000 length 0x00010000
[   11.663430] rpivid-mem feb10000.rpivid-local-intc: rpivid-intcmem initialised: Registers at 0xfeb10000 length 0x00001000
[   11.663656] rpivid-mem feb20000.h264-decoder: rpivid-h264mem initialised: Registers at 0xfeb20000 length 0x00010000
[   11.663889] rpivid-mem feb30000.vp9-decoder: rpivid-vp9mem initialised: Registers at 0xfeb30000 length 0x00010000
[   11.686466] Console: switching to colour frame buffer device 80x30
[   11.703843] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[   12.089585] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 1
```

The port hdmi1 is removed in the CM4s overlay: https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/bcm2711-rpi-cm4s.dts#L209 and therefore should be ignored by the driver.

Also after the activation of the fkms driver the active part of the screen is shrinked to only a fraction of the usable screen:

[image]

### Steps to reproduce the behaviour

1.   Edit /boot/config.txt and enable dtoverlay=vc4-fkms-v3d
2.   Reboot
3.   (Run dmesg)

### Device (s)

Raspberry Pi CM4

### System

**os**
```
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4
```

**firmware**
```
Dec  1 2021 15:01:54
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)
```

**kernel**
```
Linux RevPi6351 5.10.92-rt60-v7l+ #8 SMP PREEMPT_RT Mon, 24 Jan 2022 15:51:23 +0000 armv7l GNU/Linux
```

peterharperuk, 2022-02-02:
This might be a firmware issue. We disable HDMI1 for BCM2838 - we should be doing it for CM4S (2711) as well.

pelwell, 2022-02-03:
The firmware has a function returning the number of HDMI ports, which returns 2 on BCM2711 and 1 otherwise. The number of HDMI ports should probably be an integer trait.

pelwell, 2022-02-03:
I'll make it so.

closed 2025-03-21 by nbuchwitz