raspberrypi/linux #4872: USB/Ethernet gadget on PiOS-5.15.16: kills xHCI host controller

macmpi, 2022-02-04:
### Describe the bug

Ethernet USB gadget mode works as expected on stock PiOS lite on PiZeroW 5.10.92, but fails with 5.15.16 kernel.
Host PC xHCI host controller dies with no other remedies than reboot (x64 Mint/Ubuntu).
Can't get USB/Ethernet gadget operation anymore.

### Steps to reproduce the behaviour

On PiZeroW, I flashed 2022-01-28-raspios-bullseye-armhf-lite, set `dtoverlay=dwc2` in `config.txt`, added `modules-load=dwc2,g_ether`  in `cmdline.txt`, connected the Pi to my Ubuntu PC USB port, all worked as expected.
I could ssh into the Pi from Host PC, get to Pi to internet and do further setup: all good.
Pi kernel was at stock 5.10.92+

I then updated Pi to latest 5.15.16 kernel:
`sudo BRANCH=next rpi-update`
Update went just fine. I restarted pi, and... watched xHCI host controller dying through `dmesg`: Pi would obviously never come-up again on usb interface (and no other USB device can be used either until host reboot).

(in fact I initially had the same problem with Alpine 3.15, and it similarly killed my Hosts on Mint/elementary, but wanted to replicate on PiOS just to be sure)

### Device (s)

Raspberry Pi Zero W / WH

### System

Linux raspberrypi 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux

### Logs

on Pi side dwc2 and g_ether are loaded
```
pi@raspberrypi:~ $ lsmod | grep g_ether
g_ether                16384  0
usb_f_rndis            28672  2 g_ether
u_ether                24576  3 usb_f_ecm,g_ether,usb_f_rndis
libcomposite           57344  3 usb_f_ecm,g_ether,usb_f_rndis
```
Host dmesg log at initial successful setup:

```
[ 8402.427773] usb 1-6: new high-speed USB device number 9 using xhci_hcd
[ 8402.581291] usb 1-6: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.10
[ 8402.581304] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8402.581309] usb 1-6: Product: RNDIS/Ethernet Gadget
[ 8402.581313] usb 1-6: Manufacturer: Linux 5.10.92+ with 20980000.usb
[ 8402.664369] cdc_subset: probe of 1-6:1.0 failed with error -22
[ 8402.665751] cdc_subset 1-6:1.1 usb0: register 'cdc_subset' at usb-0000:00:14.0-6, Linux Device, 76:cc:07:17:33:b8
[ 8402.667233] usbcore: registered new interface driver cdc_subset
[ 8402.675617] cdc_ether: probe of 1-6:1.0 failed with error -16
[ 8402.675754] usbcore: registered new interface driver cdc_ether
[ 8423.929864] usb usb1-port6: disabled by hub (EMI?), re-enabling...
[ 8423.929882] usb 1-6: USB disconnect, device number 9
[ 8423.930502] cdc_subset 1-6:1.1 usb0: unregister 'cdc_subset' usb-0000:00:14.0-6, Linux Device
[ 8424.242824] usb 1-6: new high-speed USB device number 10 using xhci_hcd
[ 8424.392106] usb 1-6: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.10
[ 8424.392119] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8424.392125] usb 1-6: Product: RNDIS/Ethernet Gadget
[ 8424.392129] usb 1-6: Manufacturer: Linux 5.10.92+ with 20980000.usb
[ 8424.393491] cdc_subset: probe of 1-6:1.0 failed with error -22
[ 8424.395244] cdc_ether 1-6:1.0 usb0: register 'cdc_ether' at usb-0000:00:14.0-6, CDC Ethernet Device, b2:49:52:81:91:b6
[ 8424.453832] cdc_ether 1-6:1.0 enxb249528191b6: renamed from usb0
[ 8425.293196] usb usb1-port6: disabled by hub (EMI?), re-enabling...
[ 8425.293213] usb 1-6: USB disconnect, device number 10
[ 8425.293412] cdc_ether 1-6:1.0 enxb249528191b6: unregister 'cdc_ether' usb-0000:00:14.0-6, CDC Ethernet Device
[ 8425.602950] usb 1-6: new high-speed USB device number 11 using xhci_hcd
[ 8425.752191] usb 1-6: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.10
[ 8425.752205] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8425.752210] usb 1-6: Product: RNDIS/Ethernet Gadget
[ 8425.752214] usb 1-6: Manufacturer: Linux 5.10.92+ with 20980000.usb
[ 8425.753531] cdc_subset: probe of 1-6:1.0 failed with error -22
[ 8425.757128] cdc_ether 1-6:1.0 usb0: register 'cdc_ether' at usb-0000:00:14.0-6, CDC Ethernet Device, b2:49:52:81:91:b6
[ 8425.792668] cdc_ether 1-6:1.0 enxb249528191b6: renamed from usb0
[ 8426.489730] usb usb1-port6: disabled by hub (EMI?), re-enabling...
[ 8426.489750] usb 1-6: USB disconnect, device number 11
[ 8426.490043] cdc_ether 1-6:1.0 enxb249528191b6: unregister 'cdc_ether' usb-0000:00:14.0-6, CDC Ethernet Device
[ 8426.794901] usb 1-6: new full-speed USB device number 12 using xhci_hcd
[ 8426.943279] usb 1-6: not running at top speed; connect to a high speed hub
[ 8426.944812] usb 1-6: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.10
[ 8426.944827] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 8426.944833] usb 1-6: Product: RNDIS/Ethernet Gadget
[ 8426.944838] usb 1-6: Manufacturer: Linux 5.10.92+ with 20980000.usb
[ 8426.946793] cdc_subset: probe of 1-6:1.0 failed with error -22
[ 8426.948733] cdc_ether 1-6:1.0 usb0: register 'cdc_ether' at usb-0000:00:14.0-6, CDC Ethernet Device, b2:49:52:81:91:b6
[ 8427.010693] cdc_ether 1-6:1.0 enxb249528191b6: renamed from usb0
[ 8562.859782] bpfilter: Loaded bpfilter_umh pid 8989
[ 8562.859972] Started bpfilter
```

After Pi update and reboot:
```
[10767.380823] usb 1-6: USB disconnect, device number 12
[10767.381095] cdc_ether 1-6:1.0 enxb249528191b6: unregister 'cdc_ether' usb-0000:00:14.0-6, CDC Ethernet Device
[10782.275325] usb 1-6: new full-speed USB device number 13 using xhci_hcd
[10782.867335] usb 1-6: new high-speed USB device number 14 using xhci_hcd
[10783.016650] usb 1-6: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.15
[10783.016665] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[10783.016671] usb 1-6: Product: RNDIS/Ethernet Gadget
[10783.016676] usb 1-6: Manufacturer: Linux 5.15.16+ with 20980000.usb
[10783.017959] cdc_subset: probe of 1-6:1.0 failed with error -22
[10783.021772] cdc_ether 1-6:1.0 usb0: register 'cdc_ether' at usb-0000:00:14.0-6, CDC Ethernet Device, 22:00:5a:c2:68:33
[10783.076760] cdc_ether 1-6:1.0 enx22005ac26833: renamed from usb0
[10807.119845] usb usb1-port6: disabled by hub (EMI?), re-enabling...
[10807.119865] usb 1-6: USB disconnect, device number 14
[10807.120085] cdc_ether 1-6:1.0 enx22005ac26833: unregister 'cdc_ether' usb-0000:00:14.0-6, CDC Ethernet Device
[10807.426411] usb 1-6: new high-speed USB device number 15 using xhci_hcd
[10812.558194] usb 1-6: device descriptor read/64, error -110
[10828.173375] usb 1-6: device descriptor read/64, error -110
[10828.413497] usb 1-6: new high-speed USB device number 16 using xhci_hcd
[10833.553239] usb 1-6: device descriptor read/64, error -110
[10849.164673] usb 1-6: device descriptor read/64, error -110
[10849.272833] usb usb1-port6: attempt power cycle
[10849.924699] usb 1-6: new high-speed USB device number 17 using xhci_hcd
[10860.180303] xhci_hcd 0000:00:14.0: Abort failed to stop command ring: -110
[10860.180322] xhci_hcd 0000:00:14.0: xHCI host controller not responding, assume dead
[10860.180330] xhci_hcd 0000:00:14.0: HC died; cleaning up
[10860.180620] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[10860.184073] clocksource: timekeeping watchdog on CPU3: Marking clocksource 'tsc' as unstable because the skew is too large:
[10860.184077] clocksource:                       'acpi_pm' wd_now: c703a5 wd_last: a0bee5 mask: ffffff
[10860.184079] clocksource:                       'tsc' cs_now: cdcfdfca6d0 cs_last: cda57c49590 mask: ffffffffffffffff
[10860.184080] tsc: Marking TSC unstable due to clocksource watchdog
[10860.184097] TSC found unstable after boot, most likely due to broken BIOS. Use 'tsc=unstable'.
[10860.184097] sched_clock: Marking unstable (10860528768984, -344461202)<-(10860201573059, -17476187)
[10860.184510] clocksource: Checking clocksource tsc synchronization from CPU 1.
[10860.184551] clocksource: Switched to clocksource acpi_pm
[10860.596489] usb 1-6: device not accepting address 17, error -108
[10860.596586] usb usb1-port6: couldn't allocate usb_device
[10860.596634] usb 1-7: USB disconnect, device number 3
[10860.649522] usb 1-10: USB disconnect, device number 4
```

### Additional context

@pelwell @P33M  would you have any hint?

pelwell, 2022-02-04:
Not off the top of my head. However, since we are switching to 5.15 imminently it's probable time to break out the USB analyser.

macmpi, 2022-02-07:
Just tried again on latest 5.15.18 with `rpi-update` again: it it just works mint!
Thanks!

closed 2022-02-07 by macmpi

pelwell, 2022-02-07:
That's good timing - it was next on the list of things to look at.

macmpi, 2022-02-16:
note: 5.15.24 further fixes many remaining gadget issues, hopefully for good...
https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.15.24