raspberrypi/linux #4876: Bullseye 64-bit 6x slower than 32-bit on RPi Zero 2.

Qengineering, 2022-02-07:
### Describe the bug

The 64-bit version of Bullseye runs very slow on a Raspberry Pi Zero 2 compared to the 32-bit Bullseye version.

### Steps to reproduce the behaviour

1. Flash a 64-bit Bullseye OS on a new SD.
2. Boot as usual, following the startup instructions.
3. Get WiFi working and install the latest updates.
4. Fire-up Chromium and see how long it takes to load the default [home page](https:://www.raspberrypi.com/).
5. Repeat the same steps with a 32-bit Bullseye.
6. Repeat both steps with a RPi 3 B+ (1 GB RAM, 1.4 GHz).

The outcome:

[image]

Chromium was chosen here solely for the reproducibility of the test.

Note the almost identical load times of the 32-bit RPi 3 and RPi Zero 2 once corrected for their clock frequency.

Not only does Chromium suffers this lack of speed, all other programs as well. For instance gcc compile times.

[image]

### Device (s)

Raspberry Pi Zero 2 W

### System

[image]

pelwell, 2022-02-07:
The 64-bit image is not recommended on Zero 2 W - RAM is too constrained - but run the Lite image if you must.

JamesH65, 2022-02-07:
Also, Chromium is a major memory hog, and the 512 on the Zero is simply not enough to run it with any sort of decent performance.

pelwell, 2022-02-07:
Why? Those 64-bit pointers do take up more space, and when you are balanced on the edge of not-enough-memory, every byte counts.

closed 2022-02-14 by Qengineering