raspberrypi/linux #4879: Unable to handle kernel NULL pointer dereference at virtual address 00000028

abraunegg, 2022-02-08:
### Describe the bug

Post updating a system running `2020-05-27-raspios-buster-armhf` using `kernel7.img` (this is the *only* image archive that I can get to boot without this crash), the following system crash occurs on reboot. The crash is the same regardless of update method:
* `sudo rpi-update`
* Using `sudo apt full-upgrade -y`

### Steps to reproduce the behaviour

1. Boot system with `2020-05-27-raspios-buster-armhf` and `kernel7.img` on a RPI 2B
2. Update the system via `sudo rpi-update` or `sudo apt full-upgrade -y`

### Device (s)

Raspberry Pi 2 Mod. B

### System

System crashes before login post upgrade. This would only be valid for the system state *before* the upgrade.

### Logs

**Before Upgrade:**
```
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Feb  7 22:22:38 2022
pi@raspberrypi:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 4.19.118-v7+
#############################################################
WARNING: This update bumps to rpi-5.15.y linux tree
See: https://forums.raspberrypi.com/viewtopic.php?t=322879

'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.

DO NOT use 'rpi-update' as part of a regular update process.
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   173  100   173    0     0    239      0 --:--:-- --:--:-- --:--:--   252
100  123M    0  123M    0     0   143k      0 --:--:--  0:14:38 --:--:--  179k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 5.15.21-v8+
 *** depmod 5.15.21-v7l+
 *** depmod 5.15.21+
 *** depmod 5.15.21-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
[... 182 lines cut ...]
[  819.605028] Hardware name: BCM2835
[  819.609717] Backtrace: 
[  819.614524] [<80a3c57c>] (dump_backtrace) from [<80a3c7c4>] (show_stack+0x20/0x24)
[  819.622910]  r7:00000015 r6:80d18310 r5:00000080 r4:60000193
[  819.627370] [<80a3c7a4>] (show_stack) from [<80a40cfc>] (dump_stack_lvl+0x70/0x94)
[  819.632232] [<80a40c8c>] (dump_stack_lvl) from [<80a40d38>] (dump_stack+0x18/0x1c)
[  819.636753]  r7:00000015 r6:00000002 r5:80fe9344 r4:81050380
[  819.639372] [<80a40d20>] (dump_stack) from [<8010e6b0>] (do_handle_IPI+0x2dc/0x310)
[  819.643541] [<8010e3d4>] (do_handle_IPI) from [<8010e70c>] (ipi_handler+0x28/0x30)
[  819.647891]  r9:00000000 r8:10c5387d r7:00000015 r6:814e6340 r5:80fe9344 r4:814e3e00
[  819.652074] [<8010e6e4>] (ipi_handler) from [<8018e968>] (handle_percpu_devid_irq+0x8c/0x1d0)
[  819.657656] [<8018e8dc>] (handle_percpu_devid_irq) from [<80187c10>] (generic_handle_domain_irq+0x3c/0x4c)
[  819.663279]  r7:00000004 r6:814e3818 r5:80fe9464 r4:bc8110e0
[  819.666365] [<80187bd4>] (generic_handle_domain_irq) from [<8069a3f8>] (bcm2836_arm_irqchip_handle_ipi+0xa4/0xc4)
[  819.673328] [<8069a354>] (bcm2836_arm_irqchip_handle_ipi) from [<80188228>] (handle_domain_irq+0x6c/0x88)
[  819.679223]  r7:00000004 r6:00000000 r5:00000000 r4:80e9110c
[  819.682248] [<801881bc>] (handle_domain_irq) from [<80101228>] (bcm2836_arm_irqchip_handle_irq+0x5c/0x60)
[  819.687540]  r7:10c5387d r6:ffffffff r5:80000010 r4:76ec725c
[  819.690936] [<801011cc>] (bcm2836_arm_irqchip_handle_irq) from [<80100dac>] (__irq_usr+0x4c/0x60)
[  819.696703] Exception stack(0x82ae1fb0 to 0x82ae1ff8)
[  819.700600] 1fa0:                                     fffffffe 000001ed 00000424 00000000
[  819.707103] 1fc0: 7e821990 00000000 00000001 00000027 7e81f158 00000000 00000009 7e81f0a4
[  819.711995] 1fe0: 004cddfc 7e81f064 004b27cc 76ec725c 80000010 ffffffff
[  819.719549] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

pelwell, 2022-02-08:
It's working here, running Chromium on 5.15.21-v7+.

What software do you have running in the background for it to be in `__bpf_prog_free` - presumably something network-related.

abraunegg, 2022-02-08:
@pelwell 
This is a clean install - nothing else installed, configured or enabled except for enabling `ssh`. Nothing was disabled.

After creating the initial filesystem using `2020-05-27-raspios-buster-armhf` - the *only* things done once this is booted are:
```
pi@raspberrypi:~ $ history
    1  passwd
    2  sudo passwd root
    3  sudo raspi-config
	
		expand disk
		enable ssh
```

The only things running then would be `dhcpd` and `sshd` - anything else would be as per system defaults.

abraunegg, 2022-02-08:
If I add `ipv6.disable=1` to the boot parameters / cmdline.txt - at least I can boot, but the network interface does not even get an IPv4 address. On poweroff, however:

```
[  OK  ] Started Terminate Plymouth Boot Screen.

Raspbian GNU/Linux 10 raspberrypi ttyAMA0

raspberrypi login: root
Password: 
Linux raspberrypi 5.15.21-v7+ #1523 SMP Mon Feb 7 18:31:44 GMT 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@raspberrypi:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether 40:54:00:12:34:57  txqueuelen 1000  (Ethernet)
        RX packets 1  bytes 302 (302.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1  bytes 385 (385.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@raspberrypi:~# poweroffusbnet: failed control transaction: request 0x2143 value 0xe index 0x0 length 0x0

root@raspberrypi:~# usbnet: failed control transaction: request 0x2143 value 0xe index 0x0 length 0x0
[  OK  ] Started Show Plymouth Power Off Screen.
[  OK  ] Stopped LSB: Switch to ond…(unless shift key is pressed).
[**    ] (1 of 3) A stop job is running for …Manager for UID 0 (47s / 2min 11s)
[  763.407688] 8<--- cut here ---
[  763.414136] Unable to handle kernel NULL pointer dereference at virtual address 00000028
[  763.419521] pgd = cd0f544e
[  763.425313] [00000028] *pgd=00000000
[  763.436455] Internal error: Oops: 17 [#1] SMP ARM
[  763.444570] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc cdc_subset cdc_ether raspberrypi_hwmon uio_pdrv_genirq fixed uio i2c_dev i
p_tables x_tables ipv6
[  763.483902] CPU: 0 PID: 11 Comm: ksoftirqd/0 Tainted: G        W         5.15.21-v7+ #1523
[  763.492386] Hardware name: BCM2835
[  763.496998] PC is at bcm2835_sdhost_dma_complete+0x9c/0x1f4
[  763.503086] LR is at log_event_impl.part.0+0x90/0x94
[  763.506526] pc : [<80858b28>]    lr : [<80857200>]    psr: a0000193
[  763.510452] sp : 81557d08  ip : 81557ce8  fp : 81557d34
[  763.515173] r10: ffffe000  r9 : 00000100  r8 : 00000000
[  763.520546] r7 : 00000000  r6 : a0000113  r5 : 810b7be4  r4 : 81d28b40
[  763.529843] r3 : 81ccd640  r2 : 00000000  r1 : a0000193  r0 : 810b7be4
[  763.536767] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  763.543127] Control: 10c5387d  Table: 051dc06a  DAC: 00000055
[  763.550476] Register r0 information: non-slab/vmalloc memory
[  763.564640] Register r1 information: non-slab/vmalloc memory
[  763.572154] Register r2 information: NULL pointer
[  763.582512] Register r3 information: slab kmalloc-512 start 81ccd600 pointer offset 64 size 512
[  763.599302] Register r4 information: slab kmalloc-2k start 81d28800 pointer offset 832 size 2048
[  763.611990] Register r5 information: non-slab/vmalloc memory
[  763.615193] Register r6 information: non-slab/vmalloc memory
[  763.621826] Register r7 information: NULL pointer
[  763.630244] Register r8 information: NULL pointer
[  763.634597] Register r9 information: non-paged memory
[  763.639700] Register r10 information: non-paged memory
[  763.643385] Register r11 information: non-slab/vmalloc memory
[  763.647388] Register r12 information: non-slab/vmalloc memory
[... 165 lines cut ...]
[  764.532118] [<8010e3d4>] (do_handle_IPI) from [<8010e70c>] (ipi_handler+0x28/0x30)
[  764.538590]  r9:81560000 r8:00000001 r7:00000015 r6:814e6340 r5:80fe9344 r4:814e3e00
[  764.543505] [<8010e6e4>] (ipi_handler) from [<8018e968>] (handle_percpu_devid_irq+0x8c/0x1d0)
[  764.549875] [<8018e8dc>] (handle_percpu_devid_irq) from [<80187c10>] (generic_handle_domain_irq+0x3c/0x4c)
[  764.556781]  r7:00000004 r6:814e3818 r5:80fe9464 r4:bc8110f0
[  764.561454] [<80187bd4>] (generic_handle_domain_irq) from [<8069a3f8>] (bcm2836_arm_irqchip_handle_ipi+0xa4/0xc4)
[  764.567373] [<8069a354>] (bcm2836_arm_irqchip_handle_ipi) from [<80188228>] (handle_domain_irq+0x6c/0x88)
[  764.573780]  r7:00000004 r6:00000000 r5:00000000 r4:80e9110c
[  764.580177] [<801881bc>] (handle_domain_irq) from [<80101228>] (bcm2836_arm_irqchip_handle_irq+0x5c/0x60)
[  764.586622]  r7:81561f64 r6:ffffffff r5:60000013 r4:80108864
[  764.590823] [<801011cc>] (bcm2836_arm_irqchip_handle_irq) from [<80100aac>] (__irq_svc+0x4c/0x64)
[  764.596306] Exception stack(0x81561f30 to 0x81561f78)
[  764.599048] 1f20:                                     18b76bae 00000000 00000001 801190c0
[  764.604545] 1f40: ffffe000 00000008 80f05058 80f050a8 00000001 81037ca0 80d0eedc 81561f8c
[  764.609059] 1f60: 81561f90 81561f80 80108860 80108864 60000013 ffffffff
[  764.612837] [<8010881c>] (arch_cpu_idle) from [<80a4cc48>] (default_idle_call+0x4c/0xd8)
[  764.618322] [<80a4cbfc>] (default_idle_call) from [<8015a528>] (do_idle+0x114/0x164)
[  764.621677] [<8015a414>] (do_idle) from [<8015a89c>] (cpu_startup_entry+0x28/0x2c)
[  764.626338]  r10:00000000 r9:410fc075 r8:0000406a r7:81561ff8 r6:10c0387d r5:00000003
[  764.630338]  r4:00000096 r3:60000093
[  764.632333] [<8015a874>] (cpu_startup_entry) from [<8010ed30>] (secondary_start_kernel+0x168/0x174)
[  764.637202] [<8010ebc8>] (secondary_start_kernel) from [<001017f0>] (0x1017f0)
[  764.641462]  r5:00000055 r4:0157806a
[  764.645035] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```

pelwell, 2022-02-08:
There is something different about your system - the SD card, the power supply, or the RPi itself - because I don't get the same behaviour and nobody else has reported it.

What are you able to change? Adding `over_voltage=2`, `core_freq=250` and `arm_freq=600` to config.txt will help to rule out a few potential causes.

abraunegg, 2022-02-08:
@pelwell 
> What are you able to change? Adding `over_voltage=2`, `core_freq=250` and `arm_freq=600` to config.txt will help to rule out a few potential causes.

Will test this shortly.

Currently testing adding the following to `/etc/sysctl.conf` before upgrade (ensuring system is rebooted post this change as well):
```
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
```

abraunegg, 2022-02-10:
@pelwell 
> Adding `over_voltage=2`, `core_freq=250` and `arm_freq=600` to `config.txt` will help to rule out a few potential causes

Unfortunately adding all of these, or in any combination of fails to make any difference. Post updating the system, with those options in place, in all circumstances the kernel crash occurs at the same virtual address.

In testing however disabling IPv6 via sysctl.conf (as per above) *before* booting, the following outcomes occur.

* 2020-05-27-raspios-buster-armhf
  * before upgrade, boot no issue, system 100% stable
  * reboot post upgrade is stable, however, on first attempt to access system via SSH, system crashes
* 2020-08-20-raspios-buster-armhf
  * system hangs when trying to start a job for swap file: (1 of 7) A start job is running for ... a swap file (X min YYs / no limit)
  * does not complete booting despite being left for hours on the swap job
  * updates not attempted
* 2020-12-02-raspios-buster-armhf
  * system starts booting but starts kicking out some rcu_sched self-detected stall on CPU (not seen with prior 2 OS builds above)
  * system crashes with a kernel oops: kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:1093
  * updates not possible
* 2021-01-11-raspios-buster-armhf
  * system starts booting but starts kicking out some rcu_sched self-detected stall on CPU
  * system crashes with a kernel oops: kernel BUG at drivers/mmc/host/bcm2835-sdhost.c:1093
  * updates not possible
* 2021-03-04-raspios-buster-armhf
  * system starts booting but starts kicking out some rcu_sched self-detected stall on CPU
  * system crashes with: Unable to handle kernel NULL pointer dereference at virtual address 00000028
  * updates not possible
* 2021-05-07-raspios-buster-armhf
  * system starts booting but starts kicking out some rcu_sched self-detected stall on CPU
  * system boots eventually without issue, no crash, but it is *very very very slow*
  * access via SSH is possible - but again but it is *very very very slow* - molasses slow
  * updates not attempted ..

What else can be disabled in `config.txt` to ensure totally minimal hardware is active? I have read some options re disabling audio etc (I certainly do not need audio or any other option - IR etc) .. what about extended graphics / graphics options - again not needed.

The aim here is to have a viable | working Pi2 running as a `armv7l` so that *when* I get to install Docker, I can build & test 32bit Docker images correctly.

**Disclaimer:** I am the developer|maintainer of the OneDrive Client for Linux (https://github.com/abraunegg/onedrive) - so essentially I am trying to expand my testing footprint to ensure platforms that folk use are tested correctly.

Open to suggestions here ..

pelwell, 2022-02-10:
As well as the config.txt settings - a long shot, but worth trying - I listed a few things that make your system different to mine: the SD card, the power supply, and the RPi itself.

SD cards can go bad, an inadequate power supply (or bad power cable) will cause problems under load, and RPis do break. I'm going to rule out other environmental factors for now. All you can do is try changing each of those elements until it starts working.

I don't know how many of the millions of Pi 2s are still in use (we don't track devices), but I'm sure there are many other 2B users out there happily running 5.15 like me.

speters, 2022-06-08:
I ran into this - what appears to me like the same - problem on QEMU when emulating Raspi2b.

I can not reproduce a specific time of crash. Sometimes, it makes it into the initial setup dialogue, sometimes it crashes earlier when the serial login prompt appears.

Software is stock 2022-04-04-raspios-bullseye-armhf-lite.img just resized to 4GB (` qemu-img resize 2022-04-04-raspios-bullseye-armhf-lite.img 4G `) prior to first booting.

```
[ 9900.186706] 8<--- cut here ---
[ 9900.203823] Unable to handle kernel NULL pointer dereference at virtual address 00000028
[ 9900.210531] pgd = 63f9844a
[ 9900.217649] [00000028] *pgd=00000000
[ 9900.230398] Internal error: Oops: 5 [#1] SMP ARM
[ 9900.236249] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc joydev cdc_ether cdc_subset raspberrypi_hwmon uio_pdrv_genirq uio fixed drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[ 9900.270318] CPU: 0 PID: 33 Comm: kworker/0:1 Not tainted 5.15.32-v7+ #1538
[ 9900.272319] Hardware name: BCM2835
[ 9900.277633] Workqueue: usb_hub_wq hub_event
[ 9900.283610] PC is at bcm2835_sdhost_dma_complete+0x9c/0x1f4
[ 9900.290422] LR is at log_event_impl.part.0+0x90/0x94
[ 9900.293663] pc : [<80858f94>]    lr : [<8085766c>]    psr: a0000193
[ 9900.299945] sp : 81627a58  ip : 81627a38  fp : 81627a84
[ 9900.301257] r10: ffffe000  r9 : 00000100  r8 : 00000000
[ 9900.308805] r7 : 00000000  r6 : a0000113  r5 : 810b7b24  r4 : 81ee8b40
[ 9900.311151] r3 : 8166b640  r2 : 00000000  r1 : a0000193  r0 : 810b7b24
[ 9900.314790] Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 9900.321244] Control: 10c5387d  Table: 0506406a  DAC: 00000055
[ 9900.327170] Register r0 information: non-slab/vmalloc memory
[ 9900.338732] Register r1 information: non-slab/vmalloc memory
[ 9900.342604] Register r2 information: NULL pointer
[ 9900.350516] Register r3 information: slab kmalloc-512 start 8166b600 pointer offset 64 size 512
[ 9900.365881] Register r4 information: slab kmalloc-2k start 81ee8800 pointer offset 832 size 2048
[ 9900.373349] Register r5 information: non-slab/vmalloc memory
[ 9900.380395] Register r6 information: non-slab/vmalloc memory
[ 9900.383837] Register r7 information: NULL pointer
[ 9900.390547] Register r8 information: NULL pointer
[ 9900.393368] Register r9 information: non-paged memory
[ 9900.400888] Register r10 information: non-paged memory
[ 9900.406612] Register r11 information: non-slab/vmalloc memory
[ 9900.412035] Register r12 information: non-slab/vmalloc memory
[ 9900.420451] Process kworker/0:1 (pid: 33, stack limit = 0x8b7d4d08)
[ 9900.422798] Stack: (0x81627a58 to 0x81628000)
[ 9900.426902] 7a40:                                                       8022f0a0 8023047c
[ 9900.430405] 7a60: 8166b480 8514ad84 8514adbc 81627a58 81627a90 00000122 81627adc 81627a88
[ 9900.433068] 7a80: 806e0198 80858f04 80f05008 00000000 81627a90 81627a90 00000000 00000000
[ 9900.438497] 7aa0: 80858ef8 00000000 81ee8b40 2d9a6595 81627adc 8166b480 81ee8b64 b77142d4
[ 9900.444364] 7ac0: 00000000 00000006 81626000 ffffe000 81627b04 81627ae0 80126800 806dffc4
[ 9900.450076] 7ae0: 80f03098 00000005 00000006 00000100 000000b0 81626000 81627b14 81627b08
[ 9900.453692] 7b00: 80126844 8012672c 81627b84 81627b18 801014a8 80126828 00000080 2d9a6595
[ 9900.461094] 7b20: 801265f0 80b02160 04208060 00000001 80d0b78c 80f03d00 000ea1ea 80e8a2cc
[ 9900.464802] 7b40: 0000000a 80e91b00 81038594 80f05058 80e8a348 80f03080 8018e050 ffffe000
[ 9900.470488] 7b60: 00000000 00000000 00000008 00000402 81626000 00000000 81627b9c 81627b88
[ 9900.475156] 7b80: 801265f0 8010131c 80e9110c 00000000 81627bbc 81627ba0 80188774 80126524
[ 9900.480332] 7ba0: 80a4d57c 60000113 ffffffff 81627c04 81627bcc 81627bc0 80101268 80188710
[ 9900.484653] 7bc0: 81627c34 81627bd0 80100aec 80101218 80a4d584 60000113 60000193 60000193
[ 9900.490638] 7be0: 60000113 60000113 81e8d400 fffffffe 00000402 81c93000 00000000 81627c34
[ 9900.493504] 7c00: 81627bf8 81627c20 8022f0a0 80a4d57c 60000113 ffffffff 80a4d584 8022f054
[ 9900.499688] 7c20: 81e8af00 60000113 81627c54 81627c38 807b9b2c 80a4d53c 81e8af00 80f05008
[ 9900.502863] 7c40: 81e8af10 81c9345c 81627c8c 81627c58 807baf5c 807b9ad4 81627c8c 81c93464
[ 9900.508542] 7c60: 00000000 8175e000 81627c8c 2d9a6595 81c92000 00000000 8175e000 81c9345c
[ 9900.511997] 7c80: 81627ca4 81627c90 8086e45c 807baeec 81c92000 00000000 81627cd4 81627ca8
[ 9900.517350] 7ca0: 80870a94 8086e43c 81e9be00 00000000 81e8d400 80f05008 81e8d504 00000402
[ 9900.524742] 7cc0: 00000001 82564200 81627d1c 81627cd8 807bf610 808708e8 80a45ee8 00000000
[ 9900.530363] 7ce0: 80d6f018 00000400 80e8fc50 2d9a6595 81e8d504 81e8d400 81e8d480 81e8d504
[ 9900.534917] 7d00: 81e8d504 80f05008 00000000 ffffe000 81627d34 81627d20 807c0930 807bf584
[ 9900.540423] 7d20: 81e8d480 807c08f8 81627d5c 81627d38 80743f1c 807c0904 81e8d480 807c08f8
[ 9900.546687] 7d40: 81e8d504 00000008 80f05008 0000000c 81627d7c 81627d60 807440c0 80743ecc
[ 9900.550681] 7d60: 81e8d480 807c08f8 81e8d504 00000008 81627ddc 81627d80 807453ac 80744050
[ 9900.551479] 7d80: 81627de0 8022f32c 80745acc 2d9a6595 80173064 00000000 81627dc4 81627da8
[ 9900.556144] 7da0: 8022f32c 8010cde8 81e8d504 2d9a6595 81e8d504 81e8d480 0000000c 81e8d504
[ 9900.560764] 7dc0: 00000001 60000113 00000000 81e9a900 81627e04 81627de0 80745ae0 807452bc
[ 9900.566068] 7de0: 81e8d400 81e8d4c4 00000000 00000001 81e9aa00 00000000 81627e1c 81627e08
[... 200 lines cut ...]
[... cut ...]
qemu-system-arm -M raspi2b -kernel kernel7.img -serial stdio \
    -append 'rw earlycon=pl011,0x3f201000 console=ttyAMA0 loglevel=2 root=/dev/mmcblk0p2 fsck.repair=yes net.ifnames=0 rootwait memtest=1 dwc_otg.fiq_fsm_enable=0' \
    -drive file=2022-04-04-raspios-bullseye-armhf-lite.img,format=raw,if=sd \
    -dtb bcm2709-rpi-2-b.dtb \
    -k de -no-reboot -usb -device usb-kbd -device usb-tablet -device usb-net,netdev=ulan,mac=02:ca:fe:f0:0d:01 -netdev user,id=ulan,hostfwd=tcp::2222-:22 -vnc :0

```