raspberrypi/linux #4883: Raspberry3b+:cp: cannot stat 'arch/arm64/boot/dts/broadcom/*.dtb': No such file or directory

tcexeexe, 2022-02-10:
### Describe the bug

I follow the guide from `https://www.raspberrypi.com/documentation/computers/linux_kernel.html#default_configuration` to build the kernel, follow the step  
```
cd linux
KERNEL=kernel8
make bcm2711_defconfig
make -j4 Image modules dtbs
sudo make modules_install
```
but when l input 
`sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/`
 an error occurred cannot stat 'arch/arm64/boot/dts/broadcom/*.dtb': No such file or directory

and l affirm that their isn't any .dtb file in `arch/arm64/boot/dts/broadcom`.

Meanwhile, I notice that the destination of the dtb file is `arch/arm/boot/dts/` not the 'arch/arm64/boot/dts'
[image]

So are there any procedural issues in the official guidelines when compiling 64-bit?

### Steps to reproduce the behaviour

0

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Raspberry Pi reference 2022-01-28
Jan 20 2022 13:58:22 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
Linux raspberrypi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux

pelwell, 2022-02-10:
You are trying to build a 64-bit kernel from a 32-bit kernel (`5.10.92-v7+`), and therefore a 32-bit userland. That is not a native build - it's cross-compilation, the steps for which are here: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel

By the way, I have never tried building a 64-bit kernel on a 32-bit RPiOS - it might work, but I'm not going to guarantee it.

tcexeexe, 2022-02-10:
> You are trying to build a 64-bit kernel from a 32-bit kernel (`5.10.92-v7+`), and therefore a 32-bit userland. That is not a native build - it's cross-compilation, the steps for which are here: [...]

I am a beginner, thank you for your guidance!

closed 2022-02-10 by tcexeexe