raspberrypi/linux #4886: Request to add CONFIG_MD_MULTIPATH=m to standard RPI kernel builds

satmandu, 2022-02-13:
### Describe the bug

I use both ubuntu and Raspi-OS, and use the rpi-firmware kernels for both.

The ubuntu-server 22.04 LTS release has multipath-tools installed by default, which requires the `dm-multipath` module.
(This may be a new requirement.)

Not having the module results in errors:
```
Feb 13 15:36:44 rpi4b modprobe[4043]: modprobe: WARNING: Module dm-multipath not found in directory /lib/modules/5.15.21-v8+
Feb 13 15:36:44 rpi4b multipathd[4044]: --------start up--------
Feb 13 15:36:44 rpi4b multipathd[4044]: read /etc/multipath.conf
Feb 13 15:36:44 rpi4b multipathd[4044]: path checkers start up
Feb 13 15:36:44 rpi4b multipathd[4044]: failed to increase buffer size
Feb 13 15:36:44 rpi4b multipathd[4044]: DM multipath kernel driver not loaded
Feb 13 15:36:44 rpi4b multipathd[4044]: DM multipath kernel driver not loaded
Feb 13 15:36:44 rpi4b systemd[1]: multipathd.service: Main process exited, code=exited, status=1/FAILURE
Feb 13 15:36:44 rpi4b systemd[1]: multipathd.service: Failed with result 'exit-code'.
Feb 13 15:36:44 rpi4b systemd[1]: Failed to start Device-Mapper Multipath Device Controller.
dpkg: error processing package multipath-tools (--configure):
 installed multipath-tools package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of ubuntu-server-raspi:
 ubuntu-server-raspi depends on multipath-tools; however:
```

The current  rpi-firmware kernels do not have that set for build:
```
modprobe configs
zcat /proc/config.gz  | grep CONFIG_MD_MULTIPATH
# CONFIG_MD_MULTIPATH is not set
```

On ubuntu kernels this is set:
```
grep CONFIG_MD_MULTIPATH /boot/config-5.15.0-1002-raspi
CONFIG_MD_MULTIPATH=m
```

Is there any chance we could have that built as a module for the kernels available via [https://github.com/raspberrypi/rpi-firmware](https://github.com/raspberrypi/rpi-firmware) ?

### Steps to reproduce the behaviour

Install ubuntu 21.10 on a rpi4b.

Install rpi-firmware kernel using my own script (happy to post it.)

Update to 22.04.

Watch breakage happen:

```
Errors were encountered while processing:
 multipath-tools
 ubuntu-server-raspi
 ubuntu-server
needrestart is being skipped since dpkg has failed
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
satadru@rpi4b ~$ cat /etc/rpi-issue
cat: /etc/rpi-issue: No such file or directory
1satadru@rpi4b ~$ vcgencmd version
Feb  4 2022 12:05:40
Copyright (c) 2012 Broadcom
version a26faf97e3bf76bcc23949d7cdab2f96f399a0c3 (clean) (release) (start)
satadru@rpi4b ~$ uname -a
Linux rpi4b 5.15.21-v8+ #1523 SMP PREEMPT Mon Feb 7 18:36:51 GMT 2022 aarch64 aarch64 aarch64 GNU/Linux
satadru@rpi4b ~$
```

pelwell, 2022-02-14:
The relevant Kconfig file says:
```
config MD_MULTIPATH
	tristate "Multipath I/O support (deprecated)"
	depends on BLK_DEV_MD
	help
	  MD_MULTIPATH provides a simple multi-path personality for use
	  the MD framework.  It is not under active development.  New
	  projects should consider using DM_MULTIPATH which has more
	  features and more testing.

	  If unsure, say N.
```
Note that the warning messages mentions `dm-multipath`, not `md-multipath`. Have you tried enabling `CONFIG_DM_MULTIPATH=m` instead?

Also, a brief explanation of why users might want to use multipath tools wouldn't go amiss.

satmandu, 2022-02-14:
> Note that the warning messages mentions `dm-multipath`, not `md-multipath`.

Mea culpa. It is indeed `dm-multipath`.

I'll of course add an explainer.

dakaix, 2023-01-09:
Would also love to see this included, I've just deployed a Microk8s cluster on Raspbian backed by iSCSI, and didn't find out until several hours in that dm-multipath wasn't available! Would rather not need to compile a custom kernel everytime I need to apply some updates.

pelwell, 2023-01-11:
That's done - see https://github.com/raspberrypi/linux/commit/82efd01771e9d04d6016f6b00ff7ecac89d9d869.

Sorry, @satmandu, this one got forgotten. In future, feel free to nudge us/me.

satmandu, 2023-01-25:
I don't think this is getting set in the rpi-firmware next branch kernels? (I keep seeing commit messages which mention it getting added to the kernel configs. Did https://github.com/raspberrypi/linux/commit/5ae8b4465edc9f69d37af6a9fbd9877a71d1a24a finally fix that?

```
satadru@rpi4b ~$ uname -a
Linux rpi4b 6.1.8-v8+ #1622 SMP PREEMPT Tue Jan 24 11:39:23 GMT 2023 aarch64 aarch64 aarch64 GNU/Linux
satadru@rpi4b ~$ uname -a
Linux rpi4b 6.1.8-v8+ #1622 SMP PREEMPT Tue Jan 24 11:39:23 GMT 2023 aarch64 aarch64 aarch64 GNU/Linux
satadru@rpi4b ~$ sudo modprobe configs
satadru@rpi4b ~$ zcat /proc/config.gz  | grep CONFIG_MD_MULTIPATH
# CONFIG_MD_MULTIPATH is not set
```

pelwell, 2023-01-25:
Yes:
```
pi@raspberrypi:~$ sudo modprobe configs
pi@raspberrypi:~$ zcat /proc/config.gz  | grep CONFIG_MD_MULTIPATH
# CONFIG_MD_MULTIPATH is not set
```
But:
```
pi@raspberrypi:~$ zcat /proc/config.gz  | grep CONFIG_DM_MULTIPATH
CONFIG_DM_MULTIPATH=m
# CONFIG_DM_MULTIPATH_QL is not set
# CONFIG_DM_MULTIPATH_ST is not set
# CONFIG_DM_MULTIPATH_HST is not set
# CONFIG_DM_MULTIPATH_IOA is not set
```

satmandu, 2023-02-01:
I verified that this works with the official 6.1.9-v8+ kernel packaged with my https://gist.github.com/satmandu/a507c59d84737f6d29ff353395819d51 script, installed on ubuntu 23.04/lunar.

Running `sudo apt install multipath-tools ubuntu-server` caused no problems.