raspberrypi/linux #4888: HDMI display freezing on dual-monitor

khapin, 2022-02-13:
### Describe the bug

After a seemingly random amount of time (minutes to ~2 hours) the display freezes on both screens.
I can't CTRL+Alt+Fx to see a virtual terminal, but can still log on via ssh, where all services are still running.
Stopping lightdm doesn't restore the display, and even after issuing a reboot, the display was still showing the same as when it froze. The only way was to unplug the power for a couple of seconds.
This never happens with a single monitor connected.
It happens whether the monitors are identical or not (tried with 2 inputs of the same screen).

### Steps to reproduce the behaviour

Connect 2 monitors via HDMI.
Use the system normally. Dragging screen elements (e.g. windows) seem to make the freeze happen more quickly, but even moving the mouse around can trigger it. I haven't noticed it while not interacting with mouse/keyboard.

### Device (s)

Raspberry Pi 4 Mod. B

### System

cat /etc/rpi-issue 
```
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage4
```
vcgencmd version
```
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
```

uname -a
`Linux hostname-redacted 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux`

### Logs

```
Feb  7 19:57:54 kriek kernel: [122884.128071] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000008
Feb  7 19:57:54 kriek kernel: [122884.128093] Mem abort info:
Feb  7 19:57:54 kriek kernel: [122884.128102]   ESR = 0x96000007
Feb  7 19:57:54 kriek kernel: [122884.128111]   EC = 0x25: DABT (current EL), IL = 32 bits
Feb  7 19:57:54 kriek kernel: [122884.128119]   SET = 0, FnV = 0
Feb  7 19:57:54 kriek kernel: [122884.128127]   EA = 0, S1PTW = 0
Feb  7 19:57:54 kriek kernel: [122884.128134] Data abort info:
Feb  7 19:57:54 kriek kernel: [122884.128142]   ISV = 0, ISS = 0x00000007
Feb  7 19:57:54 kriek kernel: [122884.128150]   CM = 0, WnR = 0
Feb  7 19:57:54 kriek kernel: [122884.128160] user pgtable: 4k pages, 39-bit VAs, pgdp=000000007bcab000
Feb  7 19:57:54 kriek kernel: [122884.128168] [0000000000000008] pgd=000000007bb30003, p4d=000000007bb30003, pud=000000007bb30003, pmd=0000000055be7003, pte=0000000000000000
Feb  7 19:57:54 kriek kernel: [122884.128206] Internal error: Oops: 96000007 [#1] PREEMPT SMP
Feb  7 19:57:54 kriek kernel: [122884.128212] Modules linked in: rfcomm tcp_diag inet_diag cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc xt_nat xt_tcpudp veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_counter xt_addrtype nft_compat nf_tables nfnetlink br_netfilter bridge overlay 8021q garp stp llc hid_logitech_hidpp snd_soc_hdmi_codec joydev sg brcmfmac brcmutil hid_logitech_dj v3d gpu_sched cfg80211 bcm2835_codec(C) raspberrypi_hwmon bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) i2c_brcmstb rfkill v4l2_mem2mem vc_sm_cma(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops snd_bcm2835(C) videobuf2_v4l2 videobuf2_common videodev vc4 mc cec drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd rpivid_mem syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
Feb  7 19:57:54 kriek kernel: [122884.128470] CPU: 0 PID: 20989 Comm: Xorg Tainted: G         C        5.10.92-v8+ #1514
Feb  7 19:57:54 kriek kernel: [122884.128475] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
Feb  7 19:57:54 kriek kernel: [122884.128484] pstate: 20000005 (nzCv daif -PAN -UAO -TCO BTYPE=--)
Feb  7 19:57:54 kriek kernel: [122884.128519] pc : vc4_atomic_check+0x3f0/0x510 [vc4]
Feb  7 19:57:54 kriek kernel: [122884.128539] lr : vc4_atomic_check+0x3e8/0x510 [vc4]
Feb  7 19:57:54 kriek kernel: [122884.128544] sp : ffffffc012b6b9b0
Feb  7 19:57:54 kriek kernel: [122884.128549] x29: ffffffc012b6b9b0 x28: ffffff80490088e0 
Feb  7 19:57:54 kriek kernel: [122884.128559] x27: ffffff807924f000 x26: 0000000000000038 
Feb  7 19:57:54 kriek kernel: [122884.128570] x25: 0000000000000000 x24: ffffff80d915e880 
Feb  7 19:57:54 kriek kernel: [122884.128579] x23: 0000000000000008 x22: ffffff8091180400 
Feb  7 19:57:54 kriek kernel: [122884.128588] x21: 0000000000000000 x20: ffffff8090cb1880 
Feb  7 19:57:54 kriek kernel: [122884.128597] x19: ffffff80d915e880 x18: 0000000000000000 
Feb  7 19:57:54 kriek kernel: [122884.128606] x17: 0000000000000000 x16: 0000000000000000 
Feb  7 19:57:54 kriek kernel: [122884.128615] x15: ffffff80c11f68e0 x14: ffffff80c11f68d8 
Feb  7 19:57:54 kriek kernel: [122884.128625] x13: 0000000000000437 x12: 0000000000000002 
Feb  7 19:57:54 kriek kernel: [122884.128633] x11: 0000000000000000 x10: 000000000000000a 
Feb  7 19:57:54 kriek kernel: [122884.128643] x9 : ffffffc010ad7724 x8 : ffffffc008c2b000 
Feb  7 19:57:54 kriek kernel: [122884.128652] x7 : 0000000000000000 x6 : dead000000000100 
Feb  7 19:57:54 kriek kernel: [122884.128661] x5 : dead000000000122 x4 : ffffff804889d978 
Feb  7 19:57:54 kriek kernel: [122884.128670] x3 : 0000000000000001 x2 : ffffff804889d000 
Feb  7 19:57:54 kriek kernel: [122884.128679] x1 : 0000000000000000 x0 : fffffffffffffffc 
Feb  7 19:57:54 kriek kernel: [122884.128689] Call trace:
Feb  7 19:57:54 kriek kernel: [122884.128709]  vc4_atomic_check+0x3f0/0x510 [vc4]
Feb  7 19:57:54 kriek kernel: [122884.128786]  drm_atomic_check_only+0x454/0x770 [drm]
Feb  7 19:57:54 kriek kernel: [122884.128840]  drm_atomic_nonblocking_commit+0x24/0x68 [drm]
Feb  7 19:57:54 kriek kernel: [122884.128881]  drm_atomic_helper_page_flip+0xe8/0x128 [drm_kms_helper]
Feb  7 19:57:54 kriek kernel: [122884.128900]  vc4_page_flip+0x134/0x2f8 [vc4]
Feb  7 19:57:54 kriek kernel: [122884.128955]  drm_mode_page_flip_ioctl+0x4bc/0x528 [drm]
Feb  7 19:57:54 kriek kernel: [122884.129008]  drm_ioctl_kernel+0xc8/0x118 [drm]
Feb  7 19:57:54 kriek kernel: [122884.129059]  drm_ioctl+0x228/0x488 [drm]
Feb  7 19:57:54 kriek kernel: [122884.129125]  drm_compat_ioctl+0xe4/0xf8 [drm]
Feb  7 19:57:54 kriek kernel: [122884.129153]  __arm64_compat_sys_ioctl+0x108/0x1b0
Feb  7 19:57:54 kriek kernel: [122884.129169]  el0_svc_common.constprop.3+0xc0/0x1f8
Feb  7 19:57:54 kriek kernel: [122884.129178]  do_el0_svc_compat+0x24/0x50
Feb  7 19:57:54 kriek kernel: [122884.129190]  el0_svc_compat+0x20/0x30
Feb  7 19:57:54 kriek kernel: [122884.129197]  el0_sync_compat_handler+0xe4/0x110
Feb  7 19:57:54 kriek kernel: [122884.129204]  el0_sync_compat+0x184/0x1c0
Feb  7 19:57:54 kriek kernel: [122884.129215] Code: 91256021 97f8872d b4fffc40 f9400662 (b9400c07) 
Feb  7 19:57:54 kriek kernel: [122884.129224] ---[ end trace 077cd7c976bcb2c6 ]---

```

### Additional context

At least one other user is experiencing the same issue, as reported here: [https://forums.raspberrypi.com/viewtopic.php?p=1973506#p1973506]

popcornmix, 2023-01-24:
Is this still an issue on RPiOS bullseye and 5.15 kernel?
There's been many kms fixes since this issue was posted.

khapin, 2023-01-28:
Sorry for not updating this ticket earlier, I got rid of my second monitor a few months ago so didn't check whether the issue was still present. 
I've just tried to reproduce the issue now with kernel 5.15.84-v8+, and it seems to be working fine now.

THE-BOW, 2023-01-28:
Hi @popcornmix,

I havent used my Pi for a while due to this issue.
Ive tested last days and could not reproduce the issue...

Thank you.

f8ttyc8t, 2023-01-28:
Confirming it, freezings aren’t an issue on my Pi any longer.

closed 2023-01-28 by popcornmix