raspberrypi/linux #4892: For rpi4 allow dtoverlay i2c-rtc use i2c5 port

andres53016, 2022-02-16:
### Describe the bug

Add functionality for dtoverlay i2c-rtc to use i2c 5 port

### Steps to reproduce the behaviour

The only way to use an rtc on port 5 is activating i2c5 and run at boot, the following

echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-5/new_device

### Device (s)

Raspberry Pi 4 Mod. B

### System

Linux alarmpi 5.15.21-3-rpi-ARCH #1 armv7l GNU/Linux

pelwell, 2022-02-16:
Adding the additional I2C ports to the existing `i2c-rtc` overlay would break it on non-BCM2711 devices - the symbol `i2c5` (for example) doesn't exist in non-BCM2711 Device Trees, so the overlay cannot be applied. There is a way around the problem - create another overlay called `i2c-rtc-pi4` that supports the additional interfaces, and use the overlay_map file to load the appropriate one - but it isn't particularly elegant, and would benefit from a better way to document that some parameters are only available on Pi 4s. I'm not ruling out your request, but I'd rather have some time to think about it first.

However, I can give you a version of the overlay that's been modified to work on `i2c5` instead:
[i2c5-rtc.dtbo.zip](https://github.com/raspberrypi/linux/files/8083627/i2c5-rtc.dtbo.zip)
It was hacked together with a hex editor, but it should do the job.

andres53016, 2022-02-16:
I didn't know all that the change encompassed. Thank you very much for the quick response and for the modified file. It worked perfectly

seamusdemora, 2022-02-17:
Good news! I'll look forward to learning of your final solution. And this raises another question I've had: Currently, we can only access one of the two alarm registers available in the DS3231. I've always assumed that's a limitation of the driver, not the overlay... is that a correct assumption? IOW, I guess the overlay would also need to be modified to support both alarms, but the driver modification is also required, is it not?

pelwell, 2022-07-08:
FYI, https://github.com/raspberrypi/linux/commit/1131d880f3917c697abd64f1c79ddd8a9611286b (not yet in an rpi-update release) adds the parameters `i2c3`, `i2c4`, `i2c5` and `i2c6` to the `i2c-rtc` overlay (and others). To avoid the problem of unresolved symbols it makes use of `target-path`, with the path strings consisting of the DT aliases `i2c3`-`i2c6`.