raspberrypi/linux #4898: 5.15 kernel break Hifiberry DAC 2 HD Hat

apastuszak, 2022-02-17:
### Describe the bug

Upgrading to the 5.15 Linux kernel breaks compatibiilty with the Hifiberry DAC 2 HD. I did an rpi-update, rebooted my pi, and aplay no longer sees the HAT.

This is what is in my config.txt

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,audio=off
max_framebuffers=2
[... 507 lines cut ...]
[   15.384102] NET: Registered PF_BLUETOOTH protocol family
[   15.384115] Bluetooth: HCI device and connection manager initialized
[   15.384146] Bluetooth: HCI socket layer initialized
[   15.384164] Bluetooth: L2CAP socket layer initialized
[   15.384197] Bluetooth: SCO socket layer initialized
[   15.392322] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   15.398039] Bluetooth: HCI UART driver ver 2.3
[   15.398079] Bluetooth: HCI UART protocol H4 registered
[   15.398191] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.398617] Bluetooth: HCI UART protocol Broadcom registered
[   15.633330] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.633350] Bluetooth: BNEP filters: protocol multicast
[   15.633364] Bluetooth: BNEP socket layer initialized
[   15.648443] NET: Registered PF_ALG protocol family
[   31.709759] cam-dummy-reg: disabling
[   31.709800] cam1-reg: disabling
[  804.779109] edt_ft5x06 10-0038: Unable to fetch data, error: -121
[  805.805477] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  805.805507] edt_ft5x06 10-0038: Unable to fetch data, error: -110
[  805.805575] edt_ft5x06 10-0038: Unable to fetch data, error: -5
```

### Additional context

This was working with kernel 5.10.

apastuszak, 2022-02-17:
Hifiberry configuration instructions:

https://www.hifiberry.com/docs/software/configuring-linux-3-18-x/

apastuszak, 2022-02-17:
Uncommented the followling line in my config.txt

`#dtparam=i2s=on`

Rebooted.  No change.

pelwell, 2022-02-17:
What does `i2cdetect -y 1` report?

apastuszak, 2022-02-17:
`Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory`

apastuszak, 2022-02-17:
DOH!  Forgot to enable i2c.

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- 62 -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

pelwell, 2022-02-18:
The overlay should have enabled i2c1, so something has clearly gone wrong. What do you get from this command?
```
$ cat /proc/device-tree/chosen/*warnings
```

apastuszak, 2022-02-18:
```
cat: '/proc/device-tree/chosen/*warnings': No such file or directory
```

pelwell, 2022-02-21:
I don't have a HiFiBerry DAC2 HD HAT to test, but looking at the overlay I see it is expecting to find the DAC at I2C address 0x4c, and a PLL/clock at I2C address 0x62. Your i2cdetect output shows the PLL but no DAC, which is going to cause problems for the driver.

apastuszak, 2022-02-22:
Is there anything I need to do on my side? Do you need more information from me?

pelwell, 2022-02-22:
The working theory is that your hardware has developed a fault - I don't know what else would explain the lack of the DAC at address 0x4c. You could try downgrading to a 5.10 kernel using `sudo BRANCH=stable rpi-update` (and `sudo rpi-update` to get back to 5.15), but I don't expect it to change anything,

Do you have any suggestions, @j-schambacher?

j-schambacher, 2022-02-22:
@apastuszak, I tend to agree with Phil. Please confirm that the downgraded kernel version works (again). We have not yet tested and verified 5.15. But at a first look, I do not see an obvious reason, why it should not work. Thanks for your help

apastuszak, 2022-02-22:
uname -a

```
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
```

aplay -l

```
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplushd], device 0: HiFiBerry DAC+ HD HiFi pcm179x-hifi-0 [HiFiBerry DAC+ HD HiFi pcm179x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

sudo i2cdetect -y 1

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

j-schambacher, 2022-02-22:
@apastuszak Thanks! I'll try to dig into it.

j-schambacher, 2022-02-22:
I can confirm the a.m. behaviour with the 5.15 64bit kernel: the modules are loaded, but show no activity: I can't observe any I2C bus communication (tested with logic analyzer) nor do printk-statement while in the driver-probe function leave any trace in the logs.

@pelwell any hint?

I'm meanwhile going to check with the 5.15 32 bit kernel.

pelwell, 2022-02-22:
What does `i2cdetect -y 1` show for you?

j-schambacher, 2022-02-22:
Just as above: 0x62, the PLL is detected, the DAC needs the PLL clock, which is not configured, therefore no function on the I2C and can't be detected. I wonder, why I can't see any activity on the I2C?

pelwell, 2022-02-22:
Do you see the i2cdetect scanning?

j-schambacher, 2022-02-22:
yep, that works as expected.

pelwell, 2022-02-22:
Your overlay (or rather the device tree it creates) is unusual in that there are two instances of the dacplushd compatible string - one as an I2C device and one as a random platform device. The I2C device is relies on the platform device for its clock.

One thing that has changes since 5.10 (if my memory of the timing is correct) is that the system does more to establish that the dependencies for a device node are ready before calling the probe function. If you remove the "clocks" declaration from the pll@62 node does it get into the probe function?

j-schambacher, 2022-02-22:
Ok, I'll try.

pelwell, 2022-02-22:
Also, how does the non-I2C/platform instance ever get instantiated since there is no platform driver probe function, only one for the I2C mode?

j-schambacher, 2022-02-23:
I fixed it by removing the "clocks" declaration from either parts, DAC and PLL. I'm preparing a PR for 5.15 to fix it.
Thanks to @apastuszak  and @pelwell

apastuszak, 2022-02-23:
@j-schambacher Thank you for fixing this!

pelwell, 2022-02-23:
Joerg's fix #4904 has been merged, and the updated overlay will appear in future kernel releases. In the meantime there's a fresh build here:
[hifiberry-dacplushd.zip](https://github.com/raspberrypi/linux/files/8125299/hifiberry-dacplushd.zip)

Install it with:
```
$ unzip hifiberry-dacplushd.zip
$ sudo cp hifiberry-dacplushd.dtbo /boot/overlays
```

apastuszak, 2022-02-23:
It's working, but I had to modify the config.txt. In 5.10 it worked without me needing to modify config.txt.

But that's on the vendor to provide instructions IMHO. Thank you!

pelwell, 2022-02-23:
If the HAT EEPROM includes the contents of the overlay (and not just its name) then you will have to continue to use `dtoverlay=` at the top of the config.txt and `dtoverlay=hifiberry-dacplushd` somewhere further down (unless HiFiBerry release details on how to re-flash the EEPROM to include contain the name).

j-schambacher, 2022-02-23:
Normally, we provide the entire DT-overlay in the EEPROM. We'll discuss internally how to handle this. I'll keep you posted.

force_eeprom_read=0

in config.txt may also help to avoid reading the HAT EEPROM.

mightyoakbob, 2022-11-10:
> Joerg's fix #4904 has been merged, and the updated overlay will appear in future kernel releases. In the meantime there's a fresh build here: [...]

Hi,
I'm pretty sure I have exactly the same issue. My Raspberry pi cannot 'see' my DAC2 HD, aplay -l shows no cards at all but I'm using the 32 bit Bullseye OS. Is this fix OK for 32 bit or is it 64bit only?
 
In /boot/config.txt I have killed both the HDMI audio and the jack with
`#dtparam=audio=on`
`dtoverlay=vc4-kms-v3d,noaudio`
And I've added.
`dtoverlay=hifiberry-dacplushd`
For all other HiFiBerry cards I use, those 3 changes are sufficient all be it with a different driver.

Do I also need:

`force_eeprom_read=0`
`dtparam=i2s=on`

Could someone explain what having "dtoverlay=" at the top of the file does? I don't understand that.

Thanks very much.

Bob.

mightyoakbob, 2022-11-10:
> To recap, the suggestion is that your config.txt should contain the following:
> 
> ```
> dtoverlay=
> dtoverlay=hifiberry-dacplushd
> # force_eeprom_read=0  This isn't necessary...
> # dtparam=i2s=on       ...and neither is this
> ```
That's brilliant I'm now confident on config.txt thanks so much Phil.
Also thanks for taking the time to explain so well.

I'm sorry to be a bit slow but could I / should I install this ...  
`$` unzip hifiberry-dacplushd.zip
`$` sudo cp hifiberry-dacplushd.dtbo /boot/overlays
on my **32bit** pi?

Hoping to try this out tomorrow night and over the weekend.

Thanks again.

Bob.

pelwell, 2022-11-10:
It's not necessary to install the overlay from the downloaded zip file unless you haven't updated your kernel in the last 6 months. If you were curious you could:
```
$ unzip hifiberry-dacplushd.zip
$ md5sum hifiberry-dacplushd.dtbo /boot/overlays/hifiberry-dacplushd.dtbo
```
I tried this on a Pi 4 running a recent kernel and it said:
```
483b267b241b5a74318284904395868f  hifiberry-dacplushd.dtbo
483b267b241b5a74318284904395868f  /boot/overlays/hifiberry-dacplushd.dtbo
```
i.e. the downloaded version is the same as the one already installed.

mightyoakbob, 2022-11-10:
> It's not necessary to install the overlay from the downloaded zip file unless you haven't updated your kernel in the last 6 months.

Thanks Phil.
What a star.

Cheers,
Bob.

closed 2022-11-10 by pelwell

reinisb, 2022-11-24:
Hi folks! Unfortunately, with Kernel 5.15 my HiFiberry Digi Pro 2 DAC stopped working as well. I'd appreciate any suggestions. Thanks!

> Raspberry Pi 4 Model B Rev 1.2
> Linux raspberry 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux

```
pi@raspberry:~ $ aplay -l
aplay: device_list:274: no soundcards found...
```

```
pi@raspberry:~ $ sudo nano /boot/config.txt

#dtparam=audio=on
#force_eeprom_read=0
dtoverlay=hifiberry-digi-pro

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio

```

pelwell, 2022-11-25:
Read my answers to @mightyoakbob.

mightyoakbob, 2022-11-25:
I can confirm that Phil's advice does work for my DAC which is a HiFiBerry Dac2 HD.
My Dac requires 4 changes to /boot/config.txt

```
dtoverlay=
## dtparam=audio=on
dtoverlay=vc4-kms-v3d,noaudio
dtoverlay=hifiberry-dacplushd
```
I made "dtoverlay=" the very first line in the file and then the DAC works fine.

This DAC also has a hardware valume control built into it. Some music players have this working but unfortunately not with mpd so the only volume control is the software one. That kind od defeats the object of having a high quality dac. No one in the Linux community seems to want to look at it and I know HiFiBerry have offered support to anyone willing to have a go at this.

Cheers,

Bob.

reinisb, 2022-11-25:
I tried everything but my [HiFiberry Digi2 Pro (HW 2.1)](https://www.hifiberry.com/shop/boards/hifiberry-digi2-pro/) is simply not recognized in 5.15 on my RPi4 Rev 1.2.

```
dtoverlay=
#dtparam=audio=on
dtoverlay=vc4-kms-v3d,noaudio
dtoverlay=hifiberry-digi-pro
```

```
pi@raspberry:~ $ aplay -l
aplay: device_list:274: no soundcards found...
```

cc @j-schambacher

pelwell, 2022-11-25:
Does `sudo vcdbg log msg` show anything suspicious?

pelwell, 2022-11-25:
BTW, ideally you would have opened a new issue for this, since your card isn't a DAC 2 HD HAT.

reinisb, 2022-11-25:
@pelwell I've created a dedicated issue https://github.com/raspberrypi/linux/issues/5256 Moreover, `sudo vcdbg log msg` does not work on my 64-bit system.

6by9, 2022-11-25:
https://github.com/raspberrypi/utils/tree/master/vclog should work on both 32 and 64 bit systems to read the firmware logs.