raspberrypi/linux #4899: rpi-5.15.y arm tools too old

zhangbinghua, 2022-02-18:
### Describe the bug

I use this tools:
[raspberrypi/tools](https://github.com/raspberrypi/tools)/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian

To build the kernel rpi-5.15.y, error compiling the kernel.
It means that the toolchain is too old.
***
*** Compiler is too old.
***   Your GCC version:    4.8.3
***   Minimum GCC version: 5.1.0
***

How can i get the newest toolchain? Thank you very much
my email: 920052390@qq.com

### Steps to reproduce the behaviour

sky@sky-virtual-machine:<linux (rpi-5.15.y)>$ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (crosstool-NG linaro-1.13.1-4.8-2014.01 - Linaro GCC 2013.11) 4.8.3 20140106 (prerelease)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Step 1:
KERNEL=kernel7l
Step 2:
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig

sky@sky-virtual-machine:<linux (rpi-5.15.y)>$ make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
***
*** Compiler is too old.
***   Your GCC version:    4.8.3
***   Minimum GCC version: 5.1.0
***
scripts/Kconfig.include:44: Sorry, this compiler is not supported.
scripts/kconfig/Makefile:94: recipe for target 'bcm2711_defconfig' failed
make[1]: *** [bcm2711_defconfig] Error 1
Makefile:616: recipe for target 'bcm2711_defconfig' failed
make: *** [bcm2711_defconfig] Error 2

How can i get the newest toolchain? Thank you very much

### Device (s)

Raspberry Pi 4 Mod. B

### System

sky@sky-virtual-machine:<linux (rpi-5.15.y)>$ uname -a
Linux sky-virtual-machine 4.15.0-142-generic #146~16.04.1-Ubuntu SMP Tue Apr 13 09:27:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

popcornmix, 2022-02-18:
See the readme here: https://github.com/raspberrypi/tools

closed 2022-02-18 by popcornmix

zhangbinghua, 2022-02-19:
I got it. Now I update the ubuntu toolchain by this way: