raspberrypi/linux #4900: How to enabled the earlycon on pi4?

zhangbinghua, 2022-02-19:
### Describe the bug

When i enabled the earlycon by this, my pi4 can't boot.
If i disabled the earlycon, my uart console works properly.

I had followed by this url: https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts
earlycon=uart8250,mmio32,0xfe215040
earlycon=pl011,mmio32,0xfe201000

I've tried both above. I make sure my serial console is working properly，when i had't add the earlycon=pl011,mmio32,0xfe201000 or earlycon=uart8250,mmio32,0xfe215040 in the cmdline.txt.
How could i do?

### Steps to reproduce the behaviour

### Methods A:
If the cmdline configure like this, my console can work properly.
pi@raspberrypi-sky:~$ cat /proc/cmdline 
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CC:77:45 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=2d91291d-02 rootfstype=ext4 fsck.repair=yes rootwait

### Methods B:
If the cmdline was configured like this, my pi4 can't boot.
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CC:77:45 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 earlycon=uart8250,mmio32,0xfe215040 root=PARTUUID=2d91291d-02 rootfstype=ext4 fsck.repair=yes rootwait

### Methods C:
If the cmdline was configured like this, my pi4 can't boot.
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CC:77:45 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 earlycon=pl011,mmio32,0xfe201000 root=PARTUUID=2d91291d-02 rootfstype=ext4 fsck.repair=yes rootwait

Please help me to set earlycon right.
Thank you very much.

### Device (s)

Raspberry Pi 4 Mod. B

### System

pi@raspberrypi-sky:~$ uname  -a
Linux raspberrypi-sky 5.15.21-v7l+ #2 SMP Sun Feb 20 01:20:16 CST 2022 armv7l GNU/Linux

### Logs

pi@raspberrypi-sky:~$ ls /dev/serial* -lh
lrwxrwxrwx 1 root root 7 Feb 20 01:17 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Feb 20 01:17 /dev/serial1 -> ttyS0

zhangbinghua, 2022-02-19:
>

zhangbinghua, 2022-02-19:
some configuration about my congfig.txt
[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
#kernel=kernel7l-sky.img
#kernel=kernel-15.img
kernel=kernel-sky2.img
enable_uart=1
#dtoverlay=pi3-miniuart-bt
dtoverlay=disable-bt

zhangbinghua, 2022-02-19:
sky@sky-virtual-machine:<linux (rpi-5.15.y)>$ git remote -vv
origin	https://github.com/raspberrypi/linux.git (fetch)
origin	https://github.com/raspberrypi/linux.git (push)

zhangbinghua, 2022-02-19:
I think i need to make menuconfig about the kernel, DEBUG_LL and CONFIG_UNCOMPRESS_INCLUDE

zhangbinghua, 2022-02-19:
Should i choose this Kernel low-level debugging port (Kernel low-level debugging on BCM2836 PL011 UART) ?
Beacuse i want to enable the Uncompressing Linux... done, booting the kernel. How should i do?

zhangbinghua, 2022-02-19:
Now , i had this cmdline:
pi@raspberrypi-sky:~$ cat /proc/cmdline 
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CC:77:45 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 earlycon=pl011,mmio32,0xfe201000 root=PARTUUID=2d91291d-02 rootfstype=ext4 fsck.repair=yes rootwait

the uart console's dmesg is :

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.21-v7l+ (sky@sky-virtual-machine) (arm-linux-gnueabihf-gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609, GNU ld (GNU Binutils for Ubuntu) 2.26.1) #5 SMP Sun Feb 20 04:15:16 CST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] random: fast init done
[    0.000000] earlycon: pl11 at MMIO32 0x00000000fe201000 (options '')
[    0.000000] printk: bootconsole [pl11] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] percpu: Embedded 16 pages/cpu s35788 r8192 d21556 u65536
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2059584
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:CC:77:45 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 earlycon=pl011,mmio32,0xfe201000 root=PARTUUID=2d91291d-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000012400000-0x0000000016400000] (64MB)
[    0.000000] Memory: 7758768K/8245248K available (10240K kernel code, 1389K rwdata, 3284K rodata, 2048K init, 593K bss, 158800K reserved, 327680K cma-reserved, 7458816K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 35271 entries in 104 pages
[    0.000000] ftrace: allocated 104 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 25)
[    0.000000] random: get_random_bytes called from start_kernel+0x55c/0x754 with crng_init=1
[    0.000001] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.016820] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.026291] bcm2835: system timer (irq = 26)
[    0.031281] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.037458] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.048382] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.056490] Switching to timer-based delay loop, resolution 18ns
[    0.062883] Console: colour dummy device 80x30
[    0.067401] printk: console [tty1] enabled
[    0.071560] printk: bootconsole [pl11] disabled
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.21-v7l+ (sky@sky-virtual-machine) (arm-linux-gnueabihf-gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609, GNU ld (GNU Binutils for Ubuntu) 2.26.1) #5 SMP Sun Feb 20 04:15:16 CST 2022
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] random: fast init done
[    0.000000] earlycon: pl11 at MMIO32 0x00000000fe201000 (options '')
[    0.000000] printk: bootconsole [pl11] enabled
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[... cut ...]
[    5.196851] systemd[1]: Finished Load Kernel Modules.
[    5.210166] systemd[1]: Mounting FUSE Control File System...
[    5.234462] systemd[1]: Mounting Kernel Configuration File System...
[    5.248184] systemd[1]: Started File System Check Daemon to report status.
[    5.269847] systemd[1]: Starting Apply Kernel Variables...
[    5.311249] systemd[1]: Mounted FUSE Control File System.
[    5.317648] systemd[1]: Mounted Kernel Configuration File System.

if i enabled the DEBUG_LL and CONFIG_UNCOMPRESS , the pi4 can't boot up.

pelwell, 2022-02-19:
The problem is that the earlycon declaration doesn't enable the correct function on the GPIO pins. Just add `enable_uart=1` in config.txt.

zhangbinghua, 2022-02-19:
@pelwell 
some configuration about my congfig.txt
[pi4]
arm_boost=1

[all]
kernel=kernel-sky2.img
enable_uart=1
dtoverlay=disable-bt

Now i had boot up by earlycon=pl011,mmio32,0xfe201000 in the cmdline.
I want to print "Uncompressing Linux... done, booting the kernel."
How should i do?

pelwell, 2022-02-19:
As the kernel log shows, earlycon is enabled. I think, however, that the kernel decompression happens before the earlycon (and probably the whole kernel log) is initialised, which would explain why you are not able to see that message.

closed 2022-03-14 by pelwell