raspberrypi/linux #4902: mcp251xfd based dual can HAT and 5.10.95

nmbath, 2022-02-21:
### Describe the bug

The HAT has a dual channel mcp251xfd can chip, running on spi0.0 and spi1.0. While the driver loads and looks like it is working a number of issues are occurring,. A dmesg | grep can shows

[    7.661604] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    7.724663] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.

When trying to send and receive can messages on can0 the following dmesg messages are received for a few seconds and then stops, but traffic is sent and received on can0

[   23.398211] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.501499] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.581290] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.607407] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.619993] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.629911] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.653014] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.662845] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   23.673721] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!

However I can not get can1 running at all. 

ifconfig can1 up
ifconfig: SIOCSIFFLAGS: Invalid argument

dmesg has these messages
[  394.045735] mcp251xfd spi1.0 can1: Failed to detect MCP2518FD (osc=0x00000000).
[  394.053266] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.064157] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.075031] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.085798] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000).
[  394.096366] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.107287] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.118507] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.129372] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000).
[  394.139661] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.150838] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.161713] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.172492] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000).
[  394.182735] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.193606] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.204483] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[  394.215283] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000).

I have installed various fixes and patches as baed on https://github.com/marckleinebudde/linux-rpi/commits/v5.10-rpi/backport-performance-improvements

### Steps to reproduce the behaviour

Stock 5.10.95 pulled from https://github.com/raspberrypi/linux/tree/rpi-5.10.y and the can0 interface presented some crc error, which after installing this patch 0ecb42a4438c2e13c3d1ee058038c104c6419479 can: mcp251xfd: mcp251xfd_regmap_crc_read(): work around broken CRC on TBC register, stopped giving them.

### Device (s)

Raspberry Pi 3 Mod. B

### System

Linux raspberrypi2 5.10.95-v7 #1 SMP Fri Feb 18 07:45:17 UTC 2022 armv7l GNU/Linux

[    0.080092] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-01-20T13:58:22, variant start
[    0.090104] raspberrypi-firmware soc:firmware: Firmware hash is bd88f66f8952d34e4e0613a85c7a6d3da49e13e2

pelwell, 2022-02-21:
I'm going to leave this open for others to contribute to, but without a dual mcp251xfd HAT and some CAN hardware it's not going to get any attention from me. Have you tried asking @marckleinebudde, whose repo you linked to?

marckleinebudde, 2022-02-21:
Hey @nmbath,

the `mcp251xfd` driver in linux-v5.10 should work with your hardware, my branch is only performance improvements. If you use a newer raspi kernel, the improvements are already included.

BTW the problem:
```
[ 23.398211] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #8!!!
``` 
is fixes with v5.15.

Coming back to your problem:
```
mcp251xfd spi1.0 can1: Failed to detect MCP2518FD (osc=0x00000000).
```
The driver only reads `0x0`, i.e. none of the bits set when reading from `spi1.0`. Your analysis that the chip select might not be working looks good to me. Do you have a scope to check that? Can you post here you DT overlay configuration?

The `mcp251xfd` driver uses more resources, as the chip is more complex, it needs more SPI transfers per received CAN message, than the `mcp251x` driver.

nmbath, 2022-02-21:
Hi @marckleinebudde, Thank you for your response. The HAT is a waveshare dual mcp251xfd and understand that it uses BCM pin 7 for CS. I have used their default 2xMCP2517FD.dtbo

/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835", "brcm,bcm2836", "brcm,bcm2708", "brcm,bcm2709";
    /* disable spi-dev for spi0.0 */

    fragment@0{
	target = <&spidev0>;
	__overlay__ {
	    status = "disabled";
	};
    };

    fragment@1{
	target = <&spidev1>;
	__overlay__ {
	    status = "disabled";
	};
    };

    /* the interrupt pin of the can-controller */
    fragment@2 {
        target = <&gpio>;
        __overlay__ {
            can_int_pins: can_int_pins {
                brcm,pins = <25 16>;
                brcm,function = <0>; /* input */
            };
	    spi1_pins: spi1_pins {
			brcm,pins = <19 20 21>;
			brcm,function = <3>; /* alt4 */
	    };

	   spi1_cs_pins: spi1_cs_pins {
			brcm,pins = <26>;
			brcm,function = <1>; /* output */
	    };

        };
    };

    /* the clock/oscillator of the can-controller */
    fragment@3 {
        target-path = "/clocks";
        __overlay__ {
            /* external 40M oscillator of mcp2517fd on SPI0.0 */
            mcp2517fd_osc: mcp2517fd_osc {
                compatible = "fixed-clock";
                #clock-cells = <0>;
              clock-frequency  = <40000000>;
            };
        };
    };

    /* the spi config of the can-controller itself binding everything together */
    fragment@4 {
        target = <&spi0>;
        __overlay__ {
            /* needed to avoid dtc warning */
            #address-cells = <1>;
            #size-cells = <0>;

          status = "okay";
            can0: can@0 {
                reg = <0>;
                pinctrl-names = "default";
         	pinctrl-0 = <&can_int_pins>;
                compatible = "microchip,mcp2517fd";
                spi-max-frequency = <20000000>;
                interrupt-parent = <&gpio>;
                interrupts = <25 8>; /* IRQ_TYPE_LEVEL_LOW */
                clocks = <&mcp2517fd_osc>;
            };

        };
    };
    /* the spi config of the can-controller itself binding everything together */
    fragment@5 {
        target = <&spi1>;
        __overlay__ {
            /* needed to avoid dtc warning */
            #address-cells = <1>;
            #size-cells = <0>;

	    cs-gpios = <&gpio 26 1>;
 	    status = "okay";
            can1: can@1 {
                reg = <0>;
                pinctrl-names = "default";
         	pinctrl-0 = <&spi1_pins &spi1_cs_pins>;
                compatible = "microchip,mcp2517fd";
                spi-max-frequency = <20000000>;
                interrupt-parent = <&gpio>;
                interrupts = <16 8>; /* IRQ_TYPE_LEVEL_LOW */
                clocks = <&mcp2517fd_osc>;
            };
	};
    };
};

Looking at this file they seem to be using GPIO26 for CS on the 2nd unit. I am now in an area I have not done much work on, and I do not have a scope to look at what is going on. GPIO26 is on pin 37, but reading their material CS_1 is physical pin 26, or BCM 7. 

I tried changing the CS from 26 to 7, but then I got 

[    6.118410] pinctrl-bcm2835 3f200000.gpio: pin gpio7 already requested by 3f204000.spi; cannot claim for spi1.0
[    6.128614] pinctrl-bcm2835 3f200000.gpio: pin-7 (spi1.0) status -22
[    6.128647] mcp251xfd spi1.0: Error applying setting, reverse things back
[    6.129724] mcp251xfd: probe of spi1.0 failed with error -22

It is probably something very simple, I just cant see it and don't have enough experience with GPIO's!

[image]

marckleinebudde, 2022-02-21:
Ahhh waveshare - adding @pdp7 to the loop.
Drew, long time no see, but do have you got a working Raspi DT Overlay for the waveshare mcp251xfd board?

pelwell, 2022-02-21:
You can release GPIO 7 by adding `dtoverlay=spi0-1cs` to config.txt.

pelwell, 2022-02-21:
Once you have something that works I can help you tidy it up and get it into the standard kernel builds.

marckleinebudde, 2022-02-21:
Once it's working generate a "compiled" overlay with something like this:

```
ovmerge -c spi1-1cs-overlay.dts,cs0_pin=18,cs0_spidev=false mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi1-0,interrupt=24
```

See header of this file:

https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/seeed-can-fd-hat-v1-overlay.dts

pelwell, 2022-02-21:
Except with `,cs0_pin=7` and `interrupt=16`, i.e.
```
ovmerge -c spi1-1cs-overlay.dts,cs0_pin=7,cs0_spidev=false mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi1-0,interrupt=16
```

marckleinebudde, 2022-02-21:
ACK, that was a copy/paste from the `seeed-can-fd-hat-v1-overlay.dts` overlay, that needs to be adjusted. (Updated my comment for clarification).

pelwell, 2022-02-21:
And you can include the `spi0-1cs` in the same ovmerge command if you want.

nmbath, 2022-02-21:
I have tried adding spi0-1cs, but didn't seem to do anything and still have the error. Will try again in the morning.

nmbath, 2022-02-22:
I have just tried this configuration:

```
dtoverlay=spi0-1cs,cs0_pin=8
dtoverlay=spi1-1cs,cs0_pin=7,cs0_spidev=false
dtoverlay=mcp251xfd,spi1-0,interrupt=16
dtoverlay=mcp251xfd,spi0-0,interrupt=25

```

And had this..

```
[    5.872462] spi_master spi1: will run message pump with realtime priority
[    5.917707] mcp251xfd spi1.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).
[    5.996127] spi_master spi0: will run message pump with realtime priority
[    6.014647] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
```

Digging through Waveshares details, that is correct. Reading the wiki site again, they have two modes, both sharing spi0 and one on spi0 and the other on spi1. The default mode is apparently separate SPI's. They have a different table on the wiki and that looks like when sharing an SPI the CS for can1 is 7 and when separate SPI's is 26. So changed the configuration too

```
dtoverlay=spi0-1cs,cs0_pin=8
dtoverlay=spi1-1cs,cs0_pin=26,cs0_spidev=false
dtoverlay=mcp251xfd,spi1-0,interrupt=16
dtoverlay=mcp251xfd,spi0-0,interrupt=25

```

and got this..

```
[    5.650436] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    5.699704] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[   21.112168] can: controller area network core

```

ip link set can1 type can bitrate 250000
ifconfig can1 up
ifconfig: SIOCSIFFLAGS: No such device

and

```
[   77.044505] mcp251xfd spi1.0 can1: Failed to detect MCP2518FD (osc=0x00000000).
[   77.052125] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.062943] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.073793] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.084553] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000).
[   77.094659] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.105529] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.116342] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.127083] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000).
[   77.137193] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.148027] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.158840] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.169581] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000).
[   77.179617] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.190433] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.201236] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   77.211987] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000).

```

So maybe it's not the overlay but something else.

Just to double check, I tried this to make sure it was not on the other mode!

```
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi0-1,interrupt=16
```

and got

```
[    5.975545] spi_master spi0: will run message pump with realtime priority
[    6.021352] mcp251xfd spi0.1 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).
[    6.085145] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.

```

Thoughts?

marckleinebudde, 2022-02-22:
> ```
> [    5.650436] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
> [ [...]

This mean both mcp251xfd are properly detected. But later, during ifup you got this error messages?

```
[   77.044505] mcp251xfd spi1.0 can1: Failed to detect MCP2518FD (osc=0x00000000).
[   77.052125] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000) retrying.
```

Strange. The driver only reads `0x0` from the device. Seems the mcp2518fd is not sending anymore. Can you attach a full kernel log.

nmbath, 2022-02-22:
full dmesg:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.95-v7 (oe-user@oe-host) (arm-ve-linux-gnueabi-gcc (GCC) 9.3.0, GNU ld (GNU Binutils) 2.34.0.20200220) #1 SMP Fri Feb 18 07:45:17 UTC 2022
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 878792K/970752K available (10240K kernel code, 1312K rwdata, 2980K rodata, 1024K init, 862K bss, 26424K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32069 entries in 95 pages
[    0.000000] ftrace: allocated 94 pages with 5 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3b0/0x594 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000321] Console: colour dummy device 80x30
[    0.000388] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000424] pid_max: default: 32768 minimum: 301
[    0.000633] LSM: Security Framework initializing
[    0.000878] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000909] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002317] cgroup: Disabling memory control group subsystem
[    0.002604] CPU: Testing write buffer coherency: ok
[    0.003106] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004328] Setting up static identity map for 0x100000 - 0x10003c
[    0.004519] rcu: Hierarchical SRCU implementation.
[    0.005446] smp: Bringing up secondary CPUs ...
[    0.006616] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007907] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009096] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009252] smp: Brought up 1 node, 4 CPUs
[    0.009296] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009314] CPU: All CPU(s) started in HYP mode.
[    0.009330] CPU: Virtualization extensions available.
[    0.010429] devtmpfs: initialized
[    0.028694] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.028947] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.028984] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.031978] pinctrl core: initialized pinctrl subsystem
[... 300 lines cut ...]
[   62.439521] mcp251xfd spi1.0 can1: CRC read error at address 0x0e0c (length=4, data=00 00 00 00, CRC=0x0000).
[   62.450228] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   62.461811] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   62.473109] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[... cut ...]
[   62.554691] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   62.566077] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000) retrying.
[   62.576826] mcp251xfd spi1.0 can1: CRC read error at address 0x0000 (length=4, data=00 00 00 00, CRC=0x0000).
```

Logged in at roughly 33.x

Issued the following command, which triggered the log messages at 62.x

```
root@raspberrypi2:~#  ip link set can1 type can bitrate 250000
root@raspberrypi2:~# ifconfig can1 up
ifconfig: SIOCSIFFLAGS: No such device
```

marckleinebudde, 2022-02-22:
Which overlays are you using in this bootlog? This looks like the waveshare overlays? Which chips are on your waveshare hat? mcp2517fd or mcp2518fd?

The driver says the firmware (the device tree + overlays) are specifying a mcp2517fd, but the driver has detected the mcp2518fd.

```
[    5.940362] mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Detected MCP2518FD, but firmware specifies a MCP2517FD. Fixing up.
[    5.983905] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    6.005599] spi_master spi1: will run message pump with realtime priority
[    6.021405] mcp251xfd spi1.0 (unnamed net_device) (uninitialized): Detected MCP2518FD, but firmware specifies a MCP2517FD. Fixing up.
[    6.049346] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
```

What's your `config.txt`?
Can you past the output of:

```
dtc -I fs /proc/device-tree
```

Later in the boot log we see these error messages:
```
[   19.012973] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
```
This indicates that one driver instance is sending CAN messages to the Linux networking stack:

- Is can0 attached to a CAN bus?
- Is can1 attached to a CAN bus?
- Is the can0 interface automatically configured and brought up?
- Is the can1 interface automatically configured and brought up?

The question is, what happens between the interface is properly detected:
```
[    6.049346] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
```
and 
```
[   62.398501] mcp251xfd spi1.0 can1: Failed to detect MCP2518FD (osc=0x00000000).
```

After the driver detects the chip, the chip is shut down again. When you issue a `ip link set up` the chip is reset the driver is polling the OSC register to show the reset defaults, but register only shows `0x0`. The question is why.

nmbath, 2022-02-22:
@marckleinebudde,

Yes I am using the waveshare overlay, and this is my full config.txt

```
kernel=u-boot.bin

[pi2]
device_tree=bcm2709-rpi-2-b.dtb

[pi3]
device_tree=bcm2710-rpi-3-b.dtb
dtoverlay=miniuart-bt
core_freq=400
core_freq_min=400

[pi3+]
device_tree=bcm2710-rpi-3-b-plus.dtb
dtoverlay=miniuart-bt
core_freq=400
core_freq_min=400

[all]
#dtparam=spi=on
dtoverlay=gpio-fan,gpiopin=5,temp=55000

#dtoverlay=spi0-1cs,cs0_pin=8
#dtoverlay=spi1-1cs,cs0_pin=26,cs0_spidev=false
#dtoverlay=mcp251xfd,spi0-0,interrupt=25
#dtoverlay=mcp251xfd,spi0-1,interrupt=16

#dtoverlay=seeed-can-fd-hat-v1
dtoverlay=2xMCP2517FD
#dtoverlay=mcp251xfd,spi0-0,interrupt=25
#dtoverlay=mcp251xfd,spi1-0,interrupt=16
```

can0 is automatically configured at 250kb and starte, it is connected to a can bus, with a device running `cangen can0 -g 1 -p 10 -i -x `

can1 has nothing attached to it at present and no automatic configuration set. I can connect it to an active port, but had not until I actually get the device working.

[dtc.txt](https://github.com/raspberrypi/linux/files/8116929/dtc.txt)

marckleinebudde, 2022-02-22:
Which chips are on your waveshare hat? mcp2517fd or mcp2518fd?

So let's recap:
- you have two mcp2518fd (probably 18)
- one is connected to spi0 CS0
- one is connected to spi1 CS0

The questions are:
- which GPIO is used for spi0 CS0
- which GPIO is used for spi1 CS0
- which GPIO is used for IRQ on chip attached to spi0 CS0
- which GPIO is used for IRQ on chip attached to spi1 CS0

nmbath, 2022-02-22:
I have looked at the CHIPs on the HAT and they are labelled MCP2518FD, which aligns with their [Web site](https://www.waveshare.com/2-ch-can-fd-hat.htm)

Also on their web site in the [Wiki section ](https://www.waveshare.com/wiki/2-CH_CAN_FD_HAT)is the attached table. This would imply that the details are

spi0 CS0 is 8
spi1 CS0 is 26
IRQ on spi0 CS0 is 25
IRQ on spi1 CS0 is 16

[image]

When configured like this I get 

```
[   11.628186] mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Detected MCP2518FD, but firmware specifies a MCP2517FD. Fixing up.
[   11.649036] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[   11.692113] mcp251xfd spi1.0 (unnamed net_device) (uninitialized): Detected MCP2518FD, but firmware specifies a MCP2517FD. Fixing up.
[   11.707525] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.

```

With any other configuration of interrupt or cs0 I fail to get one or the other to initialise.

nmbath, 2022-02-22:
I have managed to compile the mcp25xxfd driver from within the source code on the [waveshare wiki](https://www.waveshare.com/wiki/File:2-CH-CAN-FD-HAT-Demo.7z)

Once I `rmmod mcp251xfd` and `modprobe mcp25xxfd` I managed to get both can0 and can1 working

```
root@raspberrypi2:~# modprobe mcp25xxfd
root@raspberrypi2:~# ip link set can1 type can bitrate 250000
root@raspberrypi2:~# ifconfig can1 up
root@raspberrypi2:~# candump can1

```

No traffic as the can port is not connected, I also got this in `dmesg [ 1341.887415] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready`

It does seem to be a code/driver issue as opposed to overlay file. I will dig through and see if I can spot where the code differs between the example waveshare has provided and the based 5.10.y code.

Also no `NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!` errors so they may have a hybrid between 5.10 and 5.15 code bases!

marckleinebudde, 2022-02-22:
If you don't receive any CAN messages you'll not get the `NOHZ tick-stop error` error. That's a very old driver, even before it got mainline.

marckleinebudde, 2022-02-22:
I found something interesting:

```
 #define MCP25XXFD_POLL_SLEEP_US (10)
-#define MCP25XXFD_POLL_TIMEOUT_US (USEC_PER_MSEC)
+#define MCP25XXFD_POLL_TIMEOUT_US 500000
 #define MCP25XXFD_SOFTRESET_RETRIES_MAX 3
 #define MCP25XXFD_READ_CRC_RETRIES_MAX 3
```

Waveshare increased the timeout from 1 ms to 500ms. Can you increase the timeout and try the mainline driver?

pelwell, 2022-02-22:
In case I forget later, thank you for looking into this, Marc.

marckleinebudde, 2022-02-23:
I strongly suggest to use the mainline driver, as the waveshare driver is old and contains several bugs, that have been fixed on mainline.

pelwell, 2022-02-23:
It's unfortunate that in mcp251xfd_chip_clock_enable the mainline driver checks for the validity of the value read from the OSC register _before_ checking if the read timed out. Presumably it has been silently timing out, and the increased timeout value makes it work.

nmbath, 2022-02-23:
@marckleinebudde, I would also like to a dd my thanks for your support.

> I strongly suggest to use the mainline driver, as the waveshare driver is old and contains several bugs, that have been fixed on mainline.

I agree that going with mainline is the correct approach. Wanted to test what waveshare supplied to try and help identify the problem. Looks like we have found a code issue rather than overlay!

> Waveshare increased the timeout from 1 ms to 500ms. Can you increase the timeout and try the mainline driver?

I have applied the following

```
diff --git a/drivers/net/can/spi/mcp251xfd/mcp251xfd.h b/drivers/net/can/spi/mcp251xfd/mcp251xfd.h
index cb6398c2a560..d69b3d4de65b 100644
--- a/drivers/net/can/spi/mcp251xfd/mcp251xfd.h
+++ b/drivers/net/can/spi/mcp251xfd/mcp251xfd.h
@@ -398,7 +398,7 @@
 #define MCP251XFD_OSC_STAB_SLEEP_US (3 * USEC_PER_MSEC)
 #define MCP251XFD_OSC_STAB_TIMEOUT_US (10 * MCP251XFD_OSC_STAB_SLEEP_US)
 #define MCP251XFD_POLL_SLEEP_US (10)
-#define MCP251XFD_POLL_TIMEOUT_US (USEC_PER_MSEC)
+#define MCP251XFD_POLL_TIMEOUT_US 500000
 #define MCP251XFD_SOFTRESET_RETRIES_MAX 3
 #define MCP251XFD_READ_CRC_RETRIES_MAX 3
 #define MCP251XFD_ECC_CNT_MAX 2
```
 The outcome was:

```
root@raspberrypi2:~# ip link set can1 type can bitrate 250000
root@raspberrypi2:~# ifconfig can1 up
ifconfig: SIOCSIFFLAGS: No such device
root@raspberrypi2:~# 

```

Is there any debugging within the mcp251xfd driver that can be enabled to give more detail.

pelwell, 2022-02-23:
You can add `#define DEBUG` at the top of `mcp241xfd-core.c` and rebuild the module. That will enable the limited debugging.

pelwell, 2022-02-23:
The next step would be to enable the kernel event tracing for the duration of the `ifup`, something like (running as root):
```
# echo 1 > /sys/kernel/debug/tracing/events/spi/enable
# ifup
# echo 0 > /sys/kernel/debug/tracing/events/spi/enable
# cp /sys/kernel/debug/tracing/trace trace.txt
```

nmbath, 2022-02-23:
Sorry for the delay, been a hectic day!

I uploaded  trace as requested

[trace.txt](https://github.com/raspberrypi/linux/files/8126744/trace.txt)

nmbath, 2022-02-23:
I have installed raspy-gpio on the device. I am not sure if this is a red hearing or not, but according to the tool it dosnt look like the SPI is properly configured in terms of PINs.

```
root@raspberrypi2:~# ./raspi-gpio get
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 func=INPUT
GPIO 1: level=1 func=INPUT
GPIO 2: level=1 func=INPUT
GPIO 3: level=1 func=INPUT
GPIO 4: level=0 alt=0 func=GPCLK0
GPIO 5: level=0 func=OUTPUT
GPIO 6: level=1 func=INPUT
GPIO 7: level=1 func=OUTPUT
GPIO 8: level=1 func=OUTPUT
GPIO 9: level=0 alt=0 func=SPI0_MISO
GPIO 10: level=0 alt=0 func=SPI0_MOSI
GPIO 11: level=0 alt=0 func=SPI0_SCLK
GPIO 12: level=0 func=INPUT
GPIO 13: level=0 func=INPUT
GPIO 14: level=1 alt=0 func=TXD0
GPIO 15: level=1 alt=0 func=RXD0
GPIO 16: level=1 func=INPUT
GPIO 17: level=0 func=INPUT
GPIO 18: level=0 alt=0 func=PCM_CLK
GPIO 19: level=0 func=INPUT
GPIO 20: level=0 alt=4 func=SPI1_MOSI
GPIO 21: level=0 func=OUTPUT
GPIO 22: level=0 func=INPUT
GPIO 23: level=0 func=INPUT
GPIO 24: level=0 func=INPUT
GPIO 25: level=1 func=INPUT
GPIO 26: level=0 func=INPUT
GPIO 27: level=0 func=INPUT
```

There is no SPI1_MISO on pin 19 and SPIO_SCLK on pin 21

nmbath, 2022-02-23:
@pelwell, 

You suggested

> You can add #define DEBUG at the top of mcp241xfd-core.c and rebuild the module. That will enable the limited debugging. 

I add this to my build:

```
diff --git a/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c b/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c
index 5f2694f5d9c0..82b9bd354134 100644
--- a/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c
+++ b/drivers/net/can/spi/mcp251xfd/mcp251xfd-core.c
@@ -11,6 +11,7 @@
 //
 // Copyright (c) 2019 Martin Sperl <kernel@martin.sperl.org>
 //
+#define DEBUG
 
 #include <linux/bitfield.h>
 #include <linux/clk.h>
```

Having done this no extra messages are displayed. Do a grep for DEBUG in the mcp151xfd folder, there are no other occurrences of DEBUG.

pelwell, 2022-02-23:
I think something is corrupting your pin alternate function settings after boot. If you compare the two traces you see that both successfully receive some non-zero data over SPI (spi0 and spi1), but then the spi1 trace changes to only receiving zeroes.

Please run `raspi-gpio get` as soon after boot as you can. If you have multiple windows or ssh sessions, run `watch -n 1 raspi-gpio get 7-21` in one of the sessions.

Then run the `ifup`, check the results and see if the pin configuration has changed.

The `#define DEBUG` should change the definition of `netdev_dbg` to a call to `netdev_printk(KERN_DEBUG, ...` (https://elixir.bootlin.com/linux/latest/source/include/linux/netdevice.h#L5335). Have you tried `dmesg -l debug`?

nmbath, 2022-02-23:
@pelwell, @marckleinebudde 

 Thank you for your support. It would seem that another part of the system I am working on was claiming 3 GPIO ports for a different function. As soon as I disabled that everything works!

```
root@raspberrypi2:~# ip link set can0 type can bitrate 250000
root@raspberrypi2:~# ip link set can1 type can bitrate 250000
root@raspberrypi2:~# ifconfig can0 up
root@raspberrypi2:~# ifconfig can1 up

```

Log and drawn out, but appreciate your support, and glad it wasn't a code issue. Strange that I managed to get it working on the old waveshare driver.

pelwell, 2022-02-23:
Thanks again, Marc - shame it was a wild goose chase.

closed 2022-02-23 by pelwell

marckleinebudde, 2022-02-23:
FYI: I ordered a waveshare board, to compile a combined overlay and send a PR.

nmbath, 2022-02-23:
@marckleinebudde I have a combined overlay should you want it. In fact I am using the one from the Waveshare zip file.

marckleinebudde, 2022-02-23:
I was thinking about a proper one, see  https://github.com/raspberrypi/linux/issues/4902#issuecomment-1047216891

nmbath, 2022-02-23:
@marckleinebudde created the attached using ovmerge. compiled the result and used it....

However, received the following:

```
[   43.899882] irq 199: nobody cared (try booting with the "irqpoll" option)
[   43.906692] CPU: 0 PID: 1373 Comm: irq/199-spi1.0 Tainted: G         C        5.10.95-v7 #1
[   43.915041] Hardware name: BCM2835
[   43.918441] Backtrace: 
[   43.920913] [<80a02fd8>] (dump_backtrace) from [<80a0335c>] (show_stack+0x20/0x24)
[   43.928488]  r7:ffffffff r6:00000000 r5:600f0193 r4:80fe5ef0
[   43.934156] [<80a0333c>] (show_stack) from [<80a07604>] (dump_stack+0xc8/0xf4)
[   43.941387] [<80a0753c>] (dump_stack) from [<80a04748>] (__report_bad_irq+0x4c/0xd0)
[   43.949137]  r10:ffffe000 r9:82b73e6c r8:814dd000 r7:000000c7 r6:82b73e6c r5:00000000
[   43.956967]  r4:82b73e00 r3:80f05008
[   43.960552] [<80a046fc>] (__report_bad_irq) from [<80189790>] (note_interrupt+0x274/0x2c0)
[   43.968823]  r9:8da26000 r8:814dd000 r7:000000c7 r6:82b73e6c r5:00000000 r4:82b73e00
[   43.976574] [<8018951c>] (note_interrupt) from [<80185ab4>] (handle_irq_event_percpu+0x94/0x9c)
[   43.985279]  r10:ffffe000 r9:8da26000 r8:814dd000 r7:00000000 r6:82b73e6c r5:00000002
[   43.993109]  r4:82b73e00 r3:00000000
[   43.996693] [<80185a20>] (handle_irq_event_percpu) from [<80185b10>] (handle_irq_event+0x54/0x78)
[   44.005566]  r5:82138840 r4:82b73e00
[   44.009149] [<80185abc>] (handle_irq_event) from [<8018a79c>] (handle_level_irq+0xd4/0x15c)
[   44.017504]  r7:00000000 r6:00000000 r5:82138840 r4:82b73e00
[   44.023170] [<8018a6c8>] (handle_level_irq) from [<80184844>] (generic_handle_irq+0x44/0x54)
[   44.031609]  r5:82138840 r4:00000018
[   44.035194] [<80184800>] (generic_handle_irq) from [<8068b5f0>] (bcm2835_gpio_irq_handle_bank+0x98/0xd8)
[   44.044682] [<8068b558>] (bcm2835_gpio_irq_handle_bank) from [<8068b734>] (bcm2835_gpio_irq_handler+0x104/0x138)
[   44.054858]  r6:82138840 r5:80fe5fb4 r4:814e8618
[   44.059483] [<8068b630>] (bcm2835_gpio_irq_handler) from [<80184844>] (generic_handle_irq+0x44/0x54)
[   44.068619]  r7:00000000 r6:00000001 r5:00000000 r4:80f05aec
[   44.074288] [<80184800>] (generic_handle_irq) from [<80682e54>] (bcm2836_chained_handle_irq+0x48/0x4c)
[   44.083603] [<80682e0c>] (bcm2836_chained_handle_irq) from [<80184844>] (generic_handle_irq+0x44/0x54)
[   44.092910]  r5:00000000 r4:80eefd90
[   44.096493] [<80184800>] (generic_handle_irq) from [<80184fa8>] (__handle_domain_irq+0x6c/0xc0)
[   44.105202] [<80184f3c>] (__handle_domain_irq) from [<801012c8>] (bcm2836_arm_irqchip_handle_irq+0x60/0x64)
[   44.114950]  r9:8da26000 r8:86e56900 r7:8da27ed4 r6:ffffffff r5:600f0113 r4:80186a14
[   44.122699] [<80101268>] (bcm2836_arm_irqchip_handle_irq) from [<80100abc>] (__irq_svc+0x5c/0x7c)
[   44.131571] Exception stack(0x8da27ea0 to 0x8da27ee8)
[   44.136628] 7ea0: 80186a10 00000000 600f0193 600f0193 82b73e00 86e56900 82b73e18 82b73e6c
[   44.144811] 7ec0: 86e56900 80186a7c ffffe000 8da27f0c ffffe000 8da27ef0 8da26000 80186a14
[   44.152988] 7ee0: 600f0113 ffffffff
[   44.156488] [<80186968>] (irq_finalize_oneshot.part.0) from [<80186afc>] (irq_thread_fn+0x80/0x88)
[   44.165450]  r7:8da26000 r6:00000000 r5:82b73e00 r4:86e56900
[   44.171118] [<80186a7c>] (irq_thread_fn) from [<801867b4>] (irq_thread+0x164/0x238)
[   44.178778]  r7:8da26000 r6:86e56924 r5:82b73e00 r4:00000000
[   44.184445] [<80186650>] (irq_thread) from [<8014414c>] (kthread+0x144/0x170)
[   44.191587]  r10:914077ac r9:86e56900 r8:80186650 r7:8da26000 r6:00000000 r5:86e56040
[   44.199415]  r4:86fb9140
[   44.201954] [<80144008>] (kthread) from [<801000ec>] (ret_from_fork+0x14/0x28)
[   44.209175] Exception stack(0x8da27fb0 to 0x8da27ff8)
[   44.214228] 7fa0:                                     00000000 00000000 00000000 00000000
[   44.222410] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   44.230590] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   44.237210]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80144008
[   44.245039]  r4:86e56040
[   44.247569] handlers:
[   44.249846] [<b02ed657>] irq_default_primary_handler threaded [<bfb5a5ff>] mcp251xfd_irq [mcp251xfd]
[   44.259006] Disabling IRQ #199
```

and proc/interupts is showing:

```
199:     100001          0          0          0  pinctrl-bcm2835  **24** Level     spi1.0
200:     399959          0          0          0  pinctrl-bcm2835  25 Level     spi0.0

```

can1 is still working and receiving packets!
[waveshare.dts.txt](https://github.com/raspberrypi/linux/files/8128065/waveshare.dts.txt)

marckleinebudde, 2022-02-23:
This doesn't look consistent:

> // redo: ovmerge -c spi1-1cs-overlay.dts,cs0_pin=18,cs0_spidev=false mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi1-0,interrupt=24

compare to you've written above:

>spi0 CS0 is 8
spi1 CS0 is 26
IRQ on spi0 CS0 is 25
IRQ on spi1 CS0 is 16

nmbath, 2022-02-23:
@marckleinebudde 

I had to re-do it as I miss typed the interrupt first time round... My local file, from second time round has

// redo: ovmerge -c spi1-1cs-overlay.dts,cs0_pin=26,cs0_spidev=false mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi1-0,interrupt=16

I musty have dragged the wrong file in here! Updated file attached 
[waveshare.dts.txt](https://github.com/raspberrypi/linux/files/8128189/waveshare.dts.txt)

I have just rebuilt it all again using..

```
ovmerge -c spi1-1cs-overlay.dts,cs0_pin=26,cs0_spidev=false mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi1-0,interrupt=16 > ~/overlay.dts 
cpp -nostdinc -I /home/mark/development/linux/include -undef -x assembler-with-cpp ~/overlay.dts > ~/overlay.tmp
dtc -O dtb -b 0 -o ~/overlay.dtbo ~/overlay.tmp 

```

and it is not setting the interrupt to 16 on spi1-0, it is setting it to 24 for some reason...

199:     100001          0          0          0  pinctrl-bcm2835  24 Level     spi1.0
200:      16086          0          0          0  pinctrl-bcm2835  25 Level     spi0.0

and throwing the above backtrace as a result

It looks like this part isn't working `mcp251xfd-overlay.dts,spi1-0,interrupt=16` and is picking interrupt=24 from somewhere.

marckleinebudde, 2022-02-23:
At least in the overlay it says IRQ 16 for SPI1. Maybe some stuff left in the `config.txt`?

pelwell, 2022-02-24:
@nmbath The most obvious explanation for your symptoms is not running the version of the overlay that you think you are running. What does the following command display?:
```
od -An -td4 --endian=big /proc/device-tree/soc/spi@7e215080/mcp251xfd@0/interrupts
```

nmbath, 2022-02-24:
@marckleinebudde 

the cold light of day, verses late in the day is a beautiful thing!

Moved the corrected overlay.dtbo onto the test box, but didn't copy it to /u-boot/overlays.

```
199:      75299          0          0          0  pinctrl-bcm2835  25 Level     spi0.0
200:          2          0          0          0  pinctrl-bcm2835  16 Level     spi1.0

```

So I can confirm that putting this in config.txt works:

```
dtoverlay=spi0-1cs,cs0_pin=8
dtoverlay=spi1-1cs,cs0_pin=26,cs0_spidev=false
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi1-0,interrupt=16

```

as well as an overlay built with:

`ovmerge -c spi1-1cs-overlay.dts,cs0_pin=26,cs0_spidev=false mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi1-0,interrupt=16`

Would you like me to do a PR, if so any name you would like to use for the overlay?
[waveshare.dts.txt](https://github.com/raspberrypi/linux/files/8131270/waveshare.dts.txt)

marckleinebudde, 2022-02-24:
Very nice!

Please add a link to the product website (https://www.waveshare.com/2-ch-can-fd-hat.htm) in the 3rd line of the dts. What about `waveshare-can-fd-hat-mode-a-overlay.dts`. Can you make a `mode-b` version, too?

pelwell, 2022-02-24:
Marc's name suggestion looks good. Each overlay will need an entry in the overlays Makefile and the README. Keep all entries in alphabetical order, and the README entries require two blank lines between them and no hard TABs.

nmbath, 2022-02-24:
> Can you make a mode-b version, too?

I can, but will not be able to test as requires a soldering iron!

marckleinebudde, 2022-02-24:
> I can, but will not be able to test as requires a soldering iron!

I think (hope?) waveshare's documentation is correct so we should get it working without actual testing. Look at the `seeed-can-fd-hat-v2-overlay.dts` as an example for both controllers on the same SPI interface.

nmbath, 2022-02-24:
submitted PR

overlays: Add overlays for WaveShare 2-Channel Isolated CAN FD HAT #4905

dav2dg, 2022-07-02:
For what it is worth, I have built 20 devices with this Waveshare card, in one batch.
All have the same image.
All configured the same through ansible.
I have tried replacing board - no luck.
I have just found this update so I will read and see if it helps me.

marckleinebudde, 2022-07-02:
Hey @dav2dg,

can you elaborate on you problem?

There's a new version of the waveshare board. Check the back side of the board. The [new boards are `Rev2.1`](https://www.waveshare.com/w/upload/0/0e/2-CH_FAN_FD_HAT_diffVer.jpg). The [old version of the board](https://m.media-amazon.com/images/I/61qzyPui0zL._AC_SX522_.jpg) doesn't have a revision number on the back.

For the old, revision-less boards, you can use the following line in your `config.txt`:

```
dtoverlay=waveshare-can-fd-hat-mode-a
```

New v5.15 raspi kernels or newer than v5.15 have this overlay.

dav2dg, 2022-07-02:
OK ... I get a range of problems.
I have both version 2.1 and the previous board (plus older seeed boards)
Issues:

- issues are usually consistent for a CAN Hat (but not always)
- I came across this issue from a search for "mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000)" which resulted in NO can devices
- I then minutes ago put the hardware back together and got another issue which is intermittent that only one can bus boots:
`pi@e45f013eb505:~ $ dmesg|grep spi
[    9.580251] spi_master spi0: will run message pump with realtime priority
[    9.587941] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    9.588950] spi_master spi1: will run message pump with realtime priority
[    9.605431] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
pi@e45f013eb505:~ $
`
- another funny result I get is Can0 and can1 swapped

I will just make sure I did not get a different result because I used a different HAT. I have quite a few bits of hardware I can try.

dav2dg, 2022-07-02:
Sorry I should have read the rest of your message.
I have now added
`dtoverlay=waveshare-can-fd-hat-mode-a`
I recall you had something similar for seeed??

interesting ... same result no can devices for HAT 2:

> pi@e45f013eb397:~ $ dmesg| grep spi
> [    9.656495] spi_master spi0: will run message pump with realtime priority
> [    9.690972] mcp251xfd spi0.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).
> [    9.708893] spi_master spi1: will run message pump with realtime priority
> [    9.747526] mcp251xfd spi1.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).

Config.txt:
[all]

> #dtoverlay=spi1-3cs
> #dtoverlay=mcp251xfd,spi0-0,interrupt=25
> #dtoverlay=mcp251xfd,spi1-0,interrupt=24
> dtoverlay=waveshare-can-fd-hat-mode-a

reboot same result
????

marckleinebudde, 2022-07-02:
> OK ... I get a range of problems.
> I have both version 2.1 and the previous board (plus older seeed boards)

Each of your boards needs different config entry. The old waveshare (without revision) works with `dtoverlay=waveshare-can-fd-hat-mode-a`.

> I use wiki config.txt
> ```
> [all]
> dtoverlay=spi1-3cs
> dtoverlay=mcp251xfd,spi0-0,interrupt=25
> dtoverlay=mcp251xfd,spi1-0,interrupt=24
> ```

I think that config snippet for the old board in mode "a" in the wiki will not work, the board uses a different pin for chip select as the overlay `spi1-3cs` specifies. Please use `dtoverlay=waveshare-can-fd-hat-mode-a`.

I don't know if the waveshare `Rev2.1` board works with the config from their wiki, I don't have that baord.

> I recall you had something similar for seeed??

Yes. Seeed has 2 variants of the board, too. The [v2 board](https://media-cdn.seeedstudio.com/media/catalog/product/cache/b2267b506d4e4594666ef83a79896a9a/1/0/103990563_preview-07.png) has a coin cell holder, while the [v1 board](https://media-cdn.seeedstudio.com/media/catalog/product/cache/b2267b506d4e4594666ef83a79896a9a/2/-/2-channel-can-bus-fd-shield-for-raspberry-pi-preview.png) doesn't have one. Use `dtoverlay=seeed-can-fd-hat-v1` or `dtoverlay=seeed-can-fd-hat-v2` in your `config.txt`.

nmbath, 2022-07-02:
I have a rev 2.1 board and confirm it generally works. The config.txt sniper that @marckleinebudde posted works. I do have issues that sometimes the can ports are detected and configured in reverse. A reboot generally sorta that. I do have other issues with high bit error rates in either TX or RX. Still trying to debug that one.

marckleinebudde, 2022-07-02:
Hey @nmbath,

See [predicable network interface names](https://lore.kernel.org/all/32e1f29b-b16e-bc8e-e57a-92c67d45a335@pengutronix.de/) for the classical `udev` or the more modern `systemd-networkd` based approach.

Can you test if this combined overlay works for the waveshare `Rev2.1` board?

[0001-overlays-add-support-for-waveshare-can-fd-hat-rev2.1.patch.TXT](https://github.com/raspberrypi/linux/files/9032746/0001-overlays-add-support-for-waveshare-can-fd-hat-rev2.1.patch.TXT)

> I do have other issues with high bit error rates in either TX or RX. Still trying to debug that one.

Can you describe the problem?

dav2dg, 2022-07-02:
@marckleinebudde 
yes I am aware each board needs a different config.
These are the 20 boards I have
[image]
This is the only one with a problem. Even if I change the Can HAT. So tomorrow I will change the base board.
All examples are for a 2.1 board

Please see my comments above, I did do as you suggest...

> I think that config snippet for the old board in mode "a" in the wiki will not work, the board uses a different pin for chip select as the overlay spi1-3cs specifies. Please use dtoverlay=waveshare-can-fd-hat-mode-a.

@nmbath I have 50 V2.1 boards, of the 11 I have tried, 9 work. Maybe it is the base board, I use a rpi CM4 with a base board from Waveshare.

marckleinebudde, 2022-07-02:
Some working boards and some not working boards of the same type on the same board suggests a hardware problem.

DavidBoJ, 2022-07-04:
2-CH CAN FD HAT Rev2.1

 ip -d link show dev can0
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 0
          bitrate 250000 sample-point 0.875
          tq 25 prop-seg 69 phase-seg1 70 phase-seg2 20 sjw 1
          mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
          mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
          clock 40000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535

uname -a
Linux cilix-19 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
On a Raspberry Pi4

In my application the /var/log/syslog is filled up with:
        mcp251xfd spi0.0 can0: Transmit Event FIFO buffer not empty
After a while the disk is full and the system can crash. Is there anyway I can disable the logging from the CAN bus?

I have two slaves on my network for which it is difficult to establish a connection. With two others slaves it seems to work (But I didn't check the logs)
I have controlled the two slaves with Pican2 in the past but not with Bullseye
The physical network is 1.2m
 Any suggestions?

dav2dg, 2022-07-05:
@marckleinebudde, I took your point and agree it really did look like a hardware problem and may well be for the specific problem I had.

I have since taken mutiple new Waveshare 2-CH CAN FD HAT Rev2.1, CM4102032, Waveshare base board. 
After testing multiple I was not able to get the mainline overlay to work...

```
Success both can0 & can1
## Setup for Waveshare Dual CAN FD v2.1
# dtoverlay=spi1-3cs
# dtoverlay=mcp251xfd,spi0-0,interrupt=25
# dtoverlay=mcp251xfd,spi1-0,interrupt=24
pi@e45f014e8730:~ $ dmesg | grep -E 'can|spi|mcp2'
[    6.602899] spi_master spi0: will run message pump with realtime priority
[    6.626836] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    6.630779] spi_master spi1: will run message pump with realtime priority
[    6.639008] mcp251xfd spi1.0 can1: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    8.245417] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[    9.247876] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   14.570723] can: controller area network core
[   14.577323] can: raw protocol
 
 
Fail - only can0
##
## NEW NEW NEW NEW NEW NEW NEW NEW 
## Mainline Mainline
##  
## dtoverlay=waveshare-can-fd-hat-mode-a
pi@e45f014e8730:~ $ dmesg | grep -E 'can|spi|mcp2'
[    6.324791] spi_master spi0: will run message pump with realtime priority
[    6.352909] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
[    6.422089] spi_master spi1: will run message pump with realtime priority
[    6.462702] mcp251xfd spi1.0 (unnamed net_device) (uninitialized): Failed to detect MCP251xFD (osc=0x00000000).
[    7.780180] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   14.644207] can: controller area network core
[   14.651967] can: raw protocol
```

Any suggestions as to what I should try next. I am going to check same results on all devices.

marckleinebudde, 2022-07-05:
The `dtoverlay=waveshare-can-fd-hat-mode-a` works for the old version only. 
Please try the overlay from the patch attached to https://github.com/raspberrypi/linux/issues/4902#issuecomment-1172870134

nmbath, 2022-07-05:
Also check that you do not have anything else opening the GPIO pins that the old version uses. It dose not use standard GPIO pins. I had issues with my old revision board and can1 due to something else wanting to use the GPIO pins that can1 required.

[15 more replies]