raspberrypi/linux #4903: Driver crash in nft_do_chain

cromefire, 2022-02-21:
### Describe the bug

dmesg and syslog note a kernel driver crash. I have also experienced a disconnect that could be resolved by a hard reset (I have no easy access to the HDMI port), but it is unclear whether both are related.

### Steps to reproduce the behaviour

Boot the Pi and maybe it'll occur.

### Device (s)

Raspberry Pi Zero W / WH

### System

- Raspberry Pi reference 2021-05-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 66cd2d17a0d2d04985b83a2ba830915c9a7d81dc, stage2
- Feb  4 2022 12:11:20 
Copyright (c) 2012 Broadcom
version a26faf97e3bf76bcc23949d7cdab2f96f399a0c3 (clean) (release) (start)
- `Linux led-control 5.15.24+ #1526 Fri Feb 18 14:53:36 GMT 2022 armv6l GNU/Linux`

### Logs

```
[  152.877573] 8<--- cut here ---
[  152.879847] Unable to handle kernel paging request at virtual address bf1535c4
[  152.882067] pgd = 078f368c
[  152.884202] [bf1535c4] *pgd=02938811, *pte=0356704f, *ppte=0356701e
[  152.886567] Internal error: Oops: 80000007 [#1] ARM
[  152.888764] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes spidev brcmfmac cdc_ether brcmutil r8152 sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) snd_bcm2835(C) spi_bcm2835 bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer snd videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_LOG nf_log_syslog nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 i2c_dev nf_tables nfnetlink ip_tables x_tables ipv6
[  152.916217] CPU: 0 PID: 758 Comm: sshd Tainted: G         C        5.15.24+ #1526
[  152.921282] Hardware name: BCM2835
[  152.923749] PC is at nft_counter_eval+0x84/0x94 [nft_counter]
[  152.926262] LR is at trace_hardirqs_on+0x4c/0x148
[  152.928771] pc : [<bf1535c4>]    lr : [<c00f8f18>]    psr: 60000013
[  152.931253] sp : c58fb9b8  ip : c58fb978  fp : c58fb9dc
[  152.933658] r10: c3b29f80  r9 : 00000000  r8 : 0002a4fa
[  152.936065] r7 : bf0d5b8c  r6 : 00000034  r5 : 00000000  r4 : 0002a4c6
[  152.938485] r3 : 60000093  r2 : 60000093  r1 : 00000000  r0 : c0028710
[  152.940848] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  152.943216] Control: 00c5387d  Table: 05434008  DAC: 00000055
[  152.945518] Register r0 information: non-slab/vmalloc memory
[  152.947790] Register r1 information: NULL pointer
[  152.949935] Register r2 information: non-paged memory
[  152.952015] Register r3 information: non-paged memory
[  152.953994] Register r4 information: non-paged memory
[  152.955864] Register r5 information: NULL pointer
[  152.957638] Register r6 information: non-paged memory
[  152.959350] Register r7 information: 52-page vmalloc region starting at 0xbf0b1000 allocated at load_module+0xa74/0x25f8
[  152.962699] Register r8 information: non-paged memory
[  152.964365] Register r9 information: NULL pointer
[  152.965974] Register r10 information: slab kmalloc-64 start c3b29f80 pointer offset 0 size 64
[  152.969267] Register r11 information: non-slab/vmalloc memory
[  152.971002] Register r12 information: non-slab/vmalloc memory
[  152.972664] Process sshd (pid: 758, stack limit = 0x42fe3b09)
[  152.974345] Stack: (0xc58fb9b8 to 0xc58fc000)
[  152.976014] b9a0:                                                       c3b29f90 c3b29fc0
[  152.979279] b9c0: bf0d5ac4 bf0d5b8c c58fbb1c bf0d5d3c c58fbb14 c58fb9e0 bf0b11b0 bf15354c
[  152.982523] b9e0: c3ae72b0 00000000 c58fba64 00000000 c3ae72b0 c356b440 00000129 0129ba8c
[  152.986047] ba00: a5906935 b5affdda 00000000 00000004 a5900035 ffffffff 00000003 00000000
[  152.990042] ba20: c58fbb74 c0d38418 00000003 00000003 c0c1f028 a0d9aac5 c09353c0 c0d38418
[  152.994429] ba40: c58fbab4 c58fba50 bf0f32a8 c0528e78 bf0f3e28 bf0f3c9c 4ab2a8c0 00000000
[  152.998972] ba60: 00000000 00000000 00021600 2bb2a8c0 00000000 00000000 00000000 bf0f6c28
[  153.003664] ba80: cd60c000 2507c65b c58fbab4 c36d73c0 c58fbb74 00000003 00000000 c58fbadc
[  153.008537] baa0: c5a0b540 f8de6b06 c58fbb34 c58fbab8 bf0f6fb0 bf0fc3dc c58fbb74 c0d38418
[  153.013578] bac0: c58fbadc 00000000 c58fbdd4 00000014 c022129c 062208c8 c58fbb0c 4ab2a8c0
[  153.018790] bae0: 00000000 2507c65b 00000000 c0c1f028 00000002 c356b850 c36d73c0 c58fbb74
[  153.024296] bb00: c356b840 c0d38418 c58fbb44 c58fbb18 bf0c4a08 bf0b10c8 c36d73c0 c36d73c0
[  153.029866] bb20: c58fbb74 00000601 00000014 2507c65b bf0fb420 00000001 c58fbb6c c58fbb48
[  153.035390] bb40: c07cd0c8 bf0c499c c36d73c0 c0c1f028 c0d38418 c3b15b80 c3b15df0 00000000
[... 67 lines cut ...]
[  153.375879]  r10:00000000 r9:00000000 r8:c6adb5a0 r7:c64f9e00 r6:c58fbe60 r5:c07ec3b0
[  153.380789]  r4:c0c1f028
[  153.383108] [<c08215e8>] (inet_recvmsg) from [<c0728694>] (sock_recvmsg+0x50/0x54)
[  153.387913]  r6:c64f9e00 r5:c08215e8 r4:c58fbe60
[  153.390351] [<c0728644>] (sock_recvmsg) from [<c072873c>] (sock_read_iter+0xa4/0xf8)
[  153.395211]  r7:c64f9e00 r6:c0c1f028 r5:c58fbf00 r4:c58fbee8
[  153.397691] [<c0728698>] (sock_read_iter) from [<c0209974>] (vfs_read+0x310/0x344)
[  153.402491]  r8:c0c1f028 r7:00004004 r6:00000000 r5:00004000 r4:c6adb5a0
[  153.404959] [<c0209664>] (vfs_read) from [<c0209ec4>] (ksys_read+0xc0/0xd8)
[  153.407518]  r10:00000000 r9:c58fa000 r8:00004000 r7:bef222ec r6:c0c1f028 r5:c6adb5a0
[  153.412393]  r4:c6adb5a0
[  153.414737] [<c0209e04>] (ksys_read) from [<c0209ef4>] (sys_read+0x18/0x1c)
[  153.417134]  r9:c58fa000 r8:c00083e4 r7:00000003 r6:016f0750 r5:00000003 r4:000bf780
[  153.421936] [<c0209edc>] (sys_read) from [<c0008260>] (ret_fast_syscall+0x0/0x1c)
[  153.426805] Exception stack(0xc58fbfa8 to 0xc58fbff0)
[  153.429234] bfa0:                   000bf780 00000003 00000003 bef222ec 00004000 00000000
[  153.434083] bfc0: 000bf780 00000003 016f0750 00000003 00000000 000c2b30 000c2d58 017095a0
[  153.438995] bfe0: 0000006c bef222d0 00021ec8 b6a4b870
[  153.441505] Code: e2833001 e59f0010 e58c3000 eb3b542d (e89dabf0) 
[  153.444297] ---[ end trace bfc0b61053b333a8 ]---
```

### Additional context

I use `netplan` and `ufw`. Unlikely to affect it, but just noting that here.

cromefire, 2022-02-21:
Second crash, that caused the disconnect (might be related):
<details>
<summary>dmesg</summary>

```
Feb 19 23:58:26 led-control kernel: [107427.096885] 8<--- cut here ---
Feb 19 23:58:26 led-control kernel: [107427.099281] Unable to handle kernel NULL pointer dereference at virtual address 00000d84
Feb 19 23:58:26 led-control kernel: [107427.103821] pgd = 2e80ff30
Feb 19 23:58:26 led-control kernel: [107427.106124] [00000d84] *pgd=00000000
Feb 19 23:58:26 led-control kernel: [107427.108376] Internal error: Oops: 5 [#6] ARM
Feb 19 23:58:26 led-control kernel: [107427.110536] Modules linked in: aes_arm aes_generic cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes spidev cdc_ether r8152 brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill raspbe
rrypi_hwmon i2c_bcm2835 bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig snd_bcm2835(C) videobuf2_memops videobuf2_v4l2 spi_bcm2835 videobuf2_common snd_pcm snd_time
r snd videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_LOG nf_log_syslog nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter nf_co
nntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink i2c_dev ip_tables x_tables ipv6
Feb 19 23:58:26 led-control kernel: [107427.135664] CPU: 0 PID: 1657 Comm: systemd-journal Tainted: G      D  C        5.15.23+ #1525
Feb 19 23:58:26 led-control kernel: [107427.140433] Hardware name: BCM2835
Feb 19 23:58:26 led-control kernel: [107427.142783] PC is at cap_capable+0x28/0x88
Feb 19 23:58:26 led-control kernel: [107427.145129] LR is at security_capable+0x4c/0x6c
Feb 19 23:58:26 led-control kernel: [107427.147419] pc : [<c046a944>]    lr : [<c046e028>]    psr: a0000013
Feb 19 23:58:26 led-control kernel: [107427.149718] sp : c68c9c88  ip : c68c9c98  fp : c68c9c94
Feb 19 23:58:26 led-control kernel: [107427.152034] r10: 00000000  r9 : c13b6080  r8 : 00000cc0
Feb 19 23:58:26 led-control kernel: [107427.154369] r7 : 00000015  r6 : 00000000  r5 : c046a91c  r4 : c0b4e714
Feb 19 23:58:26 led-control kernel: [107427.156738] r3 : c0c27990  r2 : 00000015  r1 : 00000cc0  r0 : c13b6080
Feb 19 23:58:26 led-control kernel: [107427.159068] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Feb 19 23:58:26 led-control kernel: [107427.161372] Control: 00c5387d  Table: 027b0008  DAC: 00000055
Feb 19 23:58:26 led-control kernel: [107427.163597] Register r0 information: slab cred_jar start c13b6080 pointer offset 0
Feb 19 23:58:26 led-control kernel: [107427.168028] Register r1 information: non-paged memory
Feb 19 23:58:26 led-control kernel: [107427.170245] Register r2 information: non-paged memory
Feb 19 23:58:26 led-control kernel: [107427.172347] Register r3 information: non-slab/vmalloc memory
Feb 19 23:58:26 led-control kernel: [107427.174383] Register r4 information: non-slab/vmalloc memory
Feb 19 23:58:26 led-control kernel: [107427.176304] Register r5 information: non-slab/vmalloc memory
Feb 19 23:58:26 led-control kernel: [107427.178110] Register r6 information: NULL pointer
Feb 19 23:58:26 led-control kernel: [107427.179808] Register r7 information: non-paged memory
Feb 19 23:58:26 led-control kernel: [107427.181467] Register r8 information: non-paged memory
Feb 19 23:58:26 led-control kernel: [107427.183042] Register r9 information: slab cred_jar start c13b6080 pointer offset 0
Feb 19 23:58:26 led-control kernel: [107427.186221] Register r10 information: NULL pointer
Feb 19 23:58:26 led-control kernel: [107427.187909] Register r11 information: non-slab/vmalloc memory
Feb 19 23:58:26 led-control kernel: [107427.189578] Register r12 information: non-slab/vmalloc memory
Feb 19 23:58:26 led-control kernel: [107427.191164] Process systemd-journal (pid: 1657, stack limit = 0xa3424ac4)
Feb 19 23:58:26 led-control kernel: [107427.192810] Stack: (0xc68c9c88 to 0xc68ca000)
Feb 19 23:58:26 led-control kernel: [107427.194429] 9c80:                   c68c9cbc c68c9c98 c046e028 c046a928 ffffe000 c68c9d6c
Feb 19 23:58:26 led-control kernel: [107427.197657] 9ca0: c13b6080 00000001 00000000 00000000 c68c9cd4 c68c9cc0 c002ce28 c046dfe8
Feb 19 23:58:26 led-control kernel: [107427.200985] 9cc0: c68c9ddc c68c9d6c c68c9ce4 c68c9cd8 c002ce80 c002cdf0 c68c9d34 c68c9ce8
Feb 19 23:58:26 led-control kernel: [107427.204367] 9ce0: c073f5dc c002ce70 c0df266c 00000000 c1d75700 c68c9f38 00000001 00000000
Feb 19 23:58:26 led-control kernel: [107427.208088] 9d00: 00000000 8d380ad7 c68c9d5c c68c9f38 c1d75700 00000018 c1d75700 c0c1f028
Feb 19 23:58:26 led-control kernel: [107427.212202] 9d20: c2589680 0000004c c68c9db4 c68c9d38 c086da84 c073f4ac 00000028 c68c9e3c
Feb 19 23:58:26 led-control kernel: [107427.216683] 9d40: 00000028 00000000 00000005 c04d5fe0 c68c9d84 c68c9d60 c0d38418 c68c9e7c
Feb 19 23:58:26 led-control kernel: [107427.221553] 9d60: 0000004c c68c9e3c c68c9e38 00000000 00000000 00000000 ffffffff ffffffff
Feb 19 23:58:26 led-control kernel: [107427.226525] 9d80: 00000000 8d380ad7 0000004c c68c9f38 c1d75700 00000018 c1d75700 00000000
Feb 19 23:58:26 led-control kernel: [107427.231739] 9da0: c0c1f028 c68c9ddc c68c9dcc c68c9db8 c0728c6c c086d948 c68c9f38 00004040
Feb 19 23:58:26 led-control kernel: [107427.237204] 9dc0: c68c9e2c c68c9dd0 c0729028 c0728c34 00000000 c0c1f028 c68c9e2c 00000018
[... cut ...]
Feb 19 23:58:26 led-control kernel: [107427.438734] 9fa0:                   bed3e5e4 bed3e940 00000003 bed3e218 00004000 00000000
Feb 19 23:58:26 led-control kernel: [107427.443616] 9fc0: bed3e5e4 bed3e940 bed3e218 00000128 bed3e2d4 00000005 c8ed4e00 bed3e5e4
Feb 19 23:58:26 led-control kernel: [107427.448516] 9fe0: 0000006c bed3e1e8 004ab528 b6e62970
Feb 19 23:58:26 led-control kernel: [107427.451071] Code: ea000000 e1a0100c e1530001 0a00000c (e591e0c4) 
Feb 19 23:58:26 led-control kernel: [107427.453847] ---[ end trace 8d7d38145906f393 ]---
```

</details>

pelwell, 2022-02-21:
The standard observation, rarely wrong, is that random memory faults are caused by faulty hardware or a lack of power. The fact that you have two such failures, apparently unconnected, only makes it more likely to be one of these two.

cromefire, 2022-02-21:
Well how can I proof that to raspberry pi/the seller? Because this hardware is definitely in it's warranty period. I can change the power supply, but it has been working fine and is overspeced, so it's unlikely to be the source of the problem.

pelwell, 2022-02-21:
Differential diagnosis - change something (the Pi, the supply, the SD card and image) and see what happens.

cromefire, 2022-02-21:
Will try, even tough I can't change the pi because I only have one.

closed 2022-02-21 by cromefire

pelwell, 2022-02-21:
> I can change the power supply, but it has been working fine

But has it been powering a Pi? That seems unlikely of it only have this one. If the supply has a removable USB cable then make sure that is also good enough - we've seen some cables and inline switches with shocking voltage losses.

cromefire, 2022-02-21:
> > I can change the power supply, but it has been working fine
> 
> But has it been powering a Pi? That seems unlikely if you only have this one. If the supply has a removable USB cable then make sure that is also good enough - we've seen some cables and inline switches with shocking voltage losses.

Yes I've bought it to power this Pi specially. This very Pi has been working fine for a few months (without any hardware changes in an enclosed box (and yes temperatures where fine)) until having problems after an update seemingly. But reinstalling the OS didn't help, upgrading the kernel resolved one of the kernel crashes (I had pretty consistent crashes in the SPI module), but now I get these. I have connected it to an official power supply now, so it should definitely be okay power wise now, if that doesn't work I can may borrow the SD card from somewhere else (also it's also just bought for this Pi and only a few months old with pretty much only the default logging stuff writing something to it, so unlikely) and if that doesn't work it kinda has to be the Pi if it's not a kernel bug.

cromefire, 2022-02-22:
Well that didn't help either, so I guess the hardware is just dead. Thank for helping me figure that out.