raspberrypi/linux #4906: v4l2 codec hanging with certain input data when encoding to h.264

berkutta, 2022-02-24:
### Describe the bug

When feeding raw Data from a ADV7282-M CVBS to CSI2 decoder to the v4l2 codec (GStreamer v4l2h264enc) it can happen that the encoder starts to stop outputting data and is starting to use memory without freeing. This continuous till the Application gets killed trough OOM Management.

### Steps to reproduce the behaviour

I was able to reproduce this exact phenomen with feeding raw data from `/dev/urandom` to the v4l2 codedc. Sample code is attached.

Code:
```
#include <stdio.h>
#include <stdbool.h>

#include <unistd.h>

#include <gst/gst.h>
#include <gst/app/gstappsrc.h>
#include <gst/app/gstappsink.h>
#include <gst/gstbuffer.h>
#include <gst/gstmemory.h>

#ifdef __arm__
#define IMAGE_WIDTH 720
#define IMAGE_HEIGHT 576
#endif

#define VIDEO_GST_DEBUG "*:3"
GstElement *pPipeline = NULL;
GstElement *pAppSrc = NULL;

void init_pipeline()
{
    if (pPipeline != NULL)
        return;

    unsigned int major, minor, micro, nano;

    /* initialize gstreamer */
    setenv("GST_DEBUG", VIDEO_GST_DEBUG, 5);
    gst_init(NULL, NULL);
    gst_version(&major, &minor, &micro, &nano);
    gst_update_registry();

    printf("Gstreamer initialized. \n");
    printf("Gstreamer version: %u %u %u %u \n", major, minor, micro, nano);

#ifdef __arm__
    char *pipelineStr = "appsrc name=appsrc ! video/x-raw, interlace-mode=interleaved, framerate=25/1, format=UYVY, width=720, height=576 ! "
                        "deinterlace method=linear ! videocrop top=16 left=16 right=16 bottom=16 ! videorate ! videoscale ! "
                        "video/x-raw, framerate=25/1, width=640, height=480 ! videoconvert ! "
                        "queue ! v4l2h264enc output-io-mode=2 extra-controls=\"controls,h264_profile=0,video_bitrate=1000000,h264_i_frame_period=10,repeat_sequence_header=1;\" ! "
                        "video/x-h264, level=(string)3, profile=baseline, stream-format=byte-stream ! filesink location=out.h264";

    pPipeline = gst_parse_launch(pipelineStr, NULL);

    printf("Pipeline: %s \n", pipelineStr);
#endif

    pAppSrc = gst_bin_get_by_name(GST_BIN(pPipeline), "appsrc");
    // ASSERT(pAppSrc != NULL);
    g_object_set(pAppSrc, "format", GST_FORMAT_TIME, NULL);

    if (pPipeline == NULL)
    {
        printf("Failed to init pipeline \n");
        exit(-1);
    }

    /* Start playing */
    gst_element_set_state(pPipeline, GST_STATE_PLAYING);
}

FILE *myfile = NULL;

void write_gst_data(char *buffer, size_t length)
{
    GstBuffer *gstbuffer;
    GstFlowReturn ret;
    GstMapInfo map;

    /* copy frame to gst buffer */
    gstbuffer = gst_buffer_new_and_alloc(length);
    gst_buffer_map(gstbuffer, &map, GST_MAP_WRITE);
    memcpy(map.data, buffer, length);
    gst_buffer_unmap(gstbuffer, &map);

    /* push data to appsrc */
    ret = gst_app_src_push_buffer(GST_APP_SRC(pAppSrc), gstbuffer);
    if (ret != GST_FLOW_OK)
    {
        printf("-EINVAL GST_FLOW! \n");
    }
}

int main(int argc, char *argv[])
{
    init_pipeline();

    while (true)
    {
        char *uyvybuffer = (char *)malloc(IMAGE_WIDTH * IMAGE_HEIGHT * 2);

        FILE *fp;
        fp = fopen("/dev/urandom", "r");
        fread(uyvybuffer, 1, IMAGE_WIDTH * IMAGE_HEIGHT * 2, fp);
        fclose(fp);

        write_gst_data(uyvybuffer, IMAGE_WIDTH * IMAGE_HEIGHT * 2);

        free(uyvybuffer);

        // 25FPS -> 40ms
        usleep(40 * 1000);
    }
}
```

Build:
```
gcc -g main.c -I. -o decoder `pkg-config --cflags --libs gstreamer-1.0` `pkg-config --cflags --libs gstreamer-app-1.0`
```

### Device (s)

Raspberry Pi CM4

### System

```
user@CM4:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2
user@CM4:~ $ sudo vcgencmd version
Oct 29 2021 10:47:33 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)
user@CM4:~ $ uname -a
Linux CM4 5.15.24-v7l+ #4 SMP Thu Feb 24 11:43:31 CET 2022 armv7l GNU/Linux
```

### Logs

No errors/warnings visible in GST/Syslog and dmesg Log. Also tried to enable debugging in bcm2835-v4l2-codec and mmal-vchiq but didn't get to any conclusion with it.

6by9, 2022-02-24:
Is this a vanilla copy of Raspberry Pi OS which is therefore using GStreamer 1.18.4?

berkutta, 2022-02-24:
Yep, vanilla Raspbian 11. The only modification is the Kernel, which contains a few patches in CSI2.

```
user@CM4:~ $ gst-launch-1.0 --version
gst-launch-1.0 version 1.18.4
GStreamer 1.18.4
http://packages.qa.debian.org/gstreamer1.0
```

berkutta, 2022-03-10:
Same issue also present on GStreamer 1.20.0 with ALARM Image on Raspberry Pi 3.

```
[root@alarmpi alarm]# gst-launch-1.0 --version
gst-launch-1.0 version 1.20.0
GStreamer 1.20.0
https://www.archlinux.org/
```

berkutta, 2022-03-14:
Looks like when shoving random data into mmal it can encode it without issues to h.264. So this should be an issue in the v4l2 translation layer or GStreamer.

Quedale, 2022-12-21:
I'm experiencing a similar issue in a similar context.
My camera is a USB 640x480 YUYV trying to encode with v4l2h264enc.
The first frame gets process normally and then it stops sending more frames.

After some testing, I realized that any h264_level below 3 simply won't work. (The encoder process won't start)
I also read that v4l2-codec only support maximum level of 4, although I get the same result if I use level 5.

I'm not entirely sure that the gstreamer v4l2 is at fault here since I did get it to work successfully before playing around profile and level configurations.
After playing around, the working configurations no longer works even after reboot. 
(I need to try a fresh build to see if it works back again)

I did some testing using v4l2-utils and the driver seems to accept all valid configurations without returning any error.
My next step is to test the v4l2 encoder directly without Gstreamer for a narrow test case.

As far as I can tell, this could still be a v4l2 or driver issue (bcm2835-codec).

6by9, 2022-12-21:
> After some testing, I realized that any h264_level below 3 simply won't work. (The encoder process won't start)

Correct. https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels
Level 3 will be required for VGA @ 30fps (1200MBs/frame, and 36000MBs/sec).
3.1 will be required for VGA @ 60fps.

> I also read that v4l2-codec only support maximum level of 4, although I get the same result if I use level 5.

It depends on whether you want realtime or not.
The codec was originally specced for realtime level 4.0 (1080p30), but increased clock speeds mean we can generally achieve level 4.2 (1080p60) dependent on other workload.
For transcode (non-realtime) purposes, it'll do 1080p at any framerate you fancy, but you have to be able to describe that in the stream headers, and that requires the option to request level 5 or 5.1.

> I did some testing using v4l2-utils and the driver seems to accept all valid configurations without returning any error.

The first point that the encoder can validate the configuration is at STREAMON, therefore that is the only call that can be configured to fail. Up to that point you can configure anything you like in any order you like.

You've not given any details on bitrate or profile that you're trying to encode, so there's not much further diagnosis that can be done.

Quedale, 2022-12-21:
Sorry I'm not the OP and I didn't mean to highjack it, just sharing my experience that seems to relate to it.

I'm honestly not targeting any profile in particular. I'm just trying to find a configuration that works.
I tried all profiles [0 to 4] combined with all levels [0 to 15]. (bitrate has no impact)
I tried the default value and "video_bitrate=25000000".

The result is always the same thing: The first frame is processed and sent to the sink and than it stops. (Gstreamer v4l2h264enc)

> The first point that the encoder can validate the configuration is at STREAMON, therefore that is the only call that can be configured to fail. Up to that point you can configure anything you like in any order you like.    

Thanks, that explains clearly why I'm not getting any errors setting v4l2 configurations. 
That also means Gstreamer doesn't get any feedback from the driver on this step.

[berkutta](https://github.com/berkutta) is also using Gstreamer, so there's still a possibility it is the cause.
I'll search for a "simple" v4l2-encode code sample and give it a shot outside gstreamer.

6by9, 2022-12-22:
I'm suspecting that it's a GStreamer issue as nothing in the codec can really trigger OOM.

I must confess to this having dropped off my radar (now assigned to me), but berkutta's pipeline looks overly complex. Certainly more standard pipelines are working fine, as is FFmpeg.

UVC webcams are a bit of a pain as the buffers that UVC creates come from the vmalloc allocator. Those are incompatible with the codec's requirements for contiguous buffers, so at the point of import they will fail. Forcing ```io-mode=2``` on v4l2src, and ```output-io-mode=2``` on v4l2h264enc may help, although it forces a copy from the UVC buffers to the encoder's buffers.
Getting the logging from the VPU is generally worthwhile - ```sudo vcdbg log msg```, although it may need additional logging levels enabling. Also worth a try is using ```start_debug=1``` in /boot/config.txt, and then seeing if any VPU asserts have triggered with ```sudo vcdbg log assert```.

For berkutta's use case, I did investigate a very similar one earlier this month - https://forums.raspberrypi.com/viewtopic.php?t=344001

Quedale, 2022-12-22:
I personally didn't get OOM in my scenario, but this can be caused by other elements in the pipeline.
Also I didn't let it run long enough to get there.

I do get a similar behaviour where the stream feed simply stops at the encoder. (I noticed that I sometimes get 2 frames)
While the pipeline is still running.

I tried setting io-mode and output-io-mode to 2, but I'm not even getting one frame with that.

I don't think the fault here is the source type (UVC) since I'm getting the same result using "videotestsrc".
Although I'm using the same Raw Caps for it.

I should be able to draft a very simple pipeline like "videotestsrc ! caps ! videoconvert ! v4l2h264enc extra-controls='' ! caps ! somesink" and achieve the same result. 

I'll test different input format and also try to dig in the logs see if I can find anything relevant.

6by9, 2022-12-22:
> I don't think the fault here is the source type (UVC) since I'm getting the same result using "videotestsrc".
Although I'm using the same Raw Caps for it.

Give me the command line you're using there and I can look into it.

Quedale, 2022-12-23:
My current RPi is running headless and has no means to view the stream locally.
So I've been working off the [gst-rtsp-server](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-rtsp-server/examples/test-onvif-backchannel.c) implementation in order to create a stream that I can view remotely.

My implementation is an extension of GstRtspOnvifMedia, which construct the pipeline manually instead of a launch string.
For the sake of testing, I've used the official Gstreamer one.

Here are two software encoder (openh264 and x264) launch string that works flawlessly:
```
videotestsrc is-live=true ! video/x-raw, format=YUY2, width=640, height=480, framerate=10/1 ! videoconvert !  openh264enc ! video/x-h264, width=640, height=480, framerate=10/1, profile=high, level=(string)4 ! rtph264pay name=pay0 pt=96
videotestsrc is-live=true ! video/x-raw, format=YUY2, width=640, height=480, framerate=10/1 ! videoconvert !  x264enc ! video/x-h264, width=640, height=480, framerate=10/1, profile=high, level=(string)4 ! rtph264pay name=pay0 pt=96
```

This one below is using v4l2h264enc and results with only a few frames.
```
videotestsrc is-live=true ! video/x-raw, format=YUY2, width=640, height=480, framerate=10/1 ! videoconvert !  v4l2h264enc extra-controls=\"controls,h264_profile=4,h264_level=11,video_bitrate=25000000;\"  ! video/x-h264, width=640, height=480, framerate=10/1, profile=high, level=(string)4 ! rtph264pay name=pay0 pt=96
```

Now, I understand that this bin is actually wrapped with appsrc/appsink by the GstRtspOnvifMedia class with additional handling, but the only difference in the pipeline is the v4l2h264enc element. I'm hoping that I'm just missing some mandatory config...
```
v4l2h264enc extra-controls=\"controls,h264_profile=4,h264_level=11,video_bitrate=25000000;\" 
```

When I get a chance, I will re-setup my RPi to include a display with x11 to debug locally.

Quedale, 2022-12-25:
Alright, I did better testing and I feel dumb. It seems my issue directly relates to [this old but still open issue](https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/1056)

This launch string DID worked successfully (Using videotestsrc):
```
videotestsrc is-live=true ! video/x-raw, format=YUY2, width=640, height=480, framerate=10/1 ! videoconvert !  v4l2h264enc extra-controls=\"encode,video_bitrate=25000000\" ! video/x-h264,profile=(string)high,level=(string)4,framerate=(fraction)10/1,width=640, height=480 ! h264parse ! rtph264pay name=pay0 pt=96
```
I must have made a mistake during my testing

I analyzed the logs more carefully using "v4lsrc" and "GST_DEBUG=v4l*:7" and noticed these :

v4l2 gstv4l2object.c:4212:gst_v4l2_object_set_format:<v4l2src0:src> Setting format to video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction)10/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, **colorimetry=(string)2:4:16:1**

v4l2videoenc gstv4l2videoenc.c:1022:gst_v4l2_video_enc_sink_query:<v4l2h264enc0> Returning sink caps video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)10/1, interlace-mode=(string)progressive, pixel-aspect-ratio=(fraction)1/1, format=(string)YUY2, **colorimetry=(string)bt709**

**I finally got it to work using "capssetter" as a workaround.**
Here's my successfully pipeline:
```
v4l2src ! video/x-raw, format=YUY2, width=640, height=480, framerate=10/1 ! videoconvert ! capssetter replace=true caps=\"video/x-raw,width=640,height=480,format=YUY2,framerate=10/1,colorimetry=bt709,interlace-mode=(string)progressive\" ! v4l2h264enc extra-controls=\"controls,h264_profile=4,h264_level=11,video_bitrate=25000000\" ! video/x-h264,profile=(string)high,level=(string)4,framerate=10/1,width=640, height=480 ! rtph264pay name=pay0 pt=96
```

Again, I never intended to hijack this issue, so I'll simply invite [berkutta](https://github.com/berkutta) to look into the colorimetry issue and try this workaround.

Cheers and Merry Xmas!

dp1140a, 2026-10-01:
We're seeing what looks like the same encoder stall on a CM4, on a newer stack and through FFmpeg rather than GStreamer.

  **Symptom:** the encoder stops returning capture buffers after an unpredictable time. The encoding thread then blocks forever in `ppoll()` with a NULL timeout, inside FFmpeg's `h264_v4l2m2m` dequeue.
  Nothing is logged in dmesg, and no vchiq/MMAL timeouts appear. Closing the device (or `VIDIOC_STREAMOFF` on both queues from another thread) unblocks it, and a newly opened encoder session works normally
  again. No reboot is needed.

  **Reproduces with stock ffmpeg, real camera content only:**
  - Real frames from an IMX477 via libcamera (NV21 1280x720, recorded and looped) stalled in 4 of 4 runs: at frame 623, 1639, 1631 and 39 (about 3–110 s).
  - A synthetic `testsrc2` source ran 5 minutes with no stall, using the same settings.
  - Changing `-num_capture_buffers 16 -num_output_buffers 32`, or using High instead of Constrained Baseline profile, did not help.

      ffmpeg -re -stream_loop -1 -f rawvideo -pix_fmt nv21 -s 1280x720 -r 15 -i cam_nv21_1280x720.raw \
        -vf scale=640:360,format=nv12 -c:v h264_v4l2m2m -b:v 800k -g 15 -bf 0 -profile:v 578 \
        -f h264 /dev/null

  **Environment:**
  - Raspberry Pi Compute Module 4 Rev 1.1, Ubuntu 24.04.4 LTS
  - Kernel 6.8.0-1047-raspi (Ubuntu), firmware `f4e2138c2adc8f3a92a3a65939e458f11d7298ba` (Feb 29 2024)
  - FFmpeg 6.1.1-3ubuntu5, `/dev/video11` bcm2835-codec-encode
  - config.txt: `dtoverlay=vc4-kms-v3d,cma-128`, `gpu_mem=76`

  Differences from the original report: we use FFmpeg rather than GStreamer, and kernel 6.8 rather than 5.15. We didn't measure memory during the stall, so we can't say whether it also grows.

6by9, 2026-10-05:
We don't support Ubuntu or the 6.8 kernel, but it is probably a common issue. FFmpeg 6.1.1 is also very old (Dec 2023).

I have a suspicion it'll be related to the bitrate as 800kbit/s is very low compared to the expected design. If rate control really overshoots, then it can legitimately encode a frame as nothing to drop it. That is generally where simple clients end up getting things wrong, particularly if they are relying on a 1-in, 1-out behaviour.