raspberrypi/linux #4911: wifi pwd stored in plaintext

Mayank-1234-cmd, 2022-02-27:
### Describe the bug

When I connect to WiFi, the WiFi password is stored in /etc/wpa_supplicant/wpa_supplicant.conf which is a security vulnerability that should be fixed.
If you need this:

Command used to encrypt password is
`wpa_supplicant [name] [password]` (If not working try `wpa_supplicant essid [name] key [password])
Also if using wpa_supplicant make sure to strip line 3, it's the password stored in plaintext (but commented).

### Steps to reproduce the behaviour

- Connect to WiFi
- Read the file /etc/wpa_supplicant/wpa_supplicant.conf
- It's stored in plaintext

### Device (s)

Raspberry Pi 4 Mod. B

### System

- OS, ver: Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage5
- Firmware nersion: version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
- Kernel version: Linux sdf64hostname 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux

pelwell, 2022-02-28:
Do you mean `wpa_passphrase`? `wpa_supplicant` is the daemon itself.

Using the obfuscated password is only a help against a lazy intruder - I just copied the obfuscated string into the WLAN password box on my phone and was accepted onto the network.

andrum99, 2022-02-28:
The system should not be storing passwords where non-root users can read it. As pelwell points out, obfuscating the password is pointless - the real solution is to make the password completely inaccessible to non-root users. Debian and Ubuntu both use Network Manager, which handles the wifi passwords, and prevent non-root users from access the files which contain the wifi passwords. Raspberry Pi OS does not use Network Manager.

Mayank-1234-cmd, 2022-02-28:
pelwell: I meant wpa_passphrase (and you are right, that doesn't help)
andrum99: *agrees*

pelwell, 2022-03-14:
This is not the right repo, and not the right solution. We may switch to using Network Manager in the future, but doing so will invalidate _a lot_ of tutorials.

closed 2022-03-14 by pelwell