raspberrypi/linux #4915: Kernel WARNING when starting USB sound card recording

mchazaux, 2022-03-01:
### Describe the bug

With an USB sound card : 

[   45.904710] usb 1-1.3: new high-speed USB device number 5 using dwc2
[   46.036328] usb 1-1.3: New USB device found, idVendor=2a39, idProduct=3fb0, bcdDevice= 0.01
[   46.044794] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   46.052155] usb 1-1.3: Product: Babyface Pro (73026971)
[   46.057459] usb 1-1.3: Manufacturer: RME
[   46.061421] usb 1-1.3: SerialNumber: 466682089F0E3C8
[   46.155245] usb 1-1.3: unit 2 not found!
[   46.188259] usbcore: registered new interface driver snd-usb-audio

The command : 

arecord -f  dat > /dev/null

gives a warning in the kernel log : 

[   59.608170] WARNING: CPU: 0 PID: 0 at drivers/usb/dwc2/hcd.c:2683 dwc2_assign_and_init_hc+0x860/0x900 [dwc2]

### Steps to reproduce the behaviour

1 - boot with or hotplug the USB sound card 
2 - arecord -f  dat > /dev/null
3 - hit CTRL+C to stop recording

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Kernel is 5.10.92-v7 #1 SMP Tue Mar 1 19:23:50 CET 2022 armv7l GNU/Linux

Compiled from commit ID 650082a559a570d6c9d2739ecc62843d6f951059 (from https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2022-01-28/2022-01-28-raspios-bullseye-armhf.info)

Userland has been built with buildroot.

### Logs

[   58.880214] ------------[ cut here ]------------
[   58.884925] WARNING: CPU: 0 PID: 417 at drivers/usb/dwc2/hcd.c:2683 dwc2_assign_and_init_hc+0x860/0x900 [dwc2]
[   58.894924] Modules linked in: snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device joydev rtc_ds1307 regmap_i2c vc4 cec drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops raspberrypi_ts brcmfmac brcmutil sha256_generic bcm2835_codec(C) bcm2835_isp(C) dwc2 bcm2835_v4l2(C) cfg80211 v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig rfkill videobuf2_vmalloc udc_core videobuf2_memops videobuf2_v4l2 roles raspberrypi_hwmon videobuf2_common i2c_bcm2835 videodev vc_sm_cma(C) mc rpi_backlight fixed uio_pdrv_genirq uio sch_fq_codel drm fuse drm_panel_orientation_quirks backlight ipv6
[   58.954489] CPU: 0 PID: 417 Comm: arecord Tainted: G         C        5.10.92-v7 #1
[   58.962144] Hardware name: BCM2835
[   58.965544] Backtrace:
[   58.968018] [<80a0e3b8>] (dump_backtrace) from [<80a0e740>] (show_stack+0x20/0x24)
[   58.975593]  r7:ffffffff r6:00000000 r5:60000193 r4:80fe61b0
[   58.981262] [<80a0e720>] (show_stack) from [<80a12a3c>] (dump_stack+0xc8/0xf4)
[   58.988494] [<80a12974>] (dump_stack) from [<8011f2f8>] (__warn+0xfc/0x114)
[   58.995462]  r10:00000001 r9:827424e0 r8:00000009 r7:00000a7b r6:7f5b91d0 r5:00000009
[   59.003292]  r4:7f5cfbf4 r3:80f05050
[   59.006875] [<8011f1fc>] (__warn) from [<80a0ed98>] (warn_slowpath_fmt+0x74/0xe8)
[   59.014360]  r7:7f5b91d0 r6:00000a7b r5:7f5cfbf4 r4:00000000
[   59.020082] [<80a0ed28>] (warn_slowpath_fmt) from [<7f5b91d0>] (dwc2_assign_and_init_hc+0x860/0x900 [dwc2])
[   59.029828]  r8:827424e0 r7:86e76c64 r6:86c0ff00 r5:82742040 r4:84038538
[   59.036628] [<7f5b8970>] (dwc2_assign_and_init_hc [dwc2]) from [<7f5ba94c>] (dwc2_hcd_select_transactions+0x5c/0x1a0 [dwc2])
[   59.047851]  r10:00000001 r9:827424e0 r8:82742458 r7:82742450 r6:00000000 r5:86c0ff7c
[   59.055680]  r4:82742040
[   59.058310] [<7f5ba8f0>] (dwc2_hcd_select_transactions [dwc2]) from [<7f5bdb20>] (dwc2_release_channel+0x180/0x214 [dwc2])
[   59.069360]  r10:00000001 r9:84ae3f00 r8:00000000 r7:84038480 r6:840384b8 r5:84038480
[   59.077189]  r4:82742040 r3:00000040
[   59.080861] [<7f5bd9a0>] (dwc2_release_channel [dwc2]) from [<7f5be21c>] (dwc2_complete_periodic_xfer+0x6c/0x7c [dwc2])
[   59.091647]  r7:84038480 r6:86e76c40 r5:00000000 r4:82742040
[   59.097403] [<7f5be1b0>] (dwc2_complete_periodic_xfer [dwc2]) from [<7f5bed68>] (dwc2_hc_xfercomp_intr+0x308/0x450 [dwc2])
[   59.108543] [<7f5bea60>] (dwc2_hc_xfercomp_intr [dwc2]) from [<7f5bf95c>] (dwc2_hc_n_intr+0x4e8/0x838 [dwc2])
[   59.118463]  r10:86e76c64 r9:86e76c40 r8:00000000 r7:0000050c r6:84038480 r5:00000002
[   59.126291]  r4:82742040
[   59.128920] [<7f5bf474>] (dwc2_hc_n_intr [dwc2]) from [<7f5bfe80>] (dwc2_handle_hcd_intr+0x1d4/0x6ac [dwc2])
[... 46 lines cut ...]
[   59.468154]  r10:91887ac8 r9:00000096 r8:91887ac8 r7:00000020 r6:849f7e18 r5:827ca428
[   59.475982]  r4:00000000
[   59.478523] [<802bcec0>] (page_cache_async_ra) from [<802b28d8>] (filemap_fault+0x61c/0xaec)
[   59.486964]  r7:00000096 r6:849f7ec8 r5:b9650584 r4:918879b8
[   59.492633] [<802b22bc>] (filemap_fault) from [<8042f848>] (ext4_filemap_fault+0x38/0x4c)
[   59.500817]  r10:849f7ec8 r9:00000d73 r8:00000040 r7:849f7fb0 r6:00000000 r5:849f7ec8
[   59.508645]  r4:918879a0
[   59.511188] [<8042f810>] (ext4_filemap_fault) from [<802ea4e8>] (__do_fault+0x48/0x150)
[   59.519192]  r5:90cc1de0 r4:849f7ec8
[   59.522778] [<802ea4a0>] (__do_fault) from [<802ef5f8>] (handle_mm_fault+0x9c8/0xdf4)
[   59.530611]  r7:849f7fb0 r6:00000000 r5:76e9f000 r4:00000254
[   59.536281] [<802eec30>] (handle_mm_fault) from [<80a1e148>] (do_page_fault+0x144/0x314)
[   59.544378]  r10:00000254 r9:00000000 r8:83fdfa44 r7:83fdfa00 r6:80000007 r5:76e9f324
[   59.552210]  r4:849f7fb0
[   59.554761] [<80a1e004>] (do_page_fault) from [<8011451c>] (do_PrefetchAbort+0x48/0xa0)
[   59.562777]  r10:00000000 r9:00000001 r8:80a1e004 r7:849f7fb0 r6:76e9f324 r5:80f0b144
[   59.570610]  r4:00000007
[   59.573154] [<801144d4>] (do_PrefetchAbort) from [<80101024>] (ret_from_exception+0x0/0x1c)
[   59.581506] Exception stack(0x849f7fb0 to 0x849f7ff8)
[   59.586560] 7fa0:                                     01cc1e40 7ea29494 00000003 7ea294bd
[   59.594743] 7fc0: 01cc1db0 00000003 7ea29494 01cc1e40 76ee20ec 00000001 00000000 01cc1dc0
[   59.602924] 7fe0: 76ee2484 7ea29360 76e95c74 76e9f324 60000010 ffffffff
[   59.609544]  r8:10c5387d r7:10c5383d r6:ffffffff r5:60000010 r4:76e9f324
[   59.616245] ---[ end trace d37406223ac2a2d3 ]---
[   64.624623] retire_capture_urb: 39986 callbacks suppressed

P33M, 2022-03-01:
Usage of the dwc2 driver in host mode on Pi 1, 2 and 3 is not recommended (and not supported). What happens if you use dwc_otg?

mchazaux, 2022-03-01:
Hello, 

Thanks for the tip. 
I removed the following line from my config.txt : 

dtoverlay=dwc2,dr_mode=host

The warning is gone.

I added it for the Pi compute module and as it did not seem to affect the Pi 3 I left it. My mistake, sorry !

Mickael

closed 2022-03-01 by mchazaux