raspberrypi/linux #4917: h264_v4l2m2m and ffmpeg and anomalous settings/results

hydra3333, 2022-03-03:
### Describe the bug

Hello. I am attempting to use 64bit ffmpeg (built from latest git) on a Raspberry Pi 4 (aarch64) and am seeing an unexpected result.  The input is avc interlaced 1080i25 (PAL country) constant frame rate.

I am not sure if the issue is here, or over at ffmpeg.

The ffmpeg command below should deinterlace and transcode into progressive avc using the Pi's h264_v4l2m2m hardware acceleration.
```/usr/local/bin/ffmpeg -hide_banner -v verbose \
		-i "./some_test_input_file.mp4" \
		-vsync cfr \
		-sws_flags lanczos+accurate_rnd+full_chroma_int+full_chroma_inp \
		-strict experimental \
		-filter_complex "[0:v]yadif=0:0:0,format=pix_fmts=yuv420p" \
		-c:v h264_v4l2m2m \
		-preset slow \
		-pix_fmt yuv420p \
		-qp -1 \
		-g:v 25 \
		-b:v 4000000 -minrate:v 500000 -maxrate:v 5000000 -bufsize:v 5000000 \
		-level 5.2 \
		-movflags +faststart+write_colr \
		-an \
		-y "./some_test_input_file_transcoded_h264_v4l2m2m.mp4" 2>&1 | tee ff.log```

The resulting .mp4 is flagged as VFR which is not what I expected nor wanted.
However:

1. The gop setting is ignored:   
```[code][h264_v4l2m2m @ 0x55942ed400] Failed to set gop size: Invalid argument[/code]```

2. The Level 5.2 setting is ignored (I suspect the maximum may be L4).

3. I was hoping to use VBR rather than CBR but can't find a way to set it.

mediainfo on the resulting .mp4 says:
```Format profile : High at L4
Format settings, GOP : M=1, N=60
Bit rate mode : CBR
Bit rate mode : Constant
Frame rate mode : VFR
Frame rate mode : Variable```

I am happy to try to use long-obsolete programming skills to attempt a patch, if I know where to start looking. 
(A link to linux call interface documentation for this would be handy, unless it's mentioned someplace in the relevant lunix code).

### Steps to reproduce the behaviour

```/usr/local/bin/ffmpeg -hide_banner -v verbose \
		-i "./some_test_input_file.mp4" \
		-vsync cfr \
		-sws_flags lanczos+accurate_rnd+full_chroma_int+full_chroma_inp \
		-strict experimental \
		-filter_complex "[0:v]yadif=0:0:0,format=pix_fmts=yuv420p" \
		-c:v h264_v4l2m2m \
		-preset slow \
		-pix_fmt yuv420p \
		-qp -1 \
		-g:v 25 \
		-b:v 4000000 -minrate:v 500000 -maxrate:v 5000000 -bufsize:v 5000000 \
		-level 5.2 \
		-movflags +faststart+write_colr \
		-an \
		-y "./some_test_input_file_transcoded_h264_v4l2m2m.mp4" 2>&1 | tee ff.log```

### Device (s)

Raspberry Pi 4 Mod. B

### System

```cat /etc/rpi-issue
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbe448ccdc995d295d24c7596e5f0ef62cc2488f, stage4

vcgencmd version
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

uname -a
Linux PI4B8Gb 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux```

### Logs

nil

### Additional context

```/usr/local/bin/ffmpeg -version
ffmpeg version git-2022-02-26-df27292-ffmpeg_for_RPi4B_having_h264_v4l2m2m Copyright (c) 2000-2022 the FFmpeg developers
built with gcc 10 (Debian 10.2.1-6)
configuration: --extra-version=ffmpeg_for_RPi4B_having_h264_v4l2m2m --disable-shared --enable-static --enable-pic --enable-neon --disable-w32threads --enable-pthreads --enable-gpl --enable-version3 --enable-nonfree --prefix=/usr/local --libdir=/usr/local/lib --bindir=/usr/local/bin --extra-cflags=' -I/usr/local/include -I/usr/include/aarch64-linux-gnu -I/usr/include -L/usr/local/lib -L/usr/lib/aarch64-linux-gnu -L/usr/lib ' --extra-ldflags=' -I/usr/local/include -I/usr/include/aarch64-linux-gnu -I/usr/include -L/usr/local/lib -L/usr/lib/aarch64-linux-gnu -L/usr/lib ' --extra-libs='-lpthread -lm -latomic' --pkg-config=pkg-config --pkg-config-flags=--static --disable-doc --disable-htmlpages --disable-manpages --disable-podpages --disable-txtpages --disable-avisynth --disable-vapoursynth --disable-libkvazaar --disable-schannel --enable-v4l2-m2m --enable-hardcoded-tables --enable-gray --enable-gmp --enable-gnutls --enable-iconv --enable-libaom --enable-libass --enable-libdav1d --enable-libdrm --enable-libfdk-aac --enable-libmp3lame --enable-libtwolame --enable-libfreetype --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libssh --enable-libvorbis --enable-libvpx --enable-libzimg --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-librubberband --enable-libwebp --enable-zlib --enable-lzma --extra-cflags=-DLIBTWOLAME_STATIC --extra-cflags=-DLIBXML_STATIC```

6by9, 2022-03-03:
```
pi@raspberrypi:~ $ v4l2-ctl --list-ctrls-menu -d 11

Codec Controls

             video_bitrate_mode 0x009909ce (menu)   : min=0 max=1 default=0 value=0 flags=update
				0: Variable Bitrate
				1: Constant Bitrate
                  video_bitrate 0x009909cf (int)    : min=25000 max=25000000 step=25000 default=10000000 value=10000000
           sequence_header_mode 0x009909d8 (menu)   : min=0 max=1 default=1 value=1
				0: Separate Buffer
				1: Joined With 1st Frame
         repeat_sequence_header 0x009909e2 (bool)   : default=0 value=0
                force_key_frame 0x009909e5 (button) : flags=write-only, execute-on-write
          h264_minimum_qp_value 0x00990a61 (int)    : min=0 max=51 step=1 default=20 value=20
          h264_maximum_qp_value 0x00990a62 (int)    : min=0 max=51 step=1 default=51 value=51
            h264_i_frame_period 0x00990a66 (int)    : min=0 max=2147483647 step=1 default=60 value=60
                     h264_level 0x00990a67 (menu)   : min=0 max=13 default=11 value=11
				0: 1
				1: 1b
				2: 1.1
				3: 1.2
				4: 1.3
				5: 2
				6: 2.1
				7: 2.2
				8: 3
				9: 3.1
				10: 3.2
				11: 4
				12: 4.1
				13: 4.2
                   h264_profile 0x00990a6b (menu)   : min=0 max=4 default=4 value=4
				0: Baseline
				1: Constrained Baseline
				2: Main
				4: High
```
So the maximum supported level is 4.2, although we only guarantee level 4 in real-time. Level 4 is 1080p30, and level 4.2 1080p60, so level 5.2 is excessive for your encode.

FFmpeg tries to use [V4L2_CID_MPEG_VIDEO_GOP_SIZE](https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/v4l2_m2m_enc.c#L199). 
For H264 there is V4L2_CID_MPEG_VIDEO_H264_I_PERIOD, which is what is implemented in the encoder.
Spec - https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/ext-ctrls-codec.html
```
Period between I-frames in the open GOP for H264. In case of an open GOP this is the period between two I-frames. The period between IDR (Instantaneous Decoding Refresh) frames is taken from the GOP_SIZE control. 
```
I'd need to check if we're creating IDR frames or I frames.

I'd need to know what properties of the elementary stream MediaInfo is looking at to determine the bit rate and frame rate modes.

Unless the codec has totally expended all the bits within rate control, then it will not drop frames, and it will not change the frame rate or timestamps. An ES has no inherent frame rate as timestamps are carried in the container.
What MediaInfo may be looking for is the timing information in the SPS header, which we generally omit as it is typically inaccurate.
I'm not aware of an SPS field to denote the rate control mode, so again it depends on what MediaInfo is using to determine that.

I'm surprised that yadif is fast enough to keep up at 1080i50, but I'm happy if it works for you.

hydra3333, 2022-03-03:
Thanks !

> the maximum supported level is 4.2, although we only guarantee level 4 in real-time. Level 4 is 1080p30, and level 4.2 1080p60, so level 5.2 is excessive for your encode.

Would that be a hardware/driver limitation or could it validly go up to 19 (V4L2_MPEG_VIDEO_H264_LEVEL_6_2 = 19) if the driver allowed it ?

> FFmpeg tries to use [V4L2_CID_MPEG_VIDEO_GOP_SIZE](https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/v4l2_m2m_enc.c#L199).
> For H264 there is V4L2_CID_MPEG_VIDEO_H264_I_PERIOD, which is what is implemented in the encoder.
> Spec - https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/ext-ctrls-codec.html
>  
> I'd need to know what properties of the elementary stream MediaInfo is looking at to determine the bit rate and frame rate modes.

OK. Thank you.  Other conversion tools end up reporting CFR on their resulting files.

> I'm surprised that yadif is fast enough to keep up at 1080i50, but I'm happy if it works for you.

It's happily speedy, and I'm not looking for realtime encoding.

hydra3333, 2022-03-03:
```
/usr/local/bin/ffprobe -hide_banner -probesize 200M -analyzeduration 200M -show_format  -show_private_data -i "./some_test_input_file_transcoded_h264_v4l2m2m.mp4"
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5599fbab00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 0
[h264 @ 0x5599fbb770] missing picture in access unit with size 50
[h264 @ 0x5599fbb770] no frame!
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from './some_test_input_file_transcoded_h264_v4l2m2m.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf59.17.102
  Duration: 00:00:05.96, start: 0.000000, bitrate: 3169 kb/s
  Stream #0:0[0x1](und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, progressive), 1920x1080, 3167 kb/s, SAR 1:1 DAR 16:9, 25.17 fps, 25 tbr, 12800 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
[FORMAT]
filename=./some_test_input_file_transcoded_h264_v4l2m2m.mp4
nb_streams=1
nb_programs=0
format_name=mov,mp4,m4a,3gp,3g2,mj2
format_long_name=QuickTime / MOV
start_time=0.000000
duration=5.960000
size=2361294
bit_rate=3169522
probe_score=100
TAG:major_brand=isom
TAG:minor_version=512
TAG:compatible_brands=isomiso2avc1mp41
TAG:encoder=Lavf59.17.102
[/FORMAT]
```

25.17 fps probably means it's VFR ...

hydra3333, 2022-03-03:
However
```
/usr/local/bin/ffprobe -hide_banner -probesize 200M -analyzeduration 200M -show_streams -show_private_data -i "./some_test_input_file_transcoded_h264_v4l2m2m.mp4"
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x556c50ab00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 0
[h264 @ 0x556c50b770] missing picture in access unit with size 50
[h264 @ 0x556c50b770] no frame!
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from './some_test_input_file_transcoded_h264_v4l2m2m.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf59.17.102
  Duration: 00:00:05.96, start: 0.000000, bitrate: 3169 kb/s
  Stream #0:0[0x1](und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, progressive), 1920x1080, 3167 kb/s, SAR 1:1 DAR 16:9, 25.17 fps, 25 tbr, 12800 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
[STREAM]
index=0
codec_name=h264
codec_long_name=H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
profile=High
codec_type=video
codec_tag_string=avc1
codec_tag=0x31637661
width=1920
height=1080
coded_width=1920
coded_height=1080
closed_captions=0
film_grain=0
has_b_frames=0
sample_aspect_ratio=1:1
display_aspect_ratio=16:9
pix_fmt=yuv420p
level=40
color_range=tv
color_space=unknown
color_transfer=unknown
color_primaries=unknown
chroma_location=left
field_order=progressive
refs=1
is_avc=true
nal_length_size=4
id=0x1
r_frame_rate=25/1
avg_frame_rate=3750/149
time_base=1/12800
start_pts=0
start_time=0.000000
duration_ts=76288
duration=5.960000
bit_rate=3167448
max_bit_rate=N/A
bits_per_raw_sample=8
nb_frames=150
nb_read_frames=N/A
nb_read_packets=N/A
extradata_size=57
DISPOSITION:default=1
DISPOSITION:dub=0
DISPOSITION:original=0
DISPOSITION:comment=0
DISPOSITION:lyrics=0
DISPOSITION:karaoke=0
DISPOSITION:forced=0
DISPOSITION:hearing_impaired=0
DISPOSITION:visual_impaired=0
DISPOSITION:clean_effects=0
DISPOSITION:attached_pic=0
DISPOSITION:timed_thumbnails=0
DISPOSITION:captions=0
DISPOSITION:descriptions=0
DISPOSITION:metadata=0
DISPOSITION:dependent=0
DISPOSITION:still_image=0
TAG:language=und
TAG:handler_name=VideoHandler
TAG:vendor_id=[0][0][0][0]
[/STREAM]
```
The 2nd of these 2 also seem to indicate VFR (the input file is CFR 25fps)
```
r_frame_rate=25/1
avg_frame_rate=3750/149
```

hydra3333, 2022-03-03:
Anyway, thank you for your reply !

popcornmix, 2022-03-03:
> I'm surprised that yadif is fast enough to keep up at 1080i50, but I'm happy if it works for you.

It's being run in the non-frame doubling mode (`send_frame` rather than `send_field`, so 25p output rather than 50p).

hydra3333, 2022-03-03:
OK cool. 
That gives me a really great leg-up to look into it. 
Thanks.

6by9, 2022-03-09:
You've now moved the discussion to https://forums.raspberrypi.com/viewtopic.php?p=1981727, so closing.

closed 2022-03-09 by 6by9

sliterok, 2022-10-01:
> You've now moved the discussion to https://forums.raspberrypi.com/viewtopic.php?p=1981727, so closing.

Hi. I see that this discussion is now locked. I'm new to this theme and just want to check if mental image that I got from reading this (and similar) threads is correct in my head:
There is **NO** way to encode YUV420 raw stream (or any other) of 1080p 60fps using hardware acceleration with RPI4 and x64 system?

pelwell, 2024-02-13:
What's not funny is having a LOL post on a thread that's been close for almost two years, but I bet you're going to find this hysterical - locking.