raspberrypi/linux #4919: Compile problem due to patch from raspberry pi

laroche, 2022-03-03:
### Describe the bug

I am compiling a Debian kernel with all patches from raspberry pi applied.

This breaks since a few kernel revisions due to the following patch:
  From 1a5ac44f2640e1e5f184dea59456644a8fe6854e Mon Sep 17 00:00:00 2001
  From: Dave Stevenson <dave.stevenson@raspberrypi.com>
  Date: Tue, 1 Feb 2022 12:20:20 +0000
  Subject: [PATCH 599/667] drm/panel: Add and initialise an orientation field to drm_panel

The following quick fix seems to cure the compile problems for me:

<pre>
--- a/drivers/gpu/drm/msm/disp/mdp4/mdp4_kms.h
+++ b/drivers/gpu/drm/msm/disp/mdp4/mdp4_kms.h
@@ -7,6 +7,7 @@
 #ifndef __MDP4_KMS_H__
 #define __MDP4_KMS_H__
 
+#include <drm/drm_connector.h>
 #include <drm/drm_panel.h>
 
 #include "msm_drv.h"
--- a/drivers/gpu/drm/msm/dp/dp_display.c
+++ b/drivers/gpu/drm/msm/dp/dp_display.c
@@ -10,6 +10,7 @@
 #include <linux/component.h>
 #include <linux/of_irq.h>
 #include <linux/delay.h>
+#include <drm/drm_connector.h>
 #include <drm/drm_panel.h>
 
 #include "msm_drv.h"
</pre>

Can you please review this and include this in your rpi patches for kernel 5.16 and 5.17?
Instead of these includes, maybe you want to move the definition of "enum drm_panel_orientation"
from drom_connector.h to drm_panel.h instead?

Thanks a lot,
best regards from Germany,

Florian La Roche

### Steps to reproduce the behaviour

Cross-recompile Debian kernel together with raspberry pi patches applied.

### Device (s)

Other

### System

arm64 build, crosscompile from amd64

pelwell, 2022-03-03:
An example of a set of commands that causes a build failure on an unmodified rpi-5.16.y would be helpful.

6by9, 2022-03-03:
```enum drm_panel_orientation``` can't be moved from drm_connector.h to drm_panel.h as it is used within drm_connector.h.

```#include <drm/drm_connector.h>``` from drm_panel.h is probably the better option, and drop the forward definition ```enum drm_panel_orientation;```. I suspect that mainline have tried to avoid having to cross include header files.

laroche, 2022-03-03:
A complete script doing this is available as:
https://github.com/laroche/arm-devel-infrastructure/blob/master/vmdb2-debian/kernel-5.16.sh

It does a few sudo calls to install a compile-environment. Then downloads the current Debian sources,
the current raspberry pi patches and compiles all of that. Only this one script is needed, no other file from this
repository.

Agreed, to 6by9: an include instead of the forward definition might be the best.

best regards,

Florian La Roche

pelwell, 2022-03-03:
> A complete script doing this is available as:

Hmm - that's not what I asked for - there must be a pair of config +build commands that demonstrates the bug, e.g.:
```
$ make ARCH=arm64 what_defconfig
$ make ARCH=arm64 Image modules
```