raspberrypi/linux #4938: H264 m2m encoder can't be started with 25000 kbps bitrate

mdevaev, 2022-03-14:
### Describe the bug

The sources indicate that the maximum bitrate is [25000 kbps](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c#L3147), but this does not work and leads to an error on VIDIOC_STREAMON (No such process). At the same time, 20000 kbps work.

### Steps to reproduce the behaviour

1. Use tc358743 and ustreamer.
2. Run capturing `ustreamer --dv-timings --persistent --format=uyvy --encoder=noop --workers=1 --verbose --h264-sink=test --h264-bitrate=25000`
3. See the error: `-- ERROR [9585.330    stream] -- H264: Can't start INPUT: No such process`

### Device (s)

Raspberry Pi 4 Mod. B

### System

Arch Linux ARM
```
Mar  1 2022 14:22:32
Copyright (c) 2012 Broadcom
version 969fb9b1521fc7ac2b88b15a3a9e942da7678c4d (clean) (release) (start_x)
```
`Linux pikvm 5.15.25-8-rpi-ARCH #1 SMP Tue Mar 15 06:33:22 MSK 2022 armv7l GNU/Linux`

### Logs

```
[ 9297.965563] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -3
[ 9297.975122] ------------[ cut here ]------------
[ 9297.979761] WARNING: CPU: 0 PID: 6194 at drivers/media/common/videobuf2/videobuf2-core.c:1552 vb2_start_streaming+0xe4/0x16c [videobuf2_common]
[ 9297.992770] Modules linked in: usb_f_mass_storage usb_f_hid btsdio bluetooth ecdh_generic ecc rtc_pcf8563 spidev brcmfmac brcmutil cfg80211 raspberrypi_hwmon i2c_mux_pinctrl bcm2835_unicam snd_soc_simple_card snd_soc_simple_card_utils bcm2835_codec(C) bcm2835_isp(C) i2c_mux bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) i2c_bcm2835 videobuf2_dma_contig videobuf2_vmalloc iproc_rng200 snd_soc_bcm2835_i2s rng_core videobuf2_memops spi_bcm2835 videobuf2_v4l2 videobuf2_common snd_soc_spdif_rx snd_soc_core rpivid_mem snd_compress snd_pcm_dmaengine uio_pdrv_genirq uio nvmem_rmem sch_fq_codel snd_bcm2835(C) snd_pcm snd_timer snd i2c_dev tc358743 v4l2_dv_timings v4l2_fwnode v4l2_async videodev mc cec libcomposite dwc2 roles fuse drm drm_panel_orientation_quirks ip_tables x_tables ipv6
[ 9298.063365] CPU: 0 PID: 6194 Comm: stream Tainted: G         C        5.15.25-8-rpi-ARCH #1
[ 9298.073241] Hardware name: BCM2711
[ 9298.077390] [<c020ef38>] (unwind_backtrace) from [<c020abb4>] (show_stack+0x10/0x14)
[ 9298.086634] [<c020abb4>] (show_stack) from [<c0e411c4>] (dump_stack_lvl+0x50/0x68)
[ 9298.095742] [<c0e411c4>] (dump_stack_lvl) from [<c022b52c>] (__warn+0xec/0x148)
[ 9298.104606] [<c022b52c>] (__warn) from [<c0e3931c>] (warn_slowpath_fmt+0x64/0xc0)
[ 9298.113677] [<c0e3931c>] (warn_slowpath_fmt) from [<bf2e6a3c>] (vb2_start_streaming+0xe4/0x16c [videobuf2_common])
[ 9298.125666] [<bf2e6a3c>] (vb2_start_streaming [videobuf2_common]) from [<bf2e6b44>] (vb2_core_streamon+0x80/0x170 [videobuf2_common])
[ 9298.139366] [<bf2e6b44>] (vb2_core_streamon [videobuf2_common]) from [<bf2d7ffc>] (v4l2_m2m_ioctl_streamon+0x30/0x70 [v4l2_mem2mem])
[ 9298.152985] [<bf2d7ffc>] (v4l2_m2m_ioctl_streamon [v4l2_mem2mem]) from [<bf1c3604>] (__video_do_ioctl+0x22c/0x454 [videodev])
[ 9298.166066] [<bf1c3604>] (__video_do_ioctl [videodev]) from [<bf1c45b0>] (video_usercopy+0x2ec/0x670 [videodev])
[ 9298.178047] [<bf1c45b0>] (video_usercopy [videodev]) from [<c045d68c>] (sys_ioctl+0x484/0xa74)
[ 9298.188405] [<c045d68c>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x40)
[ 9298.197665] Exception stack(0xc3a99fa8 to 0xc3a99ff0)
[ 9298.203552] 9fa0:                   00000009 b46e6908 00000009 40045612 b46e65d4 00000000
[ 9298.213399] 9fc0: 00000009 b46e6908 00000004 00000036 b46e65d4 b46e6770 00000000 c0cc5616
[ 9298.223262] 9fe0: b6cc7e58 b46e6560 00473f9c b6c36c00
[ 9298.229186] ---[ end trace 24f221837a36233b ]---
```

6by9, 2022-03-15:
Did you increase the H264 level? https://en.wikipedia.org/wiki/H.264/MPEG-4_AVC#Levels
The default is level 4.0, so a max of 20000kbit/s, but levels 4.1 and 4.2 are also selectable.

mdevaev, 2022-03-15:
No, my fault, thanks for the info!

closed 2022-03-15 by mdevaev