raspberrypi/linux #4944: rpi-5.15.y branch is frequently rebased

kraj, 2022-03-17:
### Describe the bug

Repo is defaulting to `rpi-5.15.y` branch on checkout, but the branch gets rebased frequently, in Yocto we upgraded to use 5.15 as default kernel for rpi machines but we lock to a known revision and rebase breaks that since the SHAs get re-written
so my question is, whats the ETA for this branch to become publicly consumable ?

### Steps to reproduce the behaviour

build yocto after one week of kernel SRCREV bump.

### Device (s)

Raspberry Pi 4 Mod. B

### System

cant find the revision to checkout see
https://github.com/agherzan/meta-raspberrypi/issues/1037#event-6262566909

popcornmix, 2022-03-23:
We are close to switching to 5.15 for apt, so there will be no more rebases.
Next upstream bump will be a merge.

kraj, 2022-03-23:
thanks @popcornmix

closed 2022-03-23 by pelwell