raspberrypi/linux #4946: DSI0 not reliable (Compute Modules only)

6by9, 2022-03-19:
### Describe the bug

Tracking work raised by https://forums.raspberrypi.com/viewtopic.php?t=331474

DSI0 (only exposed on Compute Modules) seems to have issues when used with vc4-kms-v3d such as the colour channels getting swapped when stopping and restarting.

This has been confirmed with a DSI analyser, so it is something within the SoC, not just one display.

### Steps to reproduce the behaviour

Configure a DSI display on DSI0 and then use xrandr to disable and reenable the display. Observe occasional colour shift (eg RGB to GBR to BRG)

### Device (s)

Raspberry Pi CM1, Raspberry Pi CM3, Raspberry Pi CM3 Lite, Raspberry Pi CM3+, Raspberry Pi CM3+ Lite, Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

5.15 kernel.

6by9, 2022-06-08:
FWIW https://github.com/6by9/linux/tree/rpi-5.15.y-dsi has my changes in it, plus some debug stuff. 
Testing has been on a CM3, but the same thing was observed on both CM3 and CM4 before.

aBUGSworstnightmare-rpi, 2024-04-29:
Hi, sorry, but can't offer support any more on topics related to the official Pi 7" panel as I don't have one in my possession any more.

aBUGSworstnightmare-rpi, 2024-04-29:
Will do some tests with DSI83 on DSI0 though!

aBUGSworstnightmare-rpi, 2024-04-29:
> Will do some tests with DSI83 on DSI0 though!
>
> It was more your wider testing of DSI devices on DSI0 that I was hoping
> for. The DSI0 fix should be generic.
>
> The fixes for the Pi panel are more that I've finally had time to sit down
> and test out all the combinations, and fiddling to determine the quirks. I
> really don't understand what that Toshiba bridge chip is doing under some
> circumstances, and why it wants a totally different timing when connected
> to DSI0 is baffling. The other panels have all just taken the DSI output,
> and the analyser doesn't show anything obviously different between DSI0 and
> DSI1.

I will pull your rpi-6.6.y-dsi0 repro, compile a custom kernel (as I need
to have 3 modules which are not part of the def_config) and then report
back.

Joerg

aBUGSworstnightmare-rpi, 2024-04-29:
running the Nexus7 panel on 2-lane DSI0 (CM4+CM4IO) with no issues found so far. 
Blanks and resumes from blanking with no issues seen so far. 
Forcing it on/off seems fine so far as well

`WAYLAND_DISPLAY="wayland-1" wlr-randr --output LVDS-1 --off
WAYLAND_DISPLAY="wayland-1" wlr-randr --output LVDS-1 --on`

6by9, 2025-01-03:
All issues believed resolved, so closing.

closed 2025-01-03 by 6by9