raspberrypi/linux #4950: "Abort failed to stop command ring: -110" when device pluged into custom CM4 PCIe-USB Bridge

MarsTechHAN, 2022-03-21:
### Describe the bug

We are working on a custom design for Raspberry Pi Compute Module 4. For a number of reasons, we have to use a PCIe to USB3.1 bridge from Asmedia. The part number is ASM3042.

The link between the PCIe/usb bridge seems to work, and we can see the root hub presented with the lsusb. 
[image]

However, when we plug in any USB device whether it is USB3 or USB2. The following message shows up and the system halts. 
```
[   42.086253] xhci_hcd 0000:01:00.0: Abort failed to stop command ring: -110
[   42.086285] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[   42.086295] xhci_hcd 0000:01:00.0: HC died; cleaning up
[   42.086429] xhci_hcd 0000:01:00.0: Error while assigning device slot ID
[   42.086436] xhci_hcd 0000:01:00.0: Max number of devices this xHCI host supports is 127.
[   42.086446] usb usb1-port1: couldn't allocate usb_device
```

We tried the newest kernel on the rpi-5.15.y branch and there is no difference. 

The design includes over current protection, and we measure the output voltage of the USB hub, there is no overcurrent events that happened during the insertion of the USB device.

### Steps to reproduce the behaviour

1. Have a Raspberry Pi CM4 with I/O board.
2. Add ```dtoverlay=pcie-32bit-dma``` to the /boot/config.txt
3. Plug in the PCIe-USB3.1 Bridge Board
4. Power up the CM4 board.
5. When the device booted, plug in any USB device.

### Device (s)

Raspberry Pi CM4

### System

vcgencmd version
```
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
```

uname -a
```
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
```

cat /etc/rpi-issue 
```
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fbe448ccdc995d295d24c7596e5f0ef62cc2488f, stage4
```

cmdline.txt
```
console=serial0,115200 console=tty1 root=PARTUUID=3c215ad9-02 rootfstype=ext4 fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles
```
config.txt
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
[... 741 lines cut ...]
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover+ Timeout+ AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [200 v1] Secondary PCI Express
		LnkCtl3: LnkEquIntrruptEn- PerformEqu-
		LaneErrStat: 0
	Capabilities: [300 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [400 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=0us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=23552ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: xhci_hcd

```

pelwell, 2022-03-21:
Something is killing the VPU - under-voltage perhaps - and the XHCI controller is similarly unhappy. I don't think this is a software issue.

MarsTechHAN, 2022-03-21:
Thanks for the idea. I notice there is some unusual stuff in the dmesg I posted. It only happened once. The power supply should be able to provide enough power to the whole system (5V 6A with 12V 3A input). We will take a measurement of the voltage drops on the pins.

MarsTechHAN, 2022-03-22:
[image]
We directly solder a pair of power wires to the socket of the CM4, and connect it to a Keithley 2231A PSU. Powered the VBUS by another PSU, connect the USB device into the socket, and we got exactly the same dmesg message.
```
[   47.615676] hwmon hwmon1: Failed to get throttled (-110)
[   50.790601] xhci_hcd 0000:01:00.0: Abort failed to stop command ring: -110
[   50.790665] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[   50.790691] xhci_hcd 0000:01:00.0: HC died; cleaning up
[   50.791035] xhci_hcd 0000:01:00.0: Error while assigning device slot ID
[   50.791059] xhci_hcd 0000:01:00.0: Max number of devices this xHCI host supports is 127.
[   50.791082] usb usb1-port1: couldn't allocate usb_device
```

For the testing result, I felt like the chance it was purely caused by power is pretty low.

pelwell, 2022-03-22:
The consistency of the error doesn't sound so much like a power problem. My debugging approach would be try to strip the system back to something that works, then add stuff until it breaks again. That might be difficult from a hardware perspective unless you have the ASMedia XHCI controller on a PCI card that you can plug into the CM4IO board. Unplug any USB devices perhaps?

It should be possible to disable the PCIe controller with a custom dtoverlay, but you could start by locating the driver for the ASMedia device and renaming the module so it isn't found. Except it's probably part of the standard XHCI driver, which is built in. It looks like you'll have to resort to an overlay or custom build of the base .dtb.

MarsTechHAN, 2022-03-22:
It is using the standard xHCI driver. We order an ASMeida PCIe card, and it should arrive today. 

I noticed similar things happened with [mainline Linux kernel](https://bbs.archlinux.org/viewtopic.php?id=270795), but for rpi, both 5.10 kernel and 5.15 kernel should not contain such problem.

MarsTechHAN, 2022-03-24:
Hi. After playing with the off-the-shelf ASMedia PCIe bridge card and Raspberry IO board, I think the kernel works perfectly without any error. Thanks for all the help, and I will close the issue.

For detailed reasons:
1. Why did VPU fail? The SMT factory uses the wrong inductor (SWPA series instead of WPN series from Sunlord, which has much lower saturation current), causing voltage drops when the raspberry pi requires sudden inrush current. We also add a 100uF POSCAP to further eliminate the power issue. 
2. Why did ASMedia fail? For some reason, the data in the flash is corrupt, and ASM3042 failed to load the firmware. We experience the same behavior when we remove the flash memory from the off-the-shelf PCIe card. We suspect that may be related to the unstable power supply. It is working now after we fix the first issue, and re-flash the firmware.

closed 2022-03-24 by MarsTechHAN

pelwell, 2022-03-24:
Thanks for reporting back your findings - I'm happy you have a solution.

tikiss, 2022-11-11:
@MarsTechHAN could you please tell me which inductor caused that problem? For which power supply? 5V for the CM4?
I am asking because, with my custom board, I have quite a similar problem with the ASM3042.
If I do not flash the Serial NOR Flash for the ASM3042, the CM4 is booting normally, but after booting, I run `dmesg` I see exactly the same log:
```
[   50.790601] xhci_hcd 0000:01:00.0: Abort failed to stop command ring: -110
[   50.790665] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[   50.790691] xhci_hcd 0000:01:00.0: HC died; cleaning up
[   50.791035] xhci_hcd 0000:01:00.0: Error while assigning device slot ID
[   50.791059] xhci_hcd 0000:01:00.0: Max number of devices this xHCI host supports is 127.
```
If I flash the serial NOR flash with the FW which was provided by ASMedia, the CM4 is not booting, it hangs:
[image]
I have tried many things in the HW to get it running, but nothing helped so far.
Would be great if you, or somebody else could give me some hints.