raspberrypi/linux #4956: Jackd doesn't work with vc4hdmi

vanfanel, 2022-03-26:
### Describe the bug

Jackd2 doesn't work with vc4hdmi audio device.
Using default bullseye **vc4hdmi.conf**
I tried building latest jack2 and it fails too.

This is what happens:
```

~/src# jackd -d alsa --device default:vc4hdmi0
jackdmp 1.9.20
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2016 Grame.
Copyright 2016-2021 Filipe Coelho.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
creating alsa driver ... default:vc4hdmi0|default:vc4hdmi0|1024|2|48000|0|0|nomon|swmeter|-|32bit
ALSA lib conf.c:5200:(snd_config_expand) Unknown parameters vc4hdmi0
ALSA lib control.c:1471:(snd_ctl_open_noupdate) Invalid CTL default:vc4hdmi0
control open "default:vc4hdmi0" (Invalid argument)
ALSA: Cannot open PCM device alsa_pcm for capture. Falling back to playback-only mode
configuring for 48000Hz, period = 1024 frames (21.3 ms), buffer = 2 periods
ALSA: final selected sample format for playback: 32bit float little-endian
You appear to be using the ALSA software "plug" layer, probably
a result of using the "default" ALSA device. This is less
efficient than it could be. Consider using a hardware device
instead rather than using the plug layer. Usually the name of the
hardwa
ALSA: use 2 periods for playback
Cannot bind server to socket err = No such file or directory
Server channel open error
JackServer::Open failed with -1
Failed to open server
```

Any idea on what's wrong, please?

### Steps to reproduce the behaviour

Try to launch jackd with the **vc4hdmi** audio device.

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400

### System

Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, c12b1df4ed6416fb0df33ba1731c5b13c1bdbdf8, stage2

Linux 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux

HiassofT, 2022-03-27:
ALSA side is fine (except you should better use hdmi:vc4hdmi0 instead of the default PCM), jack fails because it can't bind to it's socket.

A quick test with jackd2 here looks fine
```
root@raspberrypi:~# jackd -d alsa -d hdmi:vc4hdmi0 -P
jackdmp 1.9.17
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2016 Grame.
Copyright 2016-2021 Filipe Coelho.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
ALSA lib control.c:1471:(snd_ctl_open_noupdate) Invalid CTL hdmi:vc4hdmi0
control open "hdmi:vc4hdmi0" (No such file or directory)
ALSA lib control.c:1471:(snd_ctl_open_noupdate) Invalid CTL hdmi:vc4hdmi0
control open "hdmi:vc4hdmi0" (No such file or directory)
creating alsa driver ... hdmi:vc4hdmi0|-|1024|2|48000|0|0|nomon|swmeter|-|32bit
ALSA lib control.c:1471:(snd_ctl_open_noupdate) Invalid CTL hdmi:vc4hdmi0
control open "hdmi:vc4hdmi0" (No such file or directory)
configuring for 48000Hz, period = 1024 frames (21.3 ms), buffer = 2 periods
ALSA: final selected sample format for playback: 32bit integer little-endian
ALSA: use 2 periods for playback
```

I'd suggest you check into your jackd setup and/or ask on the forum, the issue you have is not related to the linux kernel.

vanfanel, 2022-03-27:
@HiassofT Sorry if this is not the place to ask then. And thanks for taking the time to answer even if it's not the right place! Will ask this on the Raspberry Pi forum.

vanfanel, 2022-03-27:
@HiassofT Maybe you can help me after all.
Running as the normal pi user, just like you did, I get this:
```

pi@minimal:/root $ jackd -d alsa -d hdmi:vc4hdmi0 -P
jackdmp 1.9.20
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2016 Grame.
Copyright 2016-2021 Filipe Coelho.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
JACK server starting in realtime mode with priority 10
self-connect-mode is "Don't restrict self connect requests"
Cannot lock down 107350048 byte memory area (Cannot allocate memory)
creating alsa driver ... hdmi:vc4hdmi0|-|1024|2|48000|0|0|nomon|swmeter|-|32bit
ALSA lib control.c:1471:(snd_ctl_open_noupdate) Invalid CTL hdmi:vc4hdmi0
control open "hdmi:vc4hdmi0" (No such file or directory)
ALSA lib conf.c:5200:(snd_config_expand) Unknown parameters vc4hdmi0
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi:vc4hdmi0
ALSA: Cannot open PCM device alsa_pcm for playback. Falling back to capture-only mode
Cannot initialize driver
JackServer::Open failed with -1
Failed to open server
```

Can you please tell me why is it saying "ALSA: Cannot open PCM device alsa_pcm for playback"? 
I mean, what should be creating the "alsa_pcm" device?

vanfanel, 2022-03-27:
@HiassofT I am reopening this because it seems I don't have ah hdmi:vc4hdmi0 device/card combination on my system.

Hence, a couple of questions arise:

Are you using ALSA emulation on top of PulseAudio, or direct ALSA?
Are you using the default vc4hdmi.conf in Bullseye?

HiassofT, 2022-03-27:
I'm running plain RPiOS Bullseye lite on RPi4 here, no pulseaudio installed and jackd from jackd2 distro package runs fine both as root and pi user.

Have you checked if you can use the alsa device at all, eg with "speaker-test -c 2 -t sine -D hdmi:vc4hdmi0"? Do you see the hdmi pcm in "aplay -L" output?

HiassofT, 2022-03-28:
Everything's still fine on the ALSA side, the error is still the same as in your first post, your self-compiled jackd can't bind to it's socket - you have to look into that.

I recommend you start with a fresh RPiOS lite installation, install jackd2 package from repository and then work down from there.

vanfanel, 2022-03-28:
@HiassofT Thanks.
Looking at the RPiOS jackd2 package, it seems to depend on DBUS.
Do you know if jackd2 uses DBUS for the socket stuff?
In other words, do you know if there's some external requeriment for the jackd sockets to work?

HiassofT, 2022-03-28:
Sorry, I'm not familiar with jack internals. I had a look at it once some 15 years ago and now only installed the jackd2 apt package but that's it.

As this is getting highly off topic and is no linux kernel issue I suggest again you better close this issue and seek for support on the RPi forums

vanfanel, 2022-03-28:
@HiassofT Understood, sir.
Again, thanks a lot for your time on this, really.

closed 2022-03-28 by vanfanel

vanfanel, 2022-03-28:
@HiassofT and future me in case of need:

Git this to work. I was missing the /dev/shm directory mounted as tmps.
So I added this to my custom init script:
```

mkdir /dev/shm
mount -t tmpfs tmpfs /dev/shm
```

..and jackd2 started working.
Found this out by debugging the `JackServerSocket::Bind` method. Should have started there, instead of annoying the kernel people with this. Sorry and thanks once more.