raspberrypi/linux #4963: Unexpected XHCI event TRB, skipping...

pcorbes, 2022-03-29:
If I connect at startup, one of these usb peripherals:
```
Bus 001 Device 026: ID 0bda:0151  Realtek Semiconductor Corp. Mass Storage Device (Multicard Reader)
Bus 001 Device 032: ID 4296:7570  VCOM
```
I get this message error:
```
USB XHCI 1.00
scanning bus xhci_pci for devices... WARM halted endpoint, queueing URB anyway.
Unexpected XHCI event TRB, skipping... (3db55dc0 00000004 13000000 06000401)
BUG at drivers/usb/host/xhci-ring.c:503/abort_td()!
BUG!
```
And it restart.
I use a Raspberry Pi 4 with 4GB.

pcorbes, 2022-03-29:
The Grub is not yest started.

closed 2022-03-29 by pelwell

pcorbes, 2022-03-29:
I don't use u-boot

pelwell, 2022-03-29:
Are you sure? Those messages aren't from the kernel (although some of them are similar) , and they aren't from our firmware, and they look a lot like these messages from U-boot (https://www.mail-archive.com/u-boot@lists.denx.de/msg391217.html):
```
U-Boot 2020.10 (Nov 28 2020 - 18:14:31 +0000)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03111)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus usb@7e980000: USB DWC2
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus usb@7e980000 for devices... 1 USB Device(s) found
scanning bus xhci_pci for devices... WARN halted endpoint, queueing URB anyway.
Unexpected XHCI event TRB, skipping... (3db6b570 00000000 13000000 02008401)
BUG at drivers/usb/host/xhci-ring.c:498/abort_td()!
BUG!
resetting ...
=====```

juergh, 2022-03-30:
This is u-boot. Are you running Ubuntu? If so, please report the issue in the Ubuntu bug tracker by running the following on the affected system:
```
$ ubuntu-bug linux-raspi
```