raspberrypi/linux #4967: Missing tag 1.20220328

oliverpool, 2022-03-30:
### Describe the bug

According to https://archive.raspberrypi.org/debian/dists/bullseye/main/binary-armhf/Packages, the kernel is downloadable in version `1.20220328`:
```
Package: raspberrypi-kernel
Source: raspberrypi-firmware
Version: 1:1.20220328-1
Architecture: armhf
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 283578
Breaks: raspberrypi-bootloader (<< 1.20160324-1)
Replaces: raspberrypi-bootloader (<< 1.20160324-1)
Provides: linux-image, wireguard-modules (= 1.0.0)
Multi-Arch: foreign
Homepage: https://github.com/raspberrypi/firmware
Priority: optional
Section: kernel
Filename: pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20220328-1_armhf.deb
Size: 81475392
SHA256: c19375cc42b6b76cdb16587c2ce4033e7f5bdab9f4409f5f57c4bdd97511fd89
SHA1: 30225fbc61919d8845a8d09bf9a309570ba906ea
MD5sum: c17c08bdba4d4ee552ebbc1a351de6f2
Description: Raspberry Pi bootloader
 This package contains the Raspberry Pi Linux kernel.
```

Usually there is a tag with the same version number, indicating the sources which were used to compile this kernel (for instance 1.20220308). This isn't the case here.

### Steps to reproduce the behaviour

Go to https://github.com/raspberrypi/linux/tags and search for the tag `1.20220328`

### Device (s)

Raspberry Pi 1 Mod. B

### System

(not relevant)

### Additional context

I use this tag to build a custom version of the kernel https://github.com/gokrazy-community/kernel-rpi-os-32/runs/5759805027

This tag is likely needed to comply with the GPL of the linux kernel (to know from which sources the compiled kernel was build).

oliverpool, 2022-03-30:
(the tag is also missing on https://github.com/raspberrypi/firmware/tags)

oliverpool, 2022-03-31:
Great, thank you! The build went fine: https://github.com/gokrazy-community/kernel-rpi-os-32/runs/5772056443

> the kernel source is always included in the source

I don't understand. In which source is it included? Is there a .deb file which provides it? (I could only find `raspberrypi-kernel-headers`, but it only has the headers I guess)

I would love to be able to make my build process more robust, and I currently rely on the presence of the tag. I would be eager to learn any better way!

(I'm closing the issue, since my original problem is solved, but I would appreciate more details :)

popcornmix, 2022-03-31:
https://www.cyberciti.biz/faq/how-to-get-source-code-of-package-using-the-apt-command-on-debian-or-ubuntu/

oliverpool, 2022-03-31:
Thank you both, this is very helpful!

Stupid question: what is the difference between the `orig-linux.tar.xz` and the `orig.tar.xz` [files](http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/)?
```
[ ]	raspberrypi-firmware_1.20220328-1.debian.tar.xz	2022-03-30 13:50 	18K	 
[ ]	raspberrypi-firmware_1.20220328-1.dsc	2022-03-30 13:51 	1.8K	 
[ ]	raspberrypi-firmware_1.20220328.orig-linux.tar.xz	2022-03-30 13:51 	123M	 
[ ]	raspberrypi-firmware_1.20220328.orig.tar.xz	2022-03-30 13:50 	130M	 
```
From the link of @popcornmix and some [other online sources](https://debian-handbook.info/browse/ro-RO/stable/sect.source-package-structure.html) I understand that the original source is in `raspberrypi-firmware_1.20220328.orig.tar.xz` and the patches in `raspberrypi-firmware_1.20220328-1.debian.tar.xz`. What then differs in  the `.orig-linux.tar.xz` file? (I couldn't find anything using search engines)

Actually, since the CI to build the kernel is running on debian, I should update my script to follow the detailed instructions of https://wiki.debian.org/BuildingTutorial#Edit_the_source_code.

oliverpool, 2022-03-31:
All right, so I should look at `orig.tar.xz` to have the exact same sources as the official Raspberry OS kernel .

I think I now have all the information I need, thank you very much for all your replies! :pray:

oliverpool, 2022-03-31:
> `orig-linux.tar.xz` will contain the kernel source files.

The same as the one tagged here, right? (the patched kernel, then)

oliverpool, 2023-03-19:
Gentle ping: the tag `1.20230317` is also missing.

Looking around, I haven't found a way to know which commit of this repo is used on a given `raspberrypi-firmware_1.***.orig-linux.tar.xz` file of http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/ without the tag (the commit is not recorded inside the archive, as far as I can tell).

- Is there a way to know which commit of this repo has been used, even if the tag is missing?
- Is there anything I help, so that the tag is automatically added to this repo?
- Or should I stop bothering you and just download and build my kernel from the `raspberrypi-firmware_1.***.orig-linux.tar.xz` file? (instead of a nice "submodule link", I will add to the readme a link to the `tar.xz`)

XECDesign, 2023-03-20:
Thanks, should be fixed now.

> Is there a way to know which commit of this repo has been used, even if the tag is missing?

99.9% of the time it will be [this](https://github.com/raspberrypi/firmware/blob/stable/extra/git_hash).

> Is there anything I help, so that the tag is automatically added to this repo?

Probably not

> Or should I stop bothering you and just download and build my kernel from the `raspberrypi-firmware_1.***.orig-linux.tar.xz` file? (instead of a nice "submodule link", I will add to the readme a link to the `tar.xz`)

The ping is not a bother, I appreciate it. If you rely on the assumption that tags go up at the same time as the apt update, then that's unlikely to happen and the tarballs will be a much better bet. The tags are a manual step in the process which is easy to overlook. Maybe there's something I can do on my end to automatically add tags whenever an apt update goes to the public repo, but on a long list of things to do, that's not near the top.

closed 2023-03-20 by oliverpool