raspberrypi/linux #4974: Rpi4: DRM render nodes for vc4 device?

dhobsong, 2022-04-01:
### Describe the bug

Hello, 

I am trying to run my application that uses DRM render nodes for buffer import/export, etc. On raspberry pi3 is seems like the modeset device (vc4) provides a render node device(/dev/dri/renderD12[8,9]), but that no longer seems to be the case on rpi4 (only one render node, corresponding to the v3d device is present).  

Looking through past issues, I found #3363, where I can see that the render node is only  enabled when v3d is present and there is a list of [device tree ids](https://github.com/raspberrypi/linux/blob/aeaa2460db088fb2c97ae56dec6d7d0058c68294/drivers/gpu/drm/vc4/vc4_v3d.c#L511) 
used to check for that the v3d device is enabled.   

On rpi4 the display controller device (bcm7211-vc5) is not in this list.  Should it be?  Is adding this to the list enough to enable the render node support? Or are render nodes just not supported on the "vc4" device? 

### Steps to reproduce the behaviour

reproducible  on boot

```
$ ls /dev/dri/renderD*
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS: Yocto (meta-raspberrypi)
Kernel version: Linux 5.4.72

6by9, 2022-04-01:
Pi 3 and Pi4 have different V3D units, hence one using vc4 driver and the other using the v3d driver.
You can not use the vc4_v3d driver on Pi4/BCM2711.

Also note that the rpi-5.4.y branch hasn't been significantly updated since Dec 2020. Most of the support requires at least the rpi-5.10.y branch.

dhobsong, 2022-04-01:
@6by9 
> Pi 3 and Pi4 have different V3D units, hence one using vc4 driver and the other using the v3d driver.
Pi4 seems to be using both, right?

> You can not use the vc4_v3d driver on Pi4/BCM2711.

Thank you, that's good to know. 

I also did a quick check on the 5.10.y branch and still no render node for the "vc4" device.  Looking at the latest source code on the 5.15.y branch it seems like the conditions for disabling the render node haven't changed.

I guess my main question is whether the render node is supposed to be enabled on the vc4 device or is it just not supported at all on Pi4?

6by9, 2022-04-01:
> I guess my main question is whether the render node is supposed to be enabled on the vc4 device or is it just not supported at all on Pi4?

No it isn't. The v3d node supports rendering, and the vc4 node controls the display pipeline.

dhobsong, 2022-04-04:
Thank you for the clarification.
Seems like it's all working as intended. 
I'll close this issue.

closed 2022-04-04 by dhobsong