raspberrypi/linux #4975: Last 20220328 has a problem with AR9271

koroki, 2022-04-01:
### Describe the bug

The Atheros driver is not working properly with the last kernel. Specifically with AR9271 chipset of the TL-WN722N.
The wlan0 is visible but cannot be active. 
Problem is solved when I downgraded to 1.20220308.

I'm using raspberry 4 with 64bit kernel. 

### Steps to reproduce the behaviour

Connect the usb wifi and simply don't work. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

firmware e5a963efa66a1974127860b42e913d2374139ff5
Kernel  aarch64

koroki, 2022-04-05:
I can confirm that the problem is NOT solved with the last version 20220331.

I can put more information.
journal shows that information in the last kernel
```
dhcpcd[992]: wlan0: failed to start wpa_supplicant
dhcpcd-run-hooks[1001]: wlan0: failed to start wpa_supplicant
dhcpcd[992]: wlan0: Successfully initialized wpa_supplicant
dhcpcd[992]: nl80211: kernel reports: Match already configured
dhcpcd[992]: nl80211: Could not configure driver mode
dhcpcd[992]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
dhcpcd[992]: wlan0: Failed to initialize driver interface
dhcpcd-run-hooks[1002]: wlan0: Successfully initialized wpa_supplicant
                                                        nl80211: kernel reports: Match already configured
                                                        nl80211: Could not configure driver mode
                                                        nl80211: deinit ifname=wlan0 disabled_11b_rates=0
                                                        wlan0: Failed to initialize driver interface
```
I cannot see any fail on dmesg, but please ask if you need. 

I also try the fix solution of this issue:
https://github.com/raspberrypi/linux/issues/4976#issue-1190832376
But it is not working.

bjajoh, 2022-09-18:
I can confirm this is still a problem!