raspberrypi/linux #4976: brcmfmac fails to connect to WPA2/WPA3 mixed-mode AP

dword1511, 2022-04-03:
### Describe the bug

# Model
Pi 3 A
# Kernel
```
$ dpkg --no-pager --list raspberrypi-kernel                                                                                                                                                                                                                                                                    
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version        Architecture Description
+++-==================-==============-============-=================================
ii  raspberrypi-kernel 1:1.20220328-1 armhf        Raspberry Pi bootloader
$ uname -a
Linux catcam 5.15.30-v7+ #1536 SMP Mon Mar 28 13:43:34 BST 2022 armv7l GNU/Linux
```

# Symptom
brcmfmac will fail to connect to WPA2/WPA3 mixed-mode AP (I am using an Archer A7 V5 running OpenWRT, where all other devices work fine). wpa_supplicant complains about auth rejected and AP sees invalid keys.

# How to fix
Followed instructions at https://iwd.wiki.kernel.org/offloading to disable SAE offloading by appending brcmfmac.feature_disable=0x82000 to the kernel command line (/boot/cmdline.txt) and it starts to connect and work normally.

### Steps to reproduce the behaviour

1. Have an AP in WPA2/WPA3 mixed-mode (pure WPA3 might also work)
2. Install `raspberrypi-kernel` `1:1.20220328-1`
3. Configure `wpa_supplicant` by adding a valid entry generated by `wpa_passphrase` to `/etc/wpa_supplicant/wpa_supplicant-wlan0.conf`
4. `sudo systemctl enable wpa_supplicant@wlan0 && sudo systemctl start wpa_supplicant@wlan0` (assuming wpa_supplicant.service was disabled)
5. `sudo journalctl -u wpa_supplicant@wlan0 -f` and watch it fail

### Device (s)

Raspberry Pi 3 Mod. A+

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2015-01-31 (armhf)
Generated using spindle, http://asbradbury.org/projects/spindle/, 56db930, stage4-lxde-edu.qed
$ vcgencmd version
Mar 24 2022 13:21:51 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start_x)
$ uname -a
Linux catcam 5.15.30-v7+ #1536 SMP Mon Mar 28 13:43:34 BST 2022 armv7l GNU/Linux
```

### Logs

I am using volatile storage for logs. Let me know if exact kernel logs or wpa_supplicant logs are needed.

taylorkline, 2022-06-04:
No issue on my end:

`/etc/wpa_supplicant/wpa_supplicant.conf`:

```
update_config=1

network={
 ssid="ssid"
 key_mgmt=WPA-PSK-SHA256
 psk=psk
 ieee80211w=2
}
```

dword1511, 2022-07-28:
I think `key_mgmt=WPA-PSK-SHA256` and `ieee80211w=2` fixed it

closed 2022-07-28 by dword1511

codeaholics, 2023-03-18:
@dword1511  Thank you for this. I recently upgraded a RPi 3 from Raspbian to Ubuntu 22.04 LTS and the wifi was terrible - taking up to half an hour to connect, dropping connections frequently, etc. I've been fighting it for 3 days. The `feature_disable` kernel command line flag has solved this for me. Thank you!

Kobunteki, 2024-05-08:
> I think `key_mgmt=WPA-PSK-SHA256` and `ieee80211w=2` fixed it

Do you know why "key_mgmt=WPA-PSK-SHA256" works? I just cannot get sussed out the link between it and WPA3/WPA2 mixed mode.

yozik04, 2024-12-15:
I can confirm that my Raspberry Pi Zero 2 W wifi started to work only after appending `brcmfmac.feature_disable=0x82000` to the end of `cmdline.txt`. I use Bookwork based Raspbian.

molecularentropy, 2025-02-05:
`brcmfmac.feature_disable=0x82000` fixed this issue for me as well.

steveja, 2026-10-11:
rpi zero-2w, using Bookworm, kernel 6.12.109+rpt-rpi-v8, wpasupplicant  2:2.10-12+deb12u3.  Eero pro7 mesh.  Connection fails randomly in 10 minutes to ~14 days with a lot of re-connection attempts in the logs before network silence.
Package firmware-brcm80211 1:20240709-2~bpo12+1+rpt4  includes /lib/modprobe.d/rpi-brcmfmac.conf:"options brcmfmac roamoff=1 feature_disable=0x282000".  So the cmdline options have no impact (note that bit 0x2000000, bit 25,  is not a valid disable in the 6.12 kernel brcmfmac driver).  No joy.