raspberrypi/linux #4977: 5.15 has dropped all USB_GSPCA modules

6by9, 2022-04-03:
### Describe the bug

Mainline patch https://github.com/torvalds/linux/commit/2a621b0859d1caf4e03f47066dad78011d274617 changed the default for USB_GSPCA from m to disabled, so all the dependent modules are no longer built.

https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/configs/bcm2711_defconfig#L803-L850 are all gone.
Whilst I could have created the PR myself, I know @pelwell has magic scripts to deal with configs :-)

### Steps to reproduce the behaviour

Reported in https://forums.raspberrypi.com/viewtopic.php?p=1990000#p1990000
gspca/ov534 camera is no longer detected.

### Device (s)

Raspberry Pi 4 Mod. B

### System

rpi-5.15.y kernel

cvicari, 2022-04-03:
I have the same problem as well. I was using an old camera that made use of the module gspca_stv06xx.ko, that disappeared with kernel 5.15

pelwell, 2022-04-03:
See 3dd028b.

6by9, 2022-04-03:
Thanks @pelwell. Not tested, but looks like it should do the intended thing, so I'll close.

closed 2022-04-03 by 6by9