raspberrypi/linux #4982: Kernel 5.18-rc1 V3D issue

Dark-Sky, 2022-04-07:
### Describe the bug

Unable to load V3D using KMS or FKMS. The kernel shows a Stack Trace and glxgears shows using llvmpipe.

### Steps to reproduce the behaviour

Enable KMS or FKMS in config.txt and boot up.

### Device (s)

Raspberry Pi 4 Mod. B

### System

[ray@pi4 Desktop]$ vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

[ray@pi4 Desktop]$ uname -a
Linux pi4 5.18.0-rc1-1-MANJARO-ARM-RPI #1 SMP PREEMPT Thu Apr 7 15:58:24 UTC 2022 aarch64 GNU/Linux

### Logs

[    5.217371] ------------[ cut here ]------------
[    5.217399] WARNING: CPU: 1 PID: 201 at drivers/gpu/drm/v3d/v3d_drv.c:283 v3d_platform_drm_probe+0x32c/0x37c [v3d]
[    5.217443] Modules linked in: snd_soc_hdmi_codec(+) v3d(+) gpu_sched crct10dif_ce joydev vc4 hci_uart drm_shmem_helper btbcm brcmfmac cec bluetooth bcm2835_v4l2(C) bcm2835_codec(C) brcmutil v4l2_mem2mem bcm2835_isp(C) raspberrypi_hwmon bcm2835_mmal_vchiq(C) cfg80211 videobuf2_vmalloc videobuf2_dma_contig ecdh_generic videobuf2_memops drm_cma_helper videobuf2_v4l2 ecc drm_kms_helper i2c_brcmstb rfkill videobuf2_common videodev snd_soc_core vc_sm_cma(C) mc snd_bcm2835(C) snd_compress snd_pcm_dmaengine snd_pcm snd_timer rpivid_mem snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem sch_fq_codel dm_multipath drm crypto_user fuse backlight ip_tables x_tables ipv6
[    5.217562] CPU: 1 PID: 201 Comm: systemd-udevd Tainted: G         C        5.18.0-rc1-1-MANJARO-ARM-RPI #1
[    5.217574] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[    5.217580] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    5.217588] pc : v3d_platform_drm_probe+0x32c/0x37c [v3d]
[    5.217608] lr : v3d_platform_drm_probe+0x108/0x37c [v3d]
[    5.217625] sp : ffff80000a88b8b0
[    5.217629] x29: ffff80000a88b8b0 x28: 0000000000000013 x27: 0000000000000100
[    5.217639] x26: ffff80000814bf90 x25: ffff80000121e0c0 x24: 0000000000000030
[    5.217648] x23: 00000fffffffffff x22: 000000000000002c x21: ffff0000401f2010
[    5.217657] x20: 0000000000000000 x19: ffff00004d4e8000 x18: ffffffffffffffff
[    5.217666] x17: 0000000000000001 x16: 00004a72ab62b108 x15: ffff00004ef8cb06
[    5.217674] x14: 0000000000000000 x13: 00657361656c6572 x12: 5f74696e695f6d65
[    5.217683] x11: ffff800008000000 x10: 00000000fec08000 x9 : ffff800008121a38
[    5.217692] x8 : 0040000000000001 x7 : 00008000f4bf4000 x6 : ffff0000fbffe280
[    5.217701] x5 : ffff8000094b9000 x4 : ffff8000094b9268 x3 : 000000000000000e
[    5.217709] x2 : 000000000000000e x1 : 00000000000000a4 x0 : 0000000000deadbe
[    5.217719] Call trace:
[    5.217722]  v3d_platform_drm_probe+0x32c/0x37c [v3d]
[    5.217739]  platform_probe+0x60/0xc0
[    5.217748]  really_probe.part.0+0xa4/0x2cc
[    5.217756]  __driver_probe_device+0xcc/0x134
[    5.217763]  driver_probe_device+0xb4/0x14c
[    5.217769]  __driver_attach+0x104/0x1a0
[    5.217775]  bus_for_each_dev+0x7c/0xe0
[    5.217781]  driver_attach+0x30/0x40
[    5.217787]  bus_add_driver+0x15c/0x200
[    5.217793]  driver_register+0x84/0x140
[    5.217799]  __platform_driver_register+0x34/0x40
[    5.217804]  v3d_platform_driver_init+0x2c/0x1000 [v3d]
[    5.217820]  do_one_initcall+0x50/0x29c
[    5.217827]  do_init_module+0x50/0x26c
[    5.217835]  load_module+0x2268/0x29d0
[    5.217841]  __do_sys_init_module+0x190/0x200
[    5.217847]  __arm64_sys_init_module+0x28/0x34
[    5.217853]  invoke_syscall+0x50/0x120
[    5.217860]  el0_svc_common.constprop.0+0x104/0x124
[    5.217866]  do_el0_svc+0x34/0x9c
[    5.217872]  el0_svc+0x38/0xc0
[    5.217880]  el0t_64_sync_handler+0xa4/0x130
[    5.217885]  el0t_64_sync+0x18c/0x190
[    5.217891] ---[ end trace 0000000000000000 ]---
[    5.242790] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[    5.243007] [drm] forcing HDMI-A-1 connector on
[... 52 lines cut ...]
[    5.436800]  v3d_platform_drm_probe+0x27c/0x37c [v3d]
[    5.436822]  platform_probe+0x60/0xc0
[    5.436836]  really_probe.part.0+0xa4/0x2cc
[    5.436854]  __driver_probe_device+0xcc/0x134
[    5.436866]  driver_probe_device+0xb4/0x14c
[    5.436878]  __driver_attach+0x104/0x1a0
[    5.437808]  bus_for_each_dev+0x7c/0xe0
[    5.438713]  driver_attach+0x30/0x40
[    5.439614]  bus_add_driver+0x15c/0x200
[    5.440517]  driver_register+0x84/0x140
[    5.441404]  __platform_driver_register+0x34/0x40
[    5.442277]  v3d_platform_driver_init+0x2c/0x1000 [v3d]
[    5.443176]  do_one_initcall+0x50/0x29c
[    5.444075]  do_init_module+0x50/0x26c
[    5.444933]  load_module+0x2268/0x29d0
[    5.445774]  __do_sys_init_module+0x190/0x200
[    5.446687]  __arm64_sys_init_module+0x28/0x34
[    5.447606]  invoke_syscall+0x50/0x120
[    5.448458]  el0_svc_common.constprop.0+0x104/0x124
[    5.449275]  do_el0_svc+0x34/0x9c
[    5.450086]  el0_svc+0x38/0xc0
[    5.450901]  el0t_64_sync_handler+0xa4/0x130
[    5.451730]  el0t_64_sync+0x18c/0x190
[    5.452552] Code: d2800000 d50332bf f8607881 91016021 (b9000023) 
[    5.453392] ---[ end trace 0000000000000000 ]---

popcornmix, 2022-04-07:
Yes - I see this too. I have a earlier branch which has working v3d and isn't too different (mostly the difference in in upstream clock driver). I'll see if I can narrow down what the difference is.

popcornmix, 2022-04-07:
I've added the commit from the working branch. Working for me now.

Dark-Sky, 2022-04-07:
I do not have a kernel stack trace now but I still do not have V3D. I still get llvmpipe. VC4 mods seem to load up but no v3d.

dmesg:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.18.0-rc1-1-MANJARO-ARM-RPI (builduser@fh-mjr-build-lon-xlarge) (gcc (GCC) 11.2.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Thu Apr 7 19:41:44 UTC 2022
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19712 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s77864 r8192 d32728 u118784
[    0.000000] pcpu-alloc: s77864 r8192 d32728 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028596
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1360x768M@60 smsc95xx.macaddr=DC:A6:32:AF:B8:7B vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  root=PARTUUID=b75fdbb1-02 rw rootwait console=ttyS0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyS0,115200 usbhid.mousepoll=8 audit=0 drm.edid_firmware=edid.dat video=HDMI-A-1:D
[    0.000000] audit: disabled (until reboot)
[    0.000000] Unknown kernel command line parameters "selinux=0", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037300000-0x000000003b300000] (64MB)
[    0.000000] Memory: 7468360K/8244224K available (12544K kernel code, 2536K rwdata, 4332K rodata, 4160K init, 1170K bss, 251576K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40235 entries in 158 pages
[    0.000000] ftrace: allocated 158 pages with 5 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000280] Console: colour dummy device 80x25
[    0.000834] printk: console [tty1] enabled
[    0.000916] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000957] pid_max: default: 32768 minimum: 301
[    0.001160] LSM: Security Framework initializing
[    0.001469] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001607] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.002902] cgroup: Disabling memory control group subsystem
[    0.004887] cblist_init_generic: Setting adjustable number of callback queues.
[    0.004920] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.005109] cblist_init_generic: Setting shift to 2 and lim to 1.
[... 993 lines cut ...]
[    23.755] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[    23.755] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 664 704 832  480 489 492 520 -hsync -vsync (37.9 kHz e)
[... cut ...]
[    23.755] (II) modeset(0): Modeline "1440x900"x0.0  136.75  1440 1536 1688 1936  900 903 909 942 -hsync +vsync (70.6 kHz e)
[    23.755] (II) modeset(0): Modeline "1600x1200"x0.0  162.00  1600 1664 1856 2160  1200 1201 1204 1250 +hsync +vsync (75.0 kHz e)
[    23.755] (II) modeset(0): Modeline "1680x1050"x0.0  119.00  1680 1728 1760 1840  1050 1053 1059 1080 +hsync -vsync (64.7 kHz e)
[    23.755] (--) modeset(0): HDMI max TMDS frequency 225000KHz
[    24.000] (II) modeset(0): Allocate new frame buffer 1440x900 stride
```

[image]

Dark-Sky, 2022-04-11:
I tested @anholt commit 70388f2326a06d9eef3fe525561d8a03f4206f66 and V3D loaded up. I did see the MMU error  that was mentioned in the commit with one instance after running glxgears and the webgl fish tank in the browser for 3 hours on my pi4 8G. Other than that there were no Desktop issues I could see.

[image]

popcornmix, 2022-04-11:
> I tested @anholt commit https://github.com/raspberrypi/linux/commit/70388f2326a06d9eef3fe525561d8a03f4206f66 and V3D loaded up.

I'm not quite understanding. Wasn't that the commit at top of rpi-5.18.y on Friday (we have rebased to v5.18-rc2 now). Are you saying v3d works or doesn't work?

Dark-Sky, 2022-04-11:
After testing today 5.18-rc2 it also appears to be ok with V3D although I am getting @10 fps reduction with the webgl fish tank in chromium than I was getting yesterday. I do not know if it was a coincidence or not but but the instant I switched from firefox to chromium the MMU error popped up with dmesg -w as the program opened.

```
[ray@pi4 ~]$ uname -a
Linux pi4 5.18.0-rc2-1-MANJARO-ARM-RPI #1 SMP PREEMPT Mon Apr 11 14:58:39 UTC 2022 aarch64 GNU/Linux
 
[ray@pi4 ~]$ lsmod | grep v3d
v3d                    86016  3
gpu_sched              53248  1 v3d
drm_shmem_helper       28672  1 v3d
drm                   544768  12 drm_cma_helper,gpu_sched,drm_kms_helper,v3d,vc4,drm_shmem_helper
 
[ray@pi4 ~]$ journalctl | grep "MMU error"
Apr 11 10:42:21 pi4 kernel: v3d fec00000.v3d: MMU error from client CLE (4) at 0xb4c1000, pte invalid
 
[ray@pi4 ~]$ glxgears -info
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
GL_RENDERER   = V3D 4.2
GL_VERSION    = 2.1 Mesa 22.0.1
GL_VENDOR     = Broadcom
...
...
376 frames in 5.0 seconds = 74.990 FPS
...
...
```

popcornmix, 2022-04-11:
>  I am getting @10 fps reduction with the webgl fish tank in chromium than I was getting yesterday

What was the difference in configuration yesterday?
Was yesterday using `-rc1` kernel and today `-rc2`?
Is this reproducible? (i.e. can you get the higher and lower fps by changing something in build?)

Dark-Sky, 2022-04-11:
I had done a make bcm2711_defconfig for a base then added the stuff people always want in the forums. Today since you did a rebase I started over again with make bcm2711_defconfig and added the extra modules. I did notice that everytime you do a rebase it starts with config: Add default configs then several commits are added on top of it. This was not the case this time. There seems to be quite a few changes with the config today.

[](https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-rpi4-rc/-/commit/b9512b7eefc7e6b9e76374193b2e3f751ac3007d#dfba7aade0868074c2861c98e2a9a92f3178a51b_1_1)

I do want to go back and check if there is a difference between KMS and FKMS and see if the fps speed changes. I did not check both either day.

Dark-Sky, 2022-04-11:
I went back and checked the current -rc kernel with both KMS/FKMS V3D with the webgl fish tank and they are both averaging 14fps and peaking at 16fps at times with chromium.

To tell you the truth this is what it has always been here with chromium and firefox averaging 10fps. When I saw it averaging 20fps yesterday and peaking to 24fps in chromium. I did know if it was improvements in chromium or RPi code but I was thrilled. I released the kernel in the repo and it was also verified by another member.

https://forum.manjaro.org/t/raspberry-pi-kernels-2-0/84885/423

Dark-Sky, 2022-04-12:
@popcornmix 

I am happy with V3D now working. After thinking about the fps in the webgl fish tank jumping up to 20-24 I kinda think that it may be a fluke mis-reporting in chromium since it has always been 14-16 fps there.

I noticed yesterday testing 5.15.33 I was also getting the _pi4 kernel: v3d fec00000.v3d: MMU error from client CLE_ there. I never really looked at dmesg before after any extended time booted up so guessing it may have been showing up along there also.

So closing this issue if you have no objections. Thanks for your help.

closed 2022-04-12 by Dark-Sky