raspberrypi/linux #4985: NFS 4.2 mount not working properly

UweDoe, 2022-04-11:
### Describe the bug

Mounting an external NFS server which is running on version 4.2 results in an hanging process if one tries to access file from the NFS server.
This problem occurs for kernel  5.15.32-v7+, whereas using the same mounting commands work properly on kernel 5.10.95-v7+
Specifying the NFS version as an option in the mount command is working, but only for   -o nfsvers=4,minorversion=0  or -o nfsvers=4 ommitting the minor version. Minorversion 1 and 2 result in the hanging process, as soon as one trys to access mounted files.

The problem was discussed in https://dietpi.com/phpbb/viewtopic.php?t=10238 earlier

### Steps to reproduce the behaviour

Working case:

root@pi2udhifi:~# umount /mnt/nfs
root@pi2udhifi:~# mount -t nfs -o "nfsvers=4,minorversion=0" 192.168.16.32:/ /mnt/nfs/
root@pi2udhifi:~# df -h
Dateisystem     Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root         60G    8,1G   49G   15% /
devtmpfs         430M       0  430M    0% /dev
tmpfs            462M       0  462M    0% /dev/shm
tmpfs            185M    7,6M  178M    5% /run
tmpfs            5,0M    8,0K  5,0M    1% /run/lock
tmpfs            1,0G    8,0K  1,0G    1% /tmp
/dev/mmcblk0p1   127M     52M   75M   41% /boot
tmpfs             93M    8,0K   93M    1% /run/user/0
tmpfs             93M     12K   93M    1% /run/user/1000
tmpfs             50M    8,0K   50M    1% /var/log
192.168.16.32:/   16M       0   16M    0% /mnt/nfs
root@pi2udhifi:~# ls -la /mnt/nfs/nfs
insgesamt 256
drwxrwxrwx 10 root   root     4096  2. Mär 12:04 .
drwxrwxrwt  3 root   root       60  7. Apr 10:02 ..
-rw-rw-rw-  1 root   root   227512  2. Mär 11:37 Alarm.mp3
drwxr-xr-x  3 root   root     4096 19. Feb 2021  dietpi-backup
drwxrwxr-x  9 dietpi dietpi   4096 27. Sep 2021  dietpi-sync
drwx------  7 root   root     4096  3. Jan 11:44 ha-repo
drwxrwxrwx  2 root   root     4096 27. Mär 15:10 @Recycle
drwxrwxrwx  2 root   root     4096  2. Mär 12:04 .@__thumb
drwxr-xr-x  3 root   root     4096  2. Feb 10:08 tmp
drwx------  7 root   root     4096  3. Jan 14:11 wp-repo
root@pi2udhifi:~# df -h
Dateisystem        Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root            60G    8,1G   49G   15% /
devtmpfs            430M       0  430M    0% /dev
tmpfs               462M       0  462M    0% /dev/shm
tmpfs               185M    7,6M  178M    5% /run
tmpfs               5,0M    8,0K  5,0M    1% /run/lock
tmpfs               1,0G    8,0K  1,0G    1% /tmp
/dev/mmcblk0p1      127M     52M   75M   41% /boot
tmpfs                93M    8,0K   93M    1% /run/user/0
tmpfs                93M     12K   93M    1% /run/user/1000
tmpfs                50M    8,0K   50M    1% /var/log
192.168.16.32:/      16M       0   16M    0% /mnt/nfs
192.168.16.32:/nfs   64G     16G   48G   25% /mnt/nfs/nfs
root@pi2udhifi:~#

Not working case:
root@pi2udhifi:~# umount /mnt/nfs
root@pi2udhifi:~# df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root        60G    8,1G   49G   15% /
devtmpfs        430M       0  430M    0% /dev
tmpfs           462M       0  462M    0% /dev/shm
tmpfs           185M    7,6M  178M    5% /run
tmpfs           5,0M    8,0K  5,0M    1% /run/lock
tmpfs           1,0G    8,0K  1,0G    1% /tmp
/dev/mmcblk0p1  127M     52M   75M   41% /boot
tmpfs            93M    8,0K   93M    1% /run/user/0
tmpfs            93M     12K   93M    1% /run/user/1000
tmpfs            50M       0   50M    0% /var/log
root@pi2udhifi:~# mount -t nfs -o "nfsvers=4,minorversion=2" 192.168.16.32:/ /mnt/nfs/
root@pi2udhifi:~# df -h
Dateisystem     Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root         60G    8,1G   49G   15% /
devtmpfs         430M       0  430M    0% /dev
tmpfs            462M       0  462M    0% /dev/shm
tmpfs            185M    7,6M  178M    5% /run
tmpfs            5,0M    8,0K  5,0M    1% /run/lock
tmpfs            1,0G    8,0K  1,0G    1% /tmp
/dev/mmcblk0p1   127M     52M   75M   41% /boot
tmpfs             93M    8,0K   93M    1% /run/user/0
tmpfs             93M     12K   93M    1% /run/user/1000
tmpfs             50M       0   50M    0% /var/log
192.168.16.32:/   16M       0   16M    0% /mnt/nfs
root@pi2udhifi:~# ls -la /mnt/nfs/nfs

^C     <-----process hangs forever
root@pi2udhifi:~#

### Device (s)

Raspberry Pi 2 Mod. B v1.2

### System

root@pi2udhifi:/etc# vcgencmd version
Mar 24 2022 13:20:54
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

root@pi2udhifi:/etc# uname -a
Linux pi2udhifi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
root@pi2udhifi:/etc#

root@pi2udhifi:/etc# cat os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@pi2udhifi:/etc#

sachapan, 2022-04-17:
I experienced the same issue on kernel 5.15.32-v8+ 64bit.  The default nfs version for mounting a share from a qnap NAS fails with this kernel.  

As a workaround, I used the suggestion from @UweDoe  above: force version 4.0.
In /etc/fstab:
server:/Share /mnt/Share nfs vers=4.0,rw,async 0 0

McTristan, 2022-04-17:
which does not seem to work in docker swarm mounts ...