raspberrypi/linux #4987: Kernel 5.17.2 AARCH64 compile issue

Dark-Sky, 2022-04-11:
### Describe the bug

Kernel 5.17.2 AARCH64 errors out compiling.

### Steps to reproduce the behaviour

Try to compile Kernel 5.17.2 AARCH64.

### Device (s)

Raspberry Pi 4 Mod. B

### System

[ray@pi4 ~]$ vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

Upstream kernel:
[ray@pi4 ~]$ uname -a
Linux pi4 5.18.0-rc2-2-MANJARO-ARM-RC #1 SMP PREEMPT Mon Apr 11 09:07:03 UTC 2022 aarch64 GNU/Linux

### Logs

  CC [M]  fs/ocfs2/uptodate.o
drivers/gpu/drm/vc4/vc4_hdmi.c: In function ‘vc4_hdmi_sink_supports_format_bpc’:
drivers/gpu/drm/vc4/vc4_hdmi.c:1614:42: error: ‘const struct drm_display_info’ has no member named ‘edid_hdmi_dc_modes’; did you mean ‘edid_hdmi_rgb444_dc_modes’?
 1614 |                 if (bpc == 10 && !(info->edid_hdmi_dc_modes & DRM_EDID_HDMI_DC_30)) {
      |                                          ^~~~~~~~~~~~~~~~~~
      |                                          edid_hdmi_rgb444_dc_modes
drivers/gpu/drm/vc4/vc4_hdmi.c:1619:42: error: ‘const struct drm_display_info’ has no member named ‘edid_hdmi_dc_modes’; did you mean ‘edid_hdmi_rgb444_dc_modes’?
 1619 |                 if (bpc == 12 && !(info->edid_hdmi_dc_modes & DRM_EDID_HDMI_DC_36)) {
      |                                          ^~~~~~~~~~~~~~~~~~
      |                                          edid_hdmi_rgb444_dc_modes
drivers/gpu/drm/vc4/vc4_hdmi.c:1653:42: error: ‘const struct drm_display_info’ has no member named ‘edid_hdmi_dc_modes’; did you mean ‘edid_hdmi_rgb444_dc_modes’?
 1653 |                 if (bpc == 10 && !(info->edid_hdmi_dc_modes & DRM_EDID_HDMI_DC_30)) {
      |                                          ^~~~~~~~~~~~~~~~~~
      |                                          edid_hdmi_rgb444_dc_modes
drivers/gpu/drm/vc4/vc4_hdmi.c:1658:42: error: ‘const struct drm_display_info’ has no member named ‘edid_hdmi_dc_modes’; did you mean ‘edid_hdmi_rgb444_dc_modes’?
 1658 |                 if (bpc == 12 && !(info->edid_hdmi_dc_modes & DRM_EDID_HDMI_DC_36)) {
      |                                          ^~~~~~~~~~~~~~~~~~
      |                                          edid_hdmi_rgb444_dc_modes
make[4]: *** [scripts/Makefile.build:288: drivers/gpu/drm/vc4/vc4_hdmi.o] Error 1
make[4]: *** Waiting for unfinished jobs....

### Additional context

Was compiling using Manjaro's build server.

popcornmix, 2022-04-12:
I believe this was fixed yesterday: https://github.com/raspberrypi/linux/commit/91b74f2134a3bb19e7e140e7d53122107e94ff5c

Dark-Sky, 2022-04-12:
Thanks. I missed that commit some how. I may run out of time before I get to test before I have to go to town for my weekly grocery shopping but will post after I get to test it.