raspberrypi/linux #4988: Kernel 5.15.33 AARCH64 mouse movement with V3D

Dark-Sky, 2022-04-11:
### Describe the bug

While booted into KMS or FKMS V3D the mouse movement is very erratic and when moving the mouse while using glxgears the gears completely stop until you cease moving the mouse. The same thing happens while playing a youtube video with the video stopping while moving the mouse. Everything seems ok if booted using llvmpipe.

### Steps to reproduce the behaviour

Boot using KMS or FKMS V3D and start glxgears -info or a youtube video in the browser and move the mouse. Also notice erratic behavior of the mouse at all times when booted into the desktop using KMS or FKMS V3D.

### Device (s)

Raspberry Pi 4 Mod. B

### System

[ray@pi4 ~]$ vcgencmd version
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
 
[ray@pi4 ~]$ uname -a
Linux pi4 5.15.33-1-MANJARO-ARM-RPI #1 SMP PREEMPT Mon Apr 11 13:35:56 UTC 2022 aarch64 GNU/Linux

popcornmix, 2022-04-11:
Did this start recently? I wonder if it could be https://github.com/raspberrypi/linux/pull/4971

Dark-Sky, 2022-04-11:
> Did this start recently? I wonder if it could be #4971

I wondered the same thing remembering the commit. It was made after I had compiled 5.15.32.

popcornmix, 2022-04-11:
Can you test latest 5.15 kernel with those commits reverted? It would be useful to confirm they are to blame (based on the timing, it seems likely).

Dark-Sky, 2022-04-11:
Sure. Be glad to....

Dark-Sky, 2022-04-11:
@popcornmix 

Sorry it took so long to respond back. My boy and his family showed up and had to wait for a thunderstorm to pass.

I reverted the 2 commits and the mouse issue went away on 5.15.33.

popcornmix, 2022-04-12:
@mripard looks like https://github.com/raspberrypi/linux/pull/4971 has caused a performance regression.

Dark-Sky, 2022-04-12:
I compiled 5.18-rc2 again today to add commits since my last compile and it now has the erratic mouse and V3D issue. Although I still can not get kernel 5.17.2 to compile yet to test it also looks to me it will have the issue also.

popcornmix, 2022-04-12:
I believe 5.15, 5.16, 5.17 and 5.18 all have the `drm/atomic-helpers: remove legacy_cursor_update hacks` commit and so I'd expect all to have this issue.

mripard, 2022-04-13:
I wonder if it's a similar issue to #4571. Does it occur only when playing a video or rendering on the GPU, or does it also stall the rest of the display (so something like displaying a huge chunk of text in a terminal would be stalled as well?)

Either way, it looks like a severe regression that won't get fixed fast, we should probably revert it for now

Dark-Sky, 2022-04-13:
I will test further in a little bit. Right now all I tested was with any graphic with movement ceased when moving the mouse but started back up when you rest the mouse. The mouse was very erratic/laggy trying to move the mouse around the Desktop when no graphic movement was being done with a program.

I have reverted all 3 kernels I maintain for our repo at the moment at released them.

Dark-Sky, 2022-04-13:
Oops. I deleted the old kernel. I can recompile if needed.

popcornmix, 2022-04-13:
Not sure I can reproduce on RPiOS 32bit. Kernel: 5.15.33-v7l+
Booted to desktop and run glxgears. Mouse moves okay and gears do not stall.
Tried 1080p60 and 4kp60. Tried running neverball too.
Tried with mutter disabled and enabled.

Dark-Sky, 2022-04-13:
Mine is 64 bit with XFCE and mesa 22.0.1-3.

Dark-Sky, 2022-04-13:
I have a 32bit arch-arm image I here with MATE DE. I will see what it does. We mirror arch-arm's repo so it should be the same program versions but 64bit.

popcornmix, 2022-04-13:
Still seems fine with RPiOS bullseye 64-bit

Dark-Sky, 2022-04-13:
Seems fine on 32bit arch-arm image with MATE DE I just compiled:
```

[alarm@alarmpi ~]$ uname -a
Linux alarmpi 5.15.33-v7l+ #1 SMP Wed Apr 13 09:16:09 CDT 2022 armv7l GNU/Linux
```

Two 64 bit kernels 5.15.33 & 5.18-rc2 with XFCE had the issue right after the new commits were made. I never got to test 5.17.2 as I had gotten around to making patches that reversed the commits by then since I found out the commits landed there also. One thing for sure is the reversing of the commits makes the issue go away; so a little puzzling to me...

popcornmix, 2022-04-13:
Does 32-bit arch-arm image have same version of mesa?
On rpios we can run fbturbo, xcompmgr, glamor, mutter in different configurations.
Do your 32-bit and 64-bit images run the same window managers/compositors/acceleration libs?

Dark-Sky, 2022-04-13:
My 32 bit image has the same mesa version as my 64bit.

 fbturbo got obsoleted as it is not maintained anymore. It did not have the proper APi for the new Xorg. We use xf86-video-fbdev now instead. We have so may images and pretty sure our people have tried the configurations you named. The images with wayland seem to give some issues but people have learned some workarounds.

Compositing is turned on in the 32 bit image but it is off in the 64bit image. My 64bit image has XFwm4 window manager. My 32bit image has Metacity (Marco) window manager

Dark-Sky, 2022-04-13:
@popcornmix I started from the ground up building the kernel with make bcm2711_defconfig and left generated .config as is and compiled the kernel and installed it on a fresh 64bit Mate image and I get the same mouse issue as with the 64bit XFCE image.

> I wonder if it's a similar issue to #4571. Does it occur only when playing a video or rendering on the GPU, or does it also stall the rest of the display (so something like displaying a huge chunk of text in a terminal would be stalled as well?)
> 

When testing the kernel with the new Mate image I tested this also and moving the mouse while typing in the terminal there is no text output until you stop the mouse and all of the text then appears.

Dark-Sky, 2022-04-14:
@popcornmix I installed the same last kernel to a fresh 64bit RPi OS image and had no issues like you. I then got to looking at your group file and saw a render group so I went back to my XFCE image and logged in as root and the mouse seemed ok along with moving the mouse and glxgears did not stop. Did you all recently add something different with your udev rules between 5.15.32 and 5.15.33 or better yet did these 2 commits in question here do something involving permissions?

Dark-Sky, 2022-04-15:
@popcornmix 

Using the 5.15.33 kernel I had compiled using "make bcm2711_defconfig" and using the generated .config as is:

I have my XFCE image some what working as a normal user with KMS but it is still not quite right with some lagging with high bit rate videos. I added a few udev rules that I saw were missing involving 12c & gpio but with my limited knowledge I believe by adding me to the render group (along with adding me to the new gpio/i2c groups) might have got it working.

I then booted my XFCE image using FKMS and the mouse issue was back. I went back to the PiOS image that has the same 5.15.33 kernel that I compiled that had no issue with KMS and booted into FKMS and the mouse issue is showed up in it.

You can see each time when I moved the mouse in this pic looking at the frames/FPS in glxgears:

[image]

Dark-Sky, 2022-04-17:
@popcornmix 

This is now showing up in the arch-arm forums:

https://archlinuxarm.org/forum/viewtopic.php?f=57&t=15999

popcornmix, 2022-04-19:
> I then got to looking at your group file and saw a render group

I have seen the render group on some systems, but pi doesn't seem to belong to it on RpiOS:
```
pi@pi4:~ $ groups pi
pi : pi adm tty dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio lpadmin
```
and I see nothing from `sudo find /dev -group render`
I think we use video group for all the 3d/drm/v4l2 nodes.
But this isn't something I set up, so I'm just going by what I see when poking around.

popcornmix, 2022-04-19:
> Did you all recently add something different with your udev rules between 5.15.32 and 5.15.33 or better yet did these 2 commits in question here do something involving permissions?

No, I don't believe so.

popcornmix, 2022-04-19:
> I have my XFCE image some what working as a normal user with KMS but it is still not quite right with some lagging with high bit rate videos. The same kernel performance wise is not as good as on the PiOS image. I added a few udev rules that I saw were missing involving 12c & gpio but with my limited knowledge I believe by adding me to the render group (along with adding me to the new gpio/i2c groups) might have got it working.

> I then booted my XFCE image using FKMS and the mouse issue was back. I went back to the PiOS image that has the same 5.15.33 kernel that I compiled that has no issue with KMS and booted into FKMS and the mouse issue showed up in it also.

So are you saying adding your user to render group fixes the regression from the two commits we've identified using kms?
What does `sudo find /dev -group render` show?

But fkms still has a similar issue with the two commits (although that is a deprecated configuration, so less critical than kms).

Dark-Sky, 2022-04-20:
> So are you saying adding your user to render group fixes the regression from the two commits we've identified using kms? What does `sudo find /dev -group render` show?

```
ray@raspberrypi:~ $ sudo find /dev -group render
/dev/dri/renderD128
```

It says this here on all of the images I have tested including PiOS.

It is fixed regarding the mouse movement issue on my XFCE image with KMS but there is lacking with performance with higher bitrate videos locally and on youtube. More dropped frames on youtube. I do not have this issue with the 2 patches reverted. 64 bit PiOS does not have this issue.

I had forgotten the PiOS Setup now lets you set up me as a user but the default in the 2022-04-04-raspios-bullseye-arm64 image has pi as a member of the render group when mounting the image on my x86. After going through the Setup and rebooting all mention of the user pi is removed from /etc/group and replaced with the user you set up:

[image]

> But fkms still has a similar issue with the two commits (although that is a deprecated configuration, so less critical than kms).

Yes I am seeing this issue with PiOS with FKMS. I am a little sad about hearing FKMS being  deprecated some of our DE's has issues from time to time with KMS like Sway and Plasma and it was nice to be able to switch back and forth. I am having a weird issue right now with the latest -rc kernel where KMS boots but FKMS will not even give a Rainbow at boot but a black screen.

popcornmix, 2022-04-20:
> ray@raspberrypi:~ $ sudo find /dev -group render
/dev/dri/renderD128

On RPiOS, render isn't used for that node:
```
$ ls -l /dev/dri/renderD128 
crw-rw----+ 1 root video 226, 128 Apr 20 11:18 /dev/dri/renderD128
```
but it should be accessible by a normal used. So you should be adding your user to the render group (or change group of that node to video, I assume in udev rules).

Dark-Sky, 2022-04-20:
> > ray@raspberrypi:~ $ sudo find /dev -group render
> > /dev/dri/renderD128
> 
> On RPiOS, render isn't used for that node:
> 
> ```
> $ ls -l /dev/dri/renderD128 
> crw-rw----+ 1 root video 226, 128 Apr 20 11:18 /dev/dri/renderD128
> ```
> 
> but it should be accessible by a normal user. So you should be adding your user to the render group (or change group of that node to video, I assume in udev rules).

Obviously we are not using the same RPiOS image. We have not been on the same page here a few times. Just so I did not mess up and got in the wrong window changing things on my desktop I have downloaded a new RPiOS image (2022-01-28-raspios-bullseye-arm64.zip) and went through the Setup where it asks me to create my user account then it did a package update then it rebooted. The default "pi" user was deleted and all pi references in /etc/group was removed and replaced with the user I created. I changed nothing and "ls -l /dev/dri/renderD128" still showed up as I previously posted.

[image]

Dark-Sky, 2022-04-22:
More reporting this on arch-arm in the last few days.

https://archlinuxarm.org/forum/viewtopic.php?f=57&t=15999

mripard, 2022-05-04:
I've been trying to reproduce it this afternoon and can't really see any meaningful difference with and without that patch applied.

glxgears will stall for a couple of frames from time to time, but it looks to me that without the reverted patch it's actually worse. So I think I can't reproduce it.

I've noticed you were using the fbdev DDX, whereas raspios uses the modeset DDX. Could you paste /var/log/Xorg.0.log and your xorg configuration?

mripard, 2022-05-06:
So just to make sure we're on the same page, you can only reproduce this on FKMS, with old firmwares?

Dark-Sky, 2022-05-06:
@mripard It does not make any difference which firmware is installed with the 2 commits installed when booted with FKMS. The mouse issue is present. I got the same thing testing with a fresh PiOS image but at that time I had a Feb eeprom flashed. I had not noticed that the new eeprom  flashes were not taking until a few days ago and had to flash a recovery image.

Every since the issue with the 2 commits started I have not released any of the kernels to manjaro's repo's with out them being reverted but I knew at some point arch-arm would have the same issue as we mirror their repo.

I would guess that maybe 50% of our users use FKMS. We release quite a few images with various DE's and some have had various issues with KMS (Plasma and Sway at the top of the list). Also there are some issues with some monitors / tv's with KMS. For these reasons we have been releasing our RPi images with FKMS as default as it has been less problematic and the users will switch to which mode that worked best for their use case. So hearing that FKMS is being depreciated and not a priority was not good news.

I had another issue a couple of weeks ago where FKMS would not boot into a DE that was about the same time where all kernels would go to a black screen that was fixed. Wonder if this might be a remnant of that issue still present coupled with the 2 commits made?

I am also seeing something else lately that may or may not be related with FKMS that is really irritating. When doing a reboot or powering on the pi4 after switching to FKMS to test it instantly at boot will give a black screen with out showing the rainbow screen and switch to my desktop's VIZIO tv's hdmi port and I have to switch back to the hdmi port my pi4 is connected to. This does not happen if I have KMS configured to boot.

mripard, 2022-05-06:
> @mripard It does not make any difference which firmware is installed with the 2 commits installed when booted with FKMS. The mouse issue is present.

But that doesn't occur with KMS, right?

> I got the same thing testing with a fresh PiOS image but at that time I had a Feb eeprom flashed. I had not noticed that the new eeprom flashes were not taking until a few days ago and had to flash [...]

It would be great to report those issues so that we can get rid of them and avoid the need / desire for FKMS

> Also there are some issues with some monitors / tv's with KMS. For these reasons we have been releasing our RPi images with FKMS as default as it has been less problematic and the users will switch [...]

I don't know fkms that well, so I can't really help you there, sorry.

Could you test the following patch to see if it prevents the mouse issue? (total shot in the dark)

```patch
diff --git a/drivers/gpu/drm/vc4/vc4_kms.c b/drivers/gpu/drm/vc4/vc4_kms.c
index dd97e1def0cb..1f20d255e6d2 100644
--- a/drivers/gpu/drm/vc4/vc4_kms.c
+++ b/drivers/gpu/drm/vc4/vc4_kms.c
@@ -450,14 +450,6 @@ static int vc4_atomic_commit_setup(struct drm_atomic_state *state)
 	struct drm_crtc *crtc;
 	unsigned int i;
 
-	/* We know for sure we don't want an async update here. Set
-	 * state->legacy_cursor_update to false to prevent
-	 * drm_atomic_helper_setup_commit() from auto-completing
-	 * commit->flip_done.
-	 */
-	if (!vc4->firmware_kms)
-		state->legacy_cursor_update = false;
-
 	hvs_state = vc4_hvs_get_new_global_state(state);
 	if (WARN_ON(IS_ERR(hvs_state)))
 		return PTR_ERR(hvs_state);
```

Dark-Sky, 2022-05-06:
Be glad to.

Dark-Sky, 2022-05-06:
@mripard There was no improvement with your patch with with mouse movement with FKMS.

VK2BEA, 2022-05-21:
My system just updated to kernel 5.15.38 and I am seeing this effect (although glxgears is OK).
I'm using the cinnamon desktop and it has become unusable.
Mouse movement is very laggy. The mouse cursor movement is smooth and starts with movement of the mouse but it catches up to the actual position after about a second. Disabling acceleration has no effect.
When moving windows, they jump after around a second  rather than following the mouse.
gdm and gnome desktop seem OK (although there is some window tearing).

What's the procedure to boot into a different kernel?

Dark-Sky, 2022-05-21:
> What's the procedure to boot into a different kernel?

If your situation is like here the question is not a different kernel generally as all of their kernels has this issue assuming you have dtoverlay=vc4-fkms-v3d enabled in /boot/config.txt. FKMS is not a priority for them now for support. You can try:

`dtoverlay=vc4-kms-v3d`

If you have issues with this with your DE then you can disable all KMS, FKMS in config.txt and install xf86-video-fbdev but you will loose V3D as a Renderer.

You can also install a upstream kernel which works pretty good here.

VK2BEA, 2022-05-21:
> 
> One last thing one person had pretty good luck with dtoverlay=vc4-kms-v3d by upgrading to the latest rpi-eeprom on his pi4 and the firmware in /boot/ if that may be your use case.

That worked - thanks!  
As you suggested, I updated the eeprom to the latest beta and changed the config from `dtoverlay=vc4-fkms-v3d` to `dtoverlay=vc4-kms-v3d`.

VK2BEA, 2022-05-21:
> If your situation is like here the question is not a different kernel generally as all of their kernels has this issue assuming you have dtoverlay=vc4-fkms-v3d enabled in /boot/config.txt. FKMS is not a priority for them now for support.

As an aside, I updated to F36 about a week ago and all was fine. 
I copied over an image I made then, disabled kernel updates and updated and still it was fine, so this does lead me to believe that there was a kernel update that broke things. (5.15.32 OK but 5.15.38 bad).  (the config in both cases had dtoverlay=vc4-fkms-v3d)

Dark-Sky, 2022-05-21:
That is correct. After 5.15.32 and all other of their kernels at that time is the point where FKMS broke.

pelwell, 2022-08-24:
See #5141 as a potential fix for the laggy mouse cursor problem.

Strit, 2022-08-24:
Thanks. Gonna test this out tonight. 

Dark-Sky had a heart attack and is recovering, so he won't be able to test this himself.

pelwell, 2022-08-24:
Oh gosh, I'm really sorry to hear that. Get well soon, Ray.

Strit, 2022-08-24:
Got some time to test the fix with fkms enabled in /boot/config.txt.

Seems to work fine for me too. Thanks for the hard work.

pelwell, 2022-08-26:
I'm closing this, since it seems the problem is fixed (at least in our source tree). Please re-open or comment if it turns out that isn't the case.

closed 2022-08-26 by pelwell