raspberrypi/linux #4992: 5.18.y - dwc2 module inactive

pavhofman, 2022-04-13:
### Describe the bug

I am trying to test the dwc2 in gadget mode on the latest 5.18.y branch, in order to test the latest usb gadget patches submitted to mainline 5.18. Loading the dwc2 module does not produce any log in dmesg, neither any feXXXXX entry gets created in /sys/kernel/debug/usb/. My compiled version includes the latest patches pushed to https://github.com/raspberrypi/linux/blob/rpi-5.18.y/arch/arm/boot/dts/overlays/dwc2-overlay.dts.

5.17 with basically identical .config (by RPi) loads dwc2 OK, working OK too.

Please is the 5.18.y branch already capable of testing, or should I wait for more rpi-specific patches to be pushed to make dwc2 work?

Thanks a lot.

Pavel.

### Steps to reproduce the behaviour

modprobe dwc2

root@raspberrypi:/sys/kernel/debug/usb# ls
devices  xhci

### Device (s)

Raspberry Pi 4 Mod. B

### System

root@raspberrypi:/sys/kernel/debug/usb# cat /etc/rpi-issue
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, c12b1df4ed6416fb0df33ba1731c5b13c1bdbdf8, stage4

root@raspberrypi:/sys/kernel/debug/usb# vcgencmd version
Oct 29 2021 10:47:33 
Copyright (c) 2012 Broadcom
version b8a114e5a9877e91ca8f26d1a5ce904b2ad3cf13 (clean) (release) (start)

root@raspberrypi:/sys/kernel/debug/usb# uname -a
Linux raspberrypi 5.18.0-rc2-v8+ #2 SMP PREEMPT Wed Apr 13 05:52:51 UTC 2022 aarch64 GNU/Linux

popcornmix, 2022-04-13:
> Please is the 5.18.y branch already capable of testing, or should I wait for more rpi-specific patches to be pushed to make dwc2 work?

The bleeding edge kernels (anything newer than stable 5.15, but especially the latest -rc version is for testing rather than for use in a stable system.
But 5.18 is believed to be feature complete - there are no known missing patches.

But there are always regressions in a new kernel. Either bugs from upstream, or changes to how modules are configured (that perhaps need new `CONFIG_*` options enabled.

A quick check doesn't look like any changes in `CONFIG_` options between 5.17 and 5.18:
```
CONFIG_USB_DWCOTG=y
# CONFIG_USB_DWC3 is not set
CONFIG_USB_DWC2=m
# CONFIG_USB_DWC2_HOST is not set
# CONFIG_USB_DWC2_PERIPHERAL is not set
CONFIG_USB_DWC2_DUAL_ROLE=y
# CONFIG_USB_DWC2_PCI is not set
# CONFIG_USB_DWC2_DEBUG is not set
# CONFIG_USB_DWC2_TRACK_MISSED_SOFS is not set
```
```
CONFIG_USB_DWCOTG=y
# CONFIG_USB_DWC3 is not set
CONFIG_USB_DWC2=m
# CONFIG_USB_DWC2_HOST is not set
# CONFIG_USB_DWC2_PERIPHERAL is not set
CONFIG_USB_DWC2_DUAL_ROLE=y
# CONFIG_USB_DWC2_PCI is not set
# CONFIG_USB_DWC2_DEBUG is not set
# CONFIG_USB_DWC2_TRACK_MISSED_SOFS is not set
```
I'd suggest you stick with a more stable kernel if this is important to you.
It will be fixed on 5.18 in time (possibly just by waiting for an upstream fix), but I can't promise when.

pavhofman, 2022-04-13:
Thanks, I asked at the mailing list https://marc.info/?l=linux-usb&m=164984983207760&w=2 , I believe it will be resolved fast.

lategoodbye, 2022-04-14:
Hi Pavel,
reporting issues with non-mainline kernel on lkml are most likely ignored. Do you have a chance to reproduce it with mainline kernel and DT?

pavhofman, 2022-04-15:
A bit of troubleshooting identified the culprit.

File  arch/arm64/configs/bcm2711_defconfig added by RPi in commit e0ad0c0927b10f2f1426f989d8f40f53aa2c3b3e to branch rpi-5.17.y (and to branch rpi-5.16.y and previous...) contains option

`CONFIG_RASPBERRYPI_POWER=y`

However, the equivalent commit 48457714f9fe3abb3088486710004a71ce305b34 pushed to branch rpi-5.18.y does not have this config set. As a result the option does not get transferred to .config when running the command 
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig`

as described in https://www.raspberrypi.com/documentation/computers/linux_kernel.html#64-bit-configs

Therefore the raspberrypi-power driver does not get built into the kernel image, causing the status of the soc:power supplier in the dwc2 module as DL_STATE_DORMANT instead of DL_STATE_AVAILABLE, leading to the deferred status of the driver upon probing.

Setting this option in .config resolved the issue, the dwc2 gadget works OK now.

popcornmix - please can you fix the issue? Both arch/arm/configs/bcm2711_defconfig and arch/arm64/configs/bcm2711_defconfig lack this config in 5.18.y.

Thanks a lot.
Pavel

Masonrf, 2022-04-15:
> A bit of troubleshooting identified the culprit.
> 
> File arch/arm64/configs/bcm2711_defconfig added by RPi in commit [e0ad0c0](https://github.com/raspberrypi/linux/commit/e0ad0c0927b10f2f1426f989d8f40f53aa2c3b3e) to branch rpi-5.17.y (and to branch rpi-5.16.y and previous...) contains option
> 
> `CONFIG_RASPBERRYPI_POWER=y`
> 
> However, the equivalent commit [4845771](https://github.com/raspberrypi/linux/commit/48457714f9fe3abb3088486710004a71ce305b34) pushed to branch rpi-5.18.y does not have this config set. As a result the option does not get transferred to .config when running the command `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig`
> 
> as described in https://www.raspberrypi.com/documentation/computers/linux_kernel.html#64-bit-configs
> 
> Therefore the raspberrypi-power driver does not get built into the kernel image, causing the status of the soc:power supplier in the dwc2 module as DL_STATE_DORMANT instead of DL_STATE_AVAILABLE, leading to the deferred status of the driver upon probing.
> 
> Setting this option in .config resolved the issue, the dwc2 gadget works OK now.
> 
> popcornmix - please can you fix the issue? Both arch/arm/configs/bcm2711_defconfig and arch/arm64/configs/bcm2711_defconfig lack this config in 5.18.y.
> 
> Thanks a lot. Pavel

I can confirm that this fix works on RPi Zero 2 W

popcornmix, 2022-04-19:
@pavhofman thanks for tracking that down. I've added `CONFIG_RASPBERRYPI_POWER` to defconfigs on rpi-5.18.y

popcornmix, 2022-05-10:
@pavhofman is this issue resolved for you now? Okay to close?

pavhofman, 2022-05-10:
Yes please.

closed 2022-05-10 by popcornmix

carylewis, 2023-09-13:
I am experiencing an issue on the Raspberry Pi 4 that sounds similar to this - I am trying to emulate a HID keyboard with the Pi, and this was working before I upgraded to "Bullseye". Now nothing is reported in dmesg when I plug an Android pixel 6a into the PI, whereas before the upgrade it worked.

How can I confirm if the dwc2 module is not in the correct power state?

pavhofman, 2023-09-13:
If your .config contains CONFIG_RASPBERRYPI_POWER=y, the cause is most likely something else.

carylewis, 2023-09-13:
I didn’t build the kernel, I just used rpi-update.

lategoodbye, 2023-09-13:
@carylewis Could you please tell the kernel version before and after the update?

carylewis, 2023-09-13:
Thanks for you help, it appears as though the power setting is correct:

```
uname -r
6.1.51-v8+

sudo modprobe configs
zgrep RASPBERRYPI_POWER /proc/config.gz
CONFIG_RASPBERRYPI_POWER=y
```

I don't know what changed but at one point the key mime pi project worked and now it doesn't.

I get this error:

`Cannot send after transport endpoint shutdown`

I will reflash to buster and see if I can get it working again.