raspberrypi/linux #4993: Compile the kernel with SYNPROXY module

Roki100, 2022-04-13:
### Describe the bug

The raspberry pi os's kernel is compiled without SYNPROXY module, which is pretty useful for some projects

### Steps to reproduce the behaviour

Try to use SYNPROXY via iptables

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-01-28
Mar 24 2022 13:19:26
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

### Additional context

*I got told to make an issue here: https://forums.raspberrypi.com/viewtopic.php?t=332870*
So, generally having SYNPROXY won't affect the general performance for everybody using the kernel itself, the memory usage impact should be zero, yet it would be helpful and useful to have it for people that want to use it
It can be used for various use cases, where one of them is a DDoS protection, and as we all know, there are a lot of people using Raspberry Pi as a small server for various things
I don't see any reason on why currently the kernel is compiled without SYNPROXY, as there is no gain on this, the real gain is on compiling it imho
Please consider compiling the kernel with SYNPROXY module, as it expands firewalling possibilities which are good for security, yet it shouldn't impact any user that won't use it.

pelwell, 2022-04-19:
There are a few "SYNPROXY" config settings, but I see from the thread that it is `CONFIG_NFT_SYNPROXY=m` that is required - which is in now.

popcornmix, 2022-04-22:
Should be in latest rpi-update kernel.

closed 2022-07-20 by pelwell