raspberrypi/linux #4997: CONFIG_IR_TOY not enabled for 5.13, 5.14, 5.15 (but was enabled older versions)

seanyoung, 2022-04-16:
### Describe the bug

The [IR Toy](https://www.irdroid.com/irdroid-usb-ir-transceiver/) has a kernel driver makes it possible to use it with standard IR tooling (ir-ctl, ir-keytable). This module is not enabled on newer kernels. It was enabled on kernel 5.10.

### Steps to reproduce the behaviour

- Insert IR Toy or IR Droid device
- Notice there is no `/dev/lirc0` device
- Device cannot be used

### Device (s)

Raspberry Pi 3 Mod. B

### System

Neither the 5.13, 5.14 and 5.15 tree has this option in the defconfig, see https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm64/configs/bcm2711_defconfig

pelwell, 2022-04-19:
I'm not sure what happened there, but IR_TOY should be enabled in future builds.

seanyoung, 2022-04-19:
@dax-se

popcornmix, 2022-04-22:
Should be in latest rpi-update kernel.

popcornmix, 2022-05-10:
@seanyoung okay to close?