raspberrypi/linux #4998: Bad merge in 5.10.110? Patches missing in drivers/gpu/drm/drm_edid.c

psgreco, 2022-04-16:
### Describe the bug

IIUC, b9288431b2da9f61cacc5ab7adba1a6cd82a5c3d is an old (unmerged version) of a patch, which seems to have been replaced by fd020eaaa24a676f11a58c527dcd928f18d66a6c and 1ba10e5c39d41ad2c410d9db13ca2bcbbc0286a2 .

Also ba09b04173242c379d2442be979ab2f1971f2f8b seems to be missing

### Steps to reproduce the behaviour

compare source code with upstream.

### Device (s)

Other

### System

v5.10.110

popcornmix, 2022-04-19:
I believe during the merge there were conflicts in `drivers/gpu/drm/drm_edid.c`, and I chose to keep the downstream changes rather than the upstream change (they both fix the same issue in reporting RGB444 support).

It debatable which is the best choice, but as 5.10 branch is legacy (bullseye is on 5.15) and doesn't get much testing there's some advantage to using a form of a fix that has been tested by us.

The missing "drm/edid: check basic audio support on CEA extension block" is a little surprising where it got lost (it's not clear to me from the git history), but it does cherry-pick cleanly, so I can add it back in.

It does look like a revert of "drm/edid: Don't clear YUV422 if using deep color" and a cherry-pick of upstream "drm/edid: Don't clear formats if using deep color" does apply cleanly ("drm/edid: Always set RGB444" is already present and cherry-picks as empty), and with the audio commit leaves `drivers/gpu/drm/drm_edid.c` almost identical to upstream (just differing by "drm/edid: Rename drm_hdmi_avi_infoframe_colorspace to _colorimetry")

Have you found any issues with the kernel as it is?
I'm happy to cherry-pick the audio commit. The upstream form of RGB444 fix I might need a bit more persuading, as I believe we have a correctly working, tested version of it.

macmpi, 2022-04-19:
> The missing "drm/edid: check basic audio support on CEA extension block is a little surprising where it got lost"

@popcornmix could that be a cause of missing audio noticed in https://github.com/raspberrypi/firmware/issues/1647#issuecomment-1059281180 ?
Thanks.

popcornmix, 2022-04-19:
> @popcornmix could that be a cause of missing audio noticed in https://github.com/raspberrypi/firmware/issues/1647#issuecomment-1059281180 ?

Not sure - I've asked user to test with rpi-update kernel (which does contain this commit from 5.15 tree).

psgreco, 2022-04-19:
I haven't had any issues, no. I just normally do a full rebase to have a clean patch on top of the corresponding upstream kernel and noticed the difference

closed 2022-07-20 by pelwell