raspberrypi/linux #4999: i2c-rtc overlay broken due to missing i2c0mux symbol

uski, 2022-04-17:
### Describe the bug

Hi,

I just intalled a fresh Raspbian Lite on a : 
```
Hardware : BCM2835
Revision : 0002
Model : Raspberry Pi Model B Rev 1
```

I have a [DS3231 module](https://www.pishop.us/product/ds3231-real-time-clock-module-for-raspberry-pi/) connected to it, which is visible via `i2cdetect -y 0` at address 0x68 on I2C bus 0, as expected.

I am able to manually register the RTC module by doing :
`echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-0/new_device`

So this shows that the hardware part is all OK.

However, and this is the bug, enabling the corresponding dtoverlay does not register the RTC module.

In config.txt I have :
```
dtparam=i2c_arm=on
dtdebug=1
dtoverlay=i2c-rtc,ds3231,i2c0
```

Running `vcdbg log msg` shows :
```
002205.698: dtparam: i2c_arm=on
002208.208: dtdebug: found override i2c_arm
002208.247: dtdebug:   override i2c_arm: string target 'status'
002213.565: brfs: File read: 1881 bytes
002220.489: dtdebug: Opened overlay file 'overlays/i2c-rtc.dtbo'
002221.952: brfs: File read: /mfs/sd/overlays/i2c-rtc.dtbo
**002244.966: dterror: can't find symbol 'i2c0mux'**
002244.979: Failed to resolve overlay 'i2c-rtc'
```

This show that the I2C interface is properly enabled via dtparam, however, the dtoverlay fails due to a missing symbol i2c0mux.

I tried removing "i2c0" parameter from the i2c-rtc overlay, however, that does not resolve the issue, I still get this error :
`002247.760: dterror: can't find symbol 'i2c0mux'`

So it really looks like this overlay cannot be parsed at all. That's the bug.

At the time of writing this message, this is a snipper from the i2c-rtc-overlay.dts file from the Raspberry Pi Kernel on GitHub :
```
	fragment@102 {
		target = <&i2c0mux>;
		__dormant__ {
			status = "okay";
		};
	};
```

So we do have a reference to i2c0mux.

I don't know how to debug this further, I get lost in the tree of overlays that get applied on top of each other.
Any help ? It seems like something is broken in the chain of overlays that is selected for each specific board.

Thanks

### Steps to reproduce the behaviour

- Install fresh Raspbian Lite on Raspberry Pi 1
- Add DS3231 module to the header
- Enable the i2c-rtc overlay with ds3231 parameter

Expected behavior : module gets picked up by the kernel, with drivers registered

Observed behavior : parse error of the overlay module

### Device (s)

Raspberry Pi 1 Mod. B

### System

```
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux
pi@raspberrypi:~ $ vcgencmd version
Dec  1 2021 15:07:06 
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)
```

pelwell, 2022-04-17:
I haven't looked at the overlay source yet, but a missing symbol like this should be easily fixable.

pelwell, 2022-04-19:
The attached zip file includes an updated version of the Rev 1 Model B dtb that declares the symbols i2c0mux, which should allow the i2c-rtc overlays to be applied.
[bcm2708-rpi-b-rev1.zip](https://github.com/raspberrypi/linux/files/8513237/bcm2708-rpi-b-rev1.zip)
Unzip it and copy it into `/boot` (or the root of the FAT partition if the card is mounted on another computer). If that works for you it will be released as an update to the official dtb.

There are a few things to note:
1. It shouldn't be necessary to use the `i2c0` parameter on the Rev 1 board - that should be the default. It would be good if you can try with and without 
2. The updated dtb still doesn't include real i2c-mux functionality for i2c1 (the I2C interface shared with the firmware). This is unlikely to cause problems since nothing has been reported so far, but it seemed worth mentioning.

pelwell, 2022-04-24:
The dts file with the added `i2c0mux` symbol has been merged. The updated dtb file will appear in future kernel releases.

closed 2022-07-08 by pelwell